WouterJD/FortiusANT #89: Unable to initialize Trainer

JordiCostaLo, 2020-05-19:
Hi @WouterJD.
Finally, I was able to return two Cyclops i purchase, and now I have two suunto movestick. As you stated they seem to work perfectly. But, now I have another problem. Find attached log file. Seem that TACX - unable to initialize traine when I try to execute FORTIUSANT.EXE.
I am sure that something I am doing is wrong.

Could you help me in this matter?

Muchas Gracias from Spain

JordiCostaLo, 2020-05-19:
sorry!!
the log file!!

[FortiusANT.2020-05-19 11-31-36.log](https://github.com/WouterJD/FortiusANT/files/4650119/FortiusANT.2020-05-19.11-31-36.log)

WouterJD, 2020-05-19:
Buenas noches!

```
11:31:37,207: GetTrainer - Found uninitialised Fortius head unit
11:31:37,347: GetTrainer - Initialising head unit, please wait 5 seconds
11:31:42,359: GetTrainer - Unable to initialise trainer
11:31:42,359: GetTrainer() returns, trainertype=0xe6be
```

It appears you have a Tacx trainer where software must be loaded in the headunit.
This software is available on github
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/tacxfortius_1942_firmware.hex

and must be placed in the same folder where the executable is stored.

Sorry, the error message is not very clear and needs improvement.

WouterJD, 2020-05-19:
hmmm... The firmware should be contained in the executable, so you have to test a bit. I have improved error-messages but those will appear in next version.
Sorry for inconvenience.

JordiCostaLo, 2020-05-19:
It works!!
Thank you very much!!
I have done all you said, and it worked.
Just one thing. Is  it normal that it does not detect the speed sensor?

JordiCostaLo, 2020-05-19:
and also, I wonder if I have to calibrate resistance

JordiCostaLo, 2020-05-19:
I advise Suunto Dongle, it works perfectly with FORTIUSANT. I've two of them in the same computer, and I have not got any problema.

Muchas gracias and Goede nacht!!!

WouterJD, 2020-05-19:
[Version 3.0 is released](https://github.com/WouterJD/FortiusANT/wiki/Version-information#version-30-released-may-19th-2020) with improved error handling and integrated firmware.
Of course of there are any issues, let me know.

If you have issues with speed-sensor, please let me know.

closed 2020-05-19 by WouterJD