WouterJD/FortiusANT #90: Tacx fortuis multiplayer

heidznz, 2020-05-23:
Hi team

New here and wanting to know if I can hook up the fortuis multiplayer to zwift.

I've been hunting online and can't find the answer.

Am looking to buy second hand trainer and need to be sure it suits me

Cheers from NZ

Heidi

WouterJD, 2020-05-23:
hi Heidi! Welcome in the FortiusANT team!

WouterJD, 2020-05-23:
I have a Tacx Fortius myself and I'm quite convinced it will work. Did not hear from Fortius owners that it does not.
You need two ANTdongles; one for FortiusANT, one for Zwift/Trainer Road/Rouvy (unless you run that from another ANTenabled device).

I would say go for it - I will assist to my best knowledge.

NZ = New Zealand? Great - you add a flagged country on the FortiusANT map!

heidznz, 2020-05-23:
Thank u so much! All the tech is a little.over my head hahha

How do I know if a device is ANT able and where wld I buy a dongle from? Anywhere?

Yes lil old NZ!!!!! Gorgeous country!

WouterJD, 2020-05-24:
Heidi, hi!

I expect you are a windows user, please read the [wiki](https://github.com/WouterJD/FortiusANT/wiki#installation);  If you have questions, drop a note.

SInce you do not know whether you have an ANT-enabled devices; you probably have not.
Read the [wiki](https://github.com/WouterJD/FortiusANT/wiki/FortiusANT-Configurations---ANT--Master-and-Slave-devices). Also: if you have questions, let me know.

And then the final: you need two ANTdongles. Read #61. Be carefull with CYCPLUS dongles, there is a series that cause a lot of headache.

Succes!
Don't worry, it will work!

WouterJD, 2020-05-26:
Hi Heidi. How are things going?

heidznz, 2020-05-26:
Hey

Thanks heaps for getting in touch

I decided to get a boost trainer and then a BLE sensors and I'm all signed
up and did a ten min ride today to get used to it

Thanks sooo much for yr help

Heidi

WouterJD, 2020-05-26:
OK have a lot of fun and Boost your performance!

closed 2020-05-26 by WouterJD