WouterJD/FortiusANT #92: Ant+ disconnection when doing Workouts

danielopez96, 2020-05-29:
Hello to everyone from Bogotá, Colombia (Southamerica).

As I mentioned in the title, everytime when I execute FortiusANT (the latest update), and then go onto Zwift to do a Workout, the Ant+ is lost, in some point of the workout the trainer just disconnects and does not connect again.

When I look at FortiusANT everything appears normal, connected and calibrated, but for some reason Zwift just does not detects the connection.

This happens only when doing workouts, not when doing free-riding. I'm using bot Cycplus Ant+ dongles, I know they may not work properly, but this problem seems strange.

Thanks for the attention, sorry if I made any grammar mistakes.

WouterJD, 2020-05-30:
Please run FortiusANT with-d127 option to create debug-logfile. Stop training when problem occurs so that logfile is not longer than needed. Create separate issue on GitHub and attach logfile.

Succes! And your grammar is 👌

WouterJD, 2020-06-08:
No follow-up in a week.
If issue needs to be reopened, feel free to do so.

closed 2020-06-08 by WouterJD