WouterJD/FortiusANT #94: Newbie/non techy  request

Langsett, 2020-06-08:
Please can you tell me if we need to load everything in the code section, if there is a specific order and  does the computer need to be connected to the Tacx.?

I know nothing about  software, programmes other than click to download, so sorry

WouterJD, 2020-06-08:
Hi @Langsett welcome in the FortiusANT team!

Perhaps the [wiki](https://github.com/WouterJD/FortiusANT/wiki) can be improved, I'll look into that.

From FortiusANT point of view, all you need to do is dowload [FortiusANT](https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe) and store it in any location on your Windows computer.

What Tacx do you have? USB-connected tacx-trainers must be connected to your computer with the USB-cable.
The you need two ANT-dongles; one for FortiusANT and one for Zwift/TrainerRoad, Rouvy or similar.

Then you have to get the USB_driver to be installed properly, which might be the most complicated part: see https://github.com/WouterJD/FortiusANT/wiki#windows

Yes, it's some work but then you can use your legacy Tacx with all the newest software.
Ping me if you hgave more questions.

WouterJD, 2020-06-08:
PS. Where are you from? I always like to know who is using the software and where!

Langsett, 2020-06-08:
Hi Wouter,

I am in Mirfield, Yorkshire ( north England)   I have the Fortius 1.36,   with brake unit  T 190008 ( v 0:19:2)  and head unit T1932

I understand to operate it must be connected  I wondered if it must be connected to load the software, as my Tacx  is in my garage  not the house ?

I am getting a 2nd Ant+  ,  and  look forward to getting  interactive again
Many thanks for your  hard work

Best wishes

Martyn

WouterJD, 2020-06-08:
Welcome again!

headunit 1932 sounds familiar.
There should be a USB connector to be plugged into the PC. FortiusANT reads from USB and transmits through ANTdongle. The second ANTdongle received and is used by Zwift - then of course reverse path back.

Even before dongles arrive you can start and get the trainer connected. In fact that's the most complicated part. The dongles are more p&p

Succes!

desscrambler, 2020-06-09:
Hello Wouter, fantastic FortiusAnt that you developed. I'm Guilherme from Brazil, I have two Fortius one T1942 and T1932, I would like to use them in Zwift. I have two Ant + Garmin. I'm already with ExplorANT and FortiusANT and already installed libusb-win32-bin-1.2.6.0 on Win10. I tested both T19 ** and was unsuccessful. can you help me. Thank you.

WouterJD, 2020-06-09:
The good news is that FortiusANT works under Windows 10 (I use Windows 10 and Windows 7) so you're on the right track.

Operationally, all you need is FortiusANT.exe. (ExplorANT is a tool to do some verifications).

What happens when you start FortiusANT?
First thing to look at are the two messages that are displayed, regarding USB and ANTdongle.

Success!

desscrambler, 2020-06-09:
I have some pictures, zwift locates ANT +, but when I pedal my avatar doesn't move, I just put it in the scroll control. tested it using one and two computers. the roller motor as soon as I start pedaling it hold and release hold and release, 
Thank you for the fast service 
[image]
[image]
[image]

desscrambler, 2020-06-09:
[image]
[image]
[image]

WouterJD, 2020-06-09:
Hi!

- You do not need ExplorANT, unless you want to debug software.
- It all looks good; start pedalling to calibrate. The Tacx Fortius T1942 will then drive the bike and find it's calibration automatically

Succes!

desscrambler, 2020-06-09:
:( In Zwift, I have no increase in power, cadence and does not control resistance, I turn the crank and the tacx brakes after a lap. What can I be doing wrong? I am on only one computer and two Ant + Garmin

[image]
[image]
[image]

desscrambler, 2020-06-09:
Already with the T1942 I have this problem, 
we'll see this later.
[image]
[image]

WouterJD, 2020-06-10:
This headunit requires firmware to be loaded.
This firmware is expected to be unpacked into the mentioned folder (..._IMEI..) 

Please check whether the tacxfortius_1942_firmware.hex file is present there.
The file is present on my system but I cannot test the process, because I do not have a 1942.

@RamonFigueras do you have a suggestion, since you have a T1942

WouterJD, 2020-06-11:
FortiusANT.exe is a self-extractor and creates IMEI and -if stopped normally- also removes it.

If FortiusANT is killed (by closing console) the folder is NOT removed.

So all IMEI folders may be deleted manually when FortiusANT not running.

Check for .hex when FortiusANT is running, not after closing

desscrambler, 2020-06-11:
@RamonFigueras 

> O FortiusANT.exe é um extrator automático e cria IMEI e, se parado normalmente, também o remove.
> 
> Se o FortiusANT for morto (fechando o console), a pasta NÃO será removida.
> 
> Portanto, todas as pastas IMEI podem ser excluídas manualmente quando o FortiusANT não estiver em execução.
> 
> Verifique .hex quando o FortiusANT estiver em execução, não após o fechamento

Good morning, and thanks for responding, as you mentioned the file creates the directory and deletes them after finishing normally. I manually deleted the temporary directories that were previously. Success

WouterJD, 2020-06-11:
:-) Success with next steps

desscrambler, 2020-06-11:
> Olá @desscrambler , @wouter ,
> 
> Em relação ao T1942, não carregue o firmware TacxFortius, verifique se o T1942 está instalado como um dispositivo libusb-win32.
> 
> Para verificá-lo:
> 
> 1 - Clique com o botão direito do mouse no símbolo do Windows no canto esquerdo da tela e selecione Gerenciador de dispositivos na lista.
> 2 - Clique em '>' à esquerda 'dispositivos lilbusb-win32', uma lista de 3 dispositivos deve abrir, provavelmente com o mesmo nome de 'ANT USB Stick 2' (se você tiver dois dongles USB conectados à porta USB do seu PC ) clique com o botão direito do mouse em todos e selecione Propriedades, vá para a guia Eventos e clique na caixa Informações. Em um deles, você deve ler 'Device USB \ VID_3561 & PID_1942 /' + mais algumas informações. Esta é a sua caixa de controle T1942.
> 3 Caso as opções acima não sejam verdadeiras, esse é o seu problema. Para resolvê-lo, siga o procedimento abaixo:
> 
> Enquanto estiver na janela do Gerenciador de dispositivos, pesquise a entrada 'Outros dispositivos'. Se o cabo USB do controlador T1942 estiver conectado, desconecte-o da tomada por 10 segundos. e conecte-o novamente. Um 'dispositivo desconhecido' deve aparecer em 'Outros dispositivos' e siga estas instruções: (em alguns casos, o dispositivo desconhecido aparecerá em dispositivos Jungo ou VR; em qualquer caso, siga as instruções abaixo)
> 
> * Clique com o botão direito do mouse e selecione "atualizar software de driver" 
>   -Selecione "Procurar no meu computador por software de driver" 
>   -Selecione "Deixe-me selecionar em uma lista de drivers de dispositivo no meu computador" 
>   -Selecione dispositivos libusb-win32 
>   -Selecione ANT USB Stick 2 e, em seguida, OK no aviso e feche Agora o seu USB-trainer está instalado como "ANT USB Stick 2", o que funciona, mas talvez não seja um nome muito claro.
> 
> Isso deve resolver o problema.
> 
> E com relação à questão do T1932, eu encontrei uma questão semelhante no meu treinador, verifique se a ponta do sensor de cadência está posicionada corretamente contra o ímã no braço da manivela e perto o suficiente. Se o software ANTForitus não detectar os pulsos do sensor de cadência, ele se comportará de maneira irregular, colocando uma resistência muito forte, seguida por quase nada como o descrito no seu comentário.
> 
> Boa sorte e sucesso,
> 
> Ramon

Right now, I'm on the T1932, those interruptions with the cadence installed, stopped. Thanks for that, I see that the moment I start FortiusANT it informs the roll calibration. at the moment I have speed and cadence on the panel, but I have no power. when I stop pedaling the roller motor and soon after. When I go back to pedaling the cadence and power speed hands are all still, thanks to you for this support. @WouterJD @RamonFigueras

WouterJD, 2020-06-11:
I do not completely understand

The process is as follows:
1. "Start Pedalling" all you need to do is to kick the pedal, then calibration starts
2. "Calibration" FortiusANT will run some time and when the calibration point is found,...
3. "Normal mode" and you can use your Tacx 

In "Normal mode" you need Zwift/Trainer Road/Rouvy to control the T1932.

I do not understand what you mean with "When I go back to pedaling the cadence and power speed hands are all still" perhaps add a foto of the screen so I can see Power, Speed, Cadence and Resistance.

desscrambler, 2020-06-11:
> Eu não entendo completamente
> 
> O processo é como se segue:
> 
> 1. "Comece a pedalar" tudo o que você precisa fazer é chutar o pedal e a calibração começa
> 2. "Calibração" O FortiusANT funcionará algum tempo e quando o ponto de calibração for encontrado, ...
> 3. "Modo normal" e você pode usar o seu Tacx
> 
> No "modo normal", você precisa do Zwift / Trainer Road / Rouvy para controlar o T1932.
> 
> Não entendo o que você quer dizer com "Quando volto a pedalar, a cadência e a velocidade da força ainda estão em movimento" talvez adicione uma foto da tela para poder ver Potência, Velocidade, Cadência e Resistência.

The panel informs me of speed and cadence when calibrating.

[image]

After calibrating, When I re-pedal the hands on the panel, the cadence and power speed are all stopped.

[image]

WouterJD, 2020-06-11:
What program do you use? Zwift?

Please start FortiusANT with -d127 flag and send the resulting logfile.

In case you start FortiusANT directly by double-clicking untill now,
you may use https://github.com/WouterJD/FortiusANT/blob/master/StartUp/FortiusAnt%20(exe).bat
to start FortiusANT.exe

```
..\WindowsExecutable\FortiusAnt.exe -d127
pause

```

WouterJD, 2020-06-11:
The images show that you are connected to a program that sets the grade (1%).

It also shows that you do not ride the bike, since Speed and Cadence are zero.
And since your speed is zero, the target power is zero (standing still does not require power)

I assume that you ARE riding the bike, so there must something strange there

desscrambler, 2020-06-11:
> As imagens mostram que você está conectado a um programa que define a nota (1%).
> 
> Também mostra que você não anda de bicicleta, pois Velocidade e Cadência são zero.
> E como sua velocidade é zero, o poder alvo é zero (ficar parado não requer energia)
> 
> Eu suponho que você está andando de bicicleta, então deve haver algo estranho lá

I'm pedaling, but the panel doesn't grow, what can I be doing wrong?

desscrambler, 2020-06-11:
> Qual programa você usa? Zwift?
> 
> Inicie o FortiusANT com o sinalizador -d127 e envie o arquivo de log resultante.
> 
> Caso você inicie o FortiusANT diretamente clicando duas vezes até agora,
> use https://github.com/WouterJD/FortiusANT/blob/master/StartUp/FortiusAnt%20(exe).bat
> para iniciar o FortiusANT.exe
> 
> ```
> ..\WindowsExecutable\FortiusAnt.exe -d127
> pause
> ```
[FortiusANT.2020-06-11 12-42-47.log](https://github.com/WouterJD/FortiusANT/files/4764817/FortiusANT.2020-06-11.12-42-47.log)

desscrambler, 2020-06-11:
He scores at the beginning and then no more. :(
[image]

desscrambler, 2020-06-11:
> > Olá @desscrambler , @wouter ,
> > Em relação ao T1942, não carrega o firmware TacxFortius, verifique se o T1942 está instalado como um dispositivo libusb-win32.
> > Para verificá-lo:
> > 1 - Clique no botão direito do mouse no símbolo do Windows no canto esquerdo da tela e selecione Gerenciador de dispositivos na lista. 
> > 2 - Clique em '>' à esquerda 'dispositivos lilbusb-win32', uma lista de 3 dispositivos deve abrir, provavelmente com o mesmo nome de 'ANT USB Stick 2' (se você tiver dois dongles USB conectados à porta USB do seu PC ) clique com o botão direito do mouse em todos e selecione Propriedades, vá para o guia Eventos e clique na caixa Informações. Em um deles, você deve ler 'Device USB \ VID_3561 & PID_1942 /' + mais algumas informações. Esta é sua caixa de controle T1942. 
> > 3 Caso as opções acima não sejam verdadeiras, esse é o seu problema. Para resolver, siga o procedimento abaixo:
> > Enquanto estiver na janela do Gerenciador de dispositivos, pesquise a entrada 'Outros dispositivos'. Se o cabo USB do driver T1942 estiver conectado, desconecte-o da tomada por 10 segundos. e conecte-o novamente. Um 'dispositivo desconhecido' deve aparecer em 'Outros dispositivos' e siga estas instruções: (em alguns casos, o dispositivo desconhecido é permitido em dispositivos Jungo ou VR; em qualquer caso, siga as instruções abaixo)
> > 
> > * Clique com o botão direito do mouse e selecione "Atualizar software de driver" - 
> >   Selecionar " Procurar no meu computador por software de driver" - 
> >   Selecionar "Deixe-me selecionar uma lista de drivers de dispositivo no meu computador" - 
> >   Selecionar dispositivos libusb- win32 
> >   -Selecione ANT USB Stick 2 e, em seguida, OK sem aviso prévio e agora O seu treinador USB está instalado como "ANT USB Stick 2", o que funciona, mas talvez não seja um nome muito claro.
> > 
> > Isso deve resolver o problema.
> > E com relação à questão do T1932, eu encontrei uma questão semelhante no meu treinador, verifique se a ponta do sensor de cadência está posicionada corretamente contra o ímã no braço da manivela e próximo ou suficiente. Se o software ANTForitus não detectar os pulsos do sensor de cadência, ele se comportará de maneira irregular, usando uma resistência muito forte, seguido por quase nada como o descrito no seu comentário.
> > Boa sorte e sucesso,
> > Ramon
> 
> No momento, estou no T1932, aquelas interrupções com a cadência instalada, pararam. Obrigado por isso, vejo que, no momento em que inicio o FortiusANT, ele informa a calibração do rolo. no momento tenho velocidade e cadência no painel, mas não tenho energia. quando paro de pedalar e logo depois. Quando volto a pedalar, os ponteiros da cadência e da velocidade da potência estão imóveis, obrigado por esse apoio. @WouterJD @RamonFigueras

T1942 :(  @WouterJD @RamonFigueras  
[image]

desscrambler, 2020-06-11:
T1932 💯  @WouterJD  👍 @RamonFigueras  👍  

[image]

WouterJD, 2020-06-11:
Thanks for sending the logfile.

I expect a tacx to return buffers of 64 or 48 bytes.
Your tacx returns 48 bytes but -every other message- an incomplete message of 24 bytes.

I have updated the reading module, skipping the incorrect messages.
Please [download](https://github.com/WouterJD/FortiusANT/blob/master/WindowsExecutable/FortiusANT.exe) version 3.1.1 and see what happens.

Succes!

WouterJD, 2020-06-11:
>

Google Translate - our friend :-)

The Tacx USB is very sensitive to loose connections; check carefully!
If a connection is loose, it causes unpredictable behaviour - no error messages!

No problem, the check for the buffer length dispappeared in the code-cleanup to release 3.0 so I'm happy it's back in.

desscrambler, 2020-06-12:
Good night @WouterJD
the T1932 here working with the .bat file, with the .exe the graphic doesn't work, running out of speed, cadence and power.

Follow log, FortiusAnt.exe -d127

[FortiusANT.2020-06-12 02-49-49.log](https://github.com/WouterJD/FortiusANT/files/4768553/FortiusANT.2020-06-12.02-49-49.log)

desscrambler, 2020-06-12:
Help me with the T1942. @WouterJD @RamonFigueras

desscrambler, 2020-06-12:
> Help me with the T1942. @WouterJD @RamonFigueras

[FortiusANT.2020-06-12 04-20-51.log](https://github.com/WouterJD/FortiusANT/files/4768739/FortiusANT.2020-06-12.04-20-51.log)

WouterJD, 2020-10-01:
Closed, if needed more support I will hear from you

closed 2020-10-01 by WouterJD