WouterJD/FortiusANT #96: Hi please support with Fortius/i-Magic

WouterJD, 2020-06-13:
Hi

I have now 2 dongles so will try and  rebuild my Fortius/imagic for Zwift 

Which trainer on Zwift do we  pair with /check on their set up once working?

How much tension should be on the resistance roller?

I am near Leeds, Yorkshire England

https://www.strava.com/athletes/304867

_Originally posted by @Langsett in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-643588675_

WouterJD, 2020-06-13:
Also: Oh dear, the FortiusAnt says no trainer detected, and I have swapped leads from brake to head unit, changed head units and still not detected brake how can I check it is getting power

WouterJD, 2020-06-13:
Tension: do you have Fortius or i-Magic? 
Use rundown and should take 7secs

WouterJD, 2020-06-13:
No trainer issue: check whether device-manager recognizes USB device first.
Then -indeed- check all cabling carefully. Loose connector does not give error message but erratic behaviour

Langsett, 2020-06-15:
Hi

I have the iMagic  ,  also  when having problems with TTS  Tacx gave me  Fortius  head unit, for  your programme  does 1932 ( blue) work with Imagic or should I use the green head unit ?

Thanks and best wishes

Martyn

WouterJD, 2020-06-15:
Martyn, to be honest I do not know. All I have is a Fortius with the T1932 headunit.
I have the impression that the headunit is the unit that decide how the trainer works; in the end TTS also worked with this configuration. The USB_ID tells the software how to operate.

The Fortius T1932 executes a calibration automatically; after you give it a pedal-kick the wheel rotates at 20km/h and when the resistance is stable you can continue.

Other systems should use the runoff option.

If you do not want the automatic calibration, start FortiusANT with the -n option.

Suggestion:
- Start FortiusANT.exe and see what happens
- If it does not work well, add the -d127 flag and ppost the logfile
- -n is always an option to skip the calibration, which is not very necessary

Succes and let me know.

PS. If you want to know more on configurations, hardware and interfaces: check [TotalReverse](https://github.com/totalreverse). You can always post an issue with a question to him

Langsett, 2020-06-16:
Hi

My works IT support connected to my lap top and asked if I had a driver for the Tacx

Will that be tacx own driver or  one for FortuiusAnt as I am not technical

Thanks

martyn

WouterJD, 2020-06-16:
Point them to https://github.com/WouterJD/FortiusANT/wiki#windows which will help them forward

WouterJD, 2020-06-19:
@Langsett any progress?

Langsett, 2020-06-23:
Not yet I have been busy with family issues , and will come back to this when those relax

Thanks

Martyn

WouterJD, 2020-06-23:
Cool. it's summer anyway :-) Let's hope to be able to go outside a lot!

WouterJD, 2020-10-01:
Closed, if needed more support I will hear from you

closed 2020-10-01 by WouterJD