WouterJD/FortiusANT #97: Probleme ANT+

regis65, 2020-06-15:
Salut
J ai reçu les clés ANT+ de la marque Anself aujourd'hui quand je lance explorerAnt sa me marque des erreurs (voir photo ci-joint) mais je lance l aplication sa fonctionne mais sa se decalibre peux tu m aidé
Merci
Regis
[image]

WouterJD, 2020-06-15:
quel versione del FortiusANT?Photo de lecran svp
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

regis65, 2020-06-15:
La version c est celle du 18.9 2019.
Sportivement
Regis

WouterJD, 2020-06-16:
En settembre 2019, FortiusANT n'existait pas...
S'il vous plais utilisez version 3.2 parce que c'est la version plus courant

regis65, 2020-06-16:
Merci mais je ne l es trouvé
FortiusAnt 3.2
Sportivement
Regis

WouterJD, 2020-06-16:
c'est au github
https://github.com/WouterJD/FortiusANT/

regis65, 2020-06-17:
Salut j ai téléchargement fortiusant 3.2 mais j ai toujours une erreur(voir photo)
[image]
 et zwift ne reconnait pas l home trainer

WouterJD, 2020-06-17:
Excuse-moi Beaucoup
J'ai trouvé Vette erreur hier, je publierai 3.2.1 leuk plus vite possibile. Je te fais savoir aujourd'hui

regis65, 2020-06-17:
Je te remercie, mais sans urge pas à la minute.
Sportivement
Régis

WouterJD, 2020-06-17:
Merci, mais c'est fait.
Hier j'avais déja solué la problème; maintenant c'etait seulement pousser qq boutons

A plus tard!

regis65, 2020-06-17:
Ok je vais attendre que tu le mettes en ligne.
Sportivement
Regis

regis65, 2020-06-17:
Quand tu l auras mis en ligne tiens moi au courant.
Merci
Regis

WouterJD, 2020-06-17:
Deja fait a 11h ce matin

regis65, 2020-06-18:
Salut j ai essayé ta nouvelle version (3.2.1) sa fonctionne sur Zwift .
Merci
Sportivement
Régis

WouterJD, 2020-06-18:
Top!

closed 2020-06-18 by WouterJD

regis65, 2020-10-15:
Salut,
Je possede le FortiusAnt depuis le mois de juin 2020 je voudrais y a t il
une nouvelle version.
Merci

WouterJD, 2020-10-16:
Regis bonjour. Ils sont des petits modifications, pas des dévélopments nouvelles.
Est-ce-que tu as des idees, desires?

regis65, 2020-10-18:
Salut, c etait juste pour savoir
Et merci encore d avoir fait cette application,sa m embêté d acheter un
autre home trainer.
Sportivement
Régis.Crouzil

regis65, 2020-10-18:
Salut dis moi as tu l intention de voir si on pourra le faire fonctionné en
Bluetooth un jour?
Cordialement
Regis

WouterJD, 2020-10-18:
Voyez #103; en effet non. Si il-y-a qc qui le fsir, je veux vien intégrer. Mais la development coûtera des temps et ne me porterá rien; et les utilisants futur seulement l'avantage de ne pas devoir acheter les deux dongles.
Je crois que la connection au telefone n'est pas une vrai avantage pour une zwifter.

regis65, 2020-10-18:
Daccord merci pour ta réponse.