WouterJD/FortiusANT #98: Compatibility with TTS4?

EvoxRider365, 2020-06-16:
I have Fortius T1932.  I have TTS4 running on Windows 10.  Can FortiusANT and TTS4 run on the same computer?  I would like to use both if possible.  Please respond.  Thanks

EvoxRider365, 2020-06-16:
Or in other words, can TTS4 and FortiusANT coexist?  I would run them separately but I would like to still use TTS4 for as long as I can.

WouterJD, 2020-06-19:
Save as #99 and since more info there, I close this isse

closed 2020-06-19 by WouterJD

EvoxRider365, 2020-06-20:
Sorry I had a busy busy week and only attempted installation last night.  
Re: installation
Windows 10 Home 65 bit
I could see my TACX Ant antenna and the Garmin ANT+ USB2 dongle that I purchased used last weekend.
I created .inf files for each but they would not install. "Error installing Driver - System policy has been modified to reject unsigned drivers."

libusb-win32 device filter installed successfully
I was able to install device filters for both ANT USB antennas.
I was not able to connect to the TACX Fortius however so I went back and installed the VR Interface as well (following the suggestion of installation option 2 for fortius ant)
I was then able to run and calibrate the Fortius Ant 3.2.1 application while connected to T1932

I was able to connect to Strava (I think)
I was more interested in getting Zwift up and running.
I started the installation but it took a long time to download and install the app.

The app can see the trainer but the trainer is not moving the rider on the screen.  When I push buttons on the Tacx head unit I sometimes move a metre or two on the screen but then it stops again.  

Sometimes the Zwift menu popped on and off the screen erratically with a prompt to end my ride to nowhere....

There are no watts showing in the upper right hand corner, no speed, etc.  

Meanwhile the pedalling I did to get started resulted in the motor spinning the wheel forever.

Any suggestions at this point?  Maybe I will try the earlier version of Fortius ant and check my usb cable from the Fortius to my computer.

I am also going to see if TTS4 will still run.

Suggestions welcome...hoping this will work...

regis65, 2020-11-28:
Bonjour,
Depuis que j ai installé FortiusAnt sur un autre pc ,
Il ne reconnait plus l interface de mon home trainer Forius en le branchant en usb,
Pouvez vous m aidé 
Sportivement
 Régis

WouterJD, 2020-11-30:
Bonjour Régis!

Si vous me racconter un peux plus, peut-etre je peux aider - bien sur.

regis65, 2020-11-30:
salut en fait sur mon pc j' ai fortius ant et Tacx Trainer Software pour
mon home trainer Fortius mais malheureusement quand je lance Tacx Trainer
je n est plus l interface qui fonctionne le vr et le moteur.
sportive
ment
regis

WouterJD, 2020-12-01:
Regis, TTS and FortiusANT do not necessarily work together on the same system.

regis65, 2020-12-02:
Ok merci, je pensais que les 2 programmes pouvais fonctionné sur le meme pc.
Tant pis
Sportivement
Régis

regis65, 2021-01-07:
Salut,
Déjà meilleurs voeux 2021 pour toi et ta famille
Dis moi hier j ai casse le cable du capteur de cadence  de mon Fortius et
je cherche un site qui vendrais des pieces détaché de home trainer et je
voudrais savoir si tu en connait un.
Merci
Regis