WouterJD/FortiusANT #99: Can TTS4 and FortiusAnt coexist?

WouterJD, 2020-06-16:
I am near Ottawa Ontario Canada and looking forward to using FortiusAnt soon.  I just picked up some used ANT USB dongles on the weekend.  I just need to know if TTS4 and FortiusAnt can coexist on the same windows 10 computer.  I still want to use TTS4 for as long as I can.  Thanks

_Originally posted by @EvoxRider365 in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-644493974_

WouterJD, 2020-06-16:
As far as I know this is not possible.

When you plug-in the USB-cable one driver is selected.
FortiusANT needs the libusb driver.
TTS needs the tacx driver.

To be honest I did not try to start TTS with libusb / FortiusANT with Tacxdriver; I have learned from the predecessor or FortiusANT ([antifier](https://github.com/john-38787364/antifier)) that it's not possible.

I do not exclude that some creativity would help.
An option could be to install a virtual machine and run FortiusANT in there.

WouterJD, 2020-06-16:
And since we did not introduce ourselves yet, my name is Wouter. ARe you on strava as well?

mattipee, 2020-06-16:
Is it not possible to disable the USB driver and have TTS4 access FortiusAnt over ANT+? I haven't read the full thread, nor tried what I am suggesting, just dropping this quick comment as a thought out loud.

iepuzaur, 2020-06-16:
I am running on Windows10 TTS4 and FortiusANT, at first I also thought I have to uninstall the Jungo driver TTS uses, but then I noticed you can have both drivers installed and there is no issue on my system. 
Of course, you cannot use them concomitantly, if you start TTS4 it will use the Jungo driver, while when you start FortiusANT the libusb drive will be selected.
So as long as you do not  intent to use them at the same moment, the drivers shall not conflict (at least this is my case).

WouterJD, 2020-06-16:
Thanks @mattipee and @iepuzaur for thinking with us.
@evoxrider let us now your experience!

WouterJD, 2020-06-19:
@EvoxRider365 what's your experience?

EvoxRider365, 2020-06-20:
Sorry I had a busy busy week and only attempted installation last night.
Re: installation
Windows 10 Home 64 bit
I could see my TACX Ant antenna and the Garmin ANT+ USB2 dongle that I purchased used last weekend.
I created .inf files for each but they would not install. "Error installing Driver - System policy has been modified to reject unsigned drivers."

libusb-win32 device filter installed successfully
I was able to install device filters for both ANT USB antennas.
I was not able to connect to the TACX Fortius however so I went back and installed the VR Interface as well (following the suggestion of installation option 2 for fortius ant)
I was then able to run and calibrate the Fortius Ant 3.2.1 application while connected to T1932

I was able to connect to Strava (I think)
I was more interested in getting Zwift up and running.
I started the installation but it took a long time to download and install the app.

The app can see the trainer but the trainer is not moving the rider on the screen. When I push buttons on the Tacx head unit I sometimes move a metre or two on the screen but then it stops again.

Sometimes the Zwift menu popped on and off the screen erratically with a prompt to end my ride to nowhere....

There are no watts showing in the upper right hand corner, no speed, etc.

Meanwhile the pedalling I did to get started resulted in the motor spinning the wheel forever.

Any suggestions at this point? Maybe I will try the earlier version of Fortius ant and check my usb cable from the Fortius to my computer.

I am also going to see if TTS4 will still run.

Suggestions welcome...hoping this will work...

EvoxRider365, 2020-06-20:
I just learned that I had to search for my heartrate monitor and the "Controllable" in Zwift
But Zwift is still getting no watts or rpm from fortius ant...

[image]

EvoxRider365, 2020-06-20:
I thought maybe I was connecting to the wrong ant antenna so I unpaired the power source and cadence in zwift and connected them to the Tacx

[image]

EvoxRider365, 2020-06-20:
Although I was able to calibrate when I started fortiusant 3.2.1, I seem to be getting only power and not speed or cadence through the fortius and fortiusant...

[image]

EvoxRider365, 2020-06-20:
I believe this confirms this interpretation
[image]

EvoxRider365, 2020-06-20:
I am going to try TTS4 now to confirm my usb cable is still ok.

EvoxRider365, 2020-06-20:
It took a restart of my computer, but TTS4 still runs

WouterJD, 2020-06-21:
Hi a lot of experimenting!

5759* are FortiusANT devices. 
A program connects to a dongle and then searches for a device on the ANT-network. Perhaps just semantics. You're connecting to the correct devices.

WouterJD, 2020-06-21:
In your case I would start with -n flag to skip calibrating and solve the other errors first.

WouterJD, 2020-06-21:
The short buffer message is worrying; 64 bytes expected of which 48 required (last 16 are zeroes anyway). This indicates a loose connection; I saw this on my configuration and was resolved after checking USBcable and restart.

WouterJD, 2020-06-21:
The short messages are ignored after four retries and since it happens every so many seconds it could work anyway.

WouterJD, 2020-06-21:
Uninstalling Tacx drivers I learned from.antifier; you are the first trying this.

Good luck

EvoxRider365, 2020-06-27:
How do you run commands to debug the app?

I found only limited info on this in the wiki

I see that fortiusant is starting with default parameters but i dont know how to change them

This is what i see briefly before the errors start

Thanks

WouterJD, 2020-06-29:
I'm planning to make a better user manual, since wiki may be difficult to follow.
https://github.com/WouterJD/FortiusANT/wiki#command-line

If you double-click the executable, the default settings are used.

There are two ways:
- create a shortcut to the executables and add the command-line flags
- create a .bat file and execute that file, see https://github.com/WouterJD/FortiusANT/tree/master/StartUp for samples

EvoxRider365, 2020-07-14:
Update as of July 12/2020
A. Previously I had Windows 10 messages that the usblib tailored drivers did not install.  I found this page helpful to install them:
https://www.maketecheasier.com/install-unsigned-drivers-windows10/
When I fully installed the VR and two ANT USB drivers, TTS4 did not work.  I will test that again...but for now I uninstalled usblib from that PC and TTS4 works again.
B. I had a VM on a MacBook Pro running a new copy of Windows 10.  I used that next and installed usblib drivers using the above mentioned advice from the web page cited above.  I installed FortiusAnt and Zwift and I successfully ran Zwift with the Fortius and a heart monitor.  But my MacBook Pro needs repairs so that was only a test run.  My USB cable and the errors on communication were all gone so I suspect the uninstalled windows drivers were the real reason it did not run on System A.
C. Zwift can run on a core2duo mac and I happen to have an extra old mac mini so I would like to try that next.  You mentioned you do not have mac hardware to test.  I have installed brewhub and I have found usblib for mac.  (running mac os 10.11 for now).  I have also downloaded the mac version of zwift.  Where would I find a mac version of the fortiusant?  Let me be your tester for this.

EvoxRider365, 2020-07-17:
So I have installed homebrew but I need to update my mac os as Python 3.8.x won't install.  To do that on core2duo i need to install the dosdude patch and I want to make sure that I won't lose my wifi card functionality as a result.  There were conflicting information on macos 10.11 being current enough for Python 3.8.x to install.  
Which .py files do I need to copy or should I be installing?  Thanks

WouterJD, 2020-07-17:
We'll, all from pythoncode
And if you want to reduce to the minimum (why?) Ecplude explorANT* and test*

EvoxRider365, 2020-07-17:
Thanks. Upgrading MacOS this afternoon and then I will try again this evening.

EvoxRider365, 2020-07-17:
And someone gave me a FREE fortius-so now I have two : )  If it still works ok I will put one in the garage and one in the basement.

EvoxRider365, 2020-07-17:
Gonna need more ant adapters though....  : (

EvoxRider365, 2020-07-18:
Ok, well my mac mini 2009 is now running macos 10.14 so that homebrew will stop complaining I am running an unsupported macos.  I have reinstalled python 3.8.3, libusb, and pyenv.  I have all the python scripts and supporting files in a folder on my desktop.  I have plugged the Fortius, and two ant adapters into my mac mini usb ports.

[image]

When I run python3 fortiusant.py from my desktop folder I get the following error message: 

File "fortiusant.py", line 13, in <module>
import numpy
ModuleNotFoundError: No module named 'numpy'

Am I missing something?
Thanks

WouterJD, 2020-07-18:
Yes, run pyinstall numpy
It appears this module is missing in requirements.txt

I do not have exact syntax at hand

EvoxRider365, 2020-07-18:
Going to try that in a moment.  I also got the same tpe of error "No module named 'usb' when running antdongle.py for example

WouterJD, 2020-07-18:
usb should be covered by pyusb.

EvoxRider365, 2020-07-18:
making some progress...

on a mac the default python is 2.7 
the preferred python installed by homebrew is python 3 (3.8.4 has been installed)
that means when i run the python scripts I need to specify python3 and then the script name or pip3 and then the .txt name.
i was able to find instructions to install numby and I ran the requirements.txt but  I got an error.
I am now being told there is no wx module when i run fortiusant.py
I am going to search for that and try to figure out what that other error was.

WouterJD, 2020-07-18:
ex missing in requirements too?
Please note them all and let me know

EvoxRider365, 2020-07-18:
It seems the error was related to pygame being installed when trying to install pyusb.  
no resolution to that as well.

I have tried different online suggestions to install wx as part of wxpython using brew and I am getting no where

EvoxRider365, 2020-07-18:
Part of the challenge is that installed homebrew before upgrading the macos to 10.14 because homebrew said it was compatible with 10.11 and it is no longer compatible with that macos version.  Now it is a challenge to update homebrew and python and I am having to do it in pieces.

EvoxRider365, 2020-07-18:
I just ran requirements.txt again.  Attached is the start of hundreds of lines of errors all related to pygame installation failure.

[image]

EvoxRider365, 2020-07-18:
pip3 install pygame seems to be working but is still executing...

EvoxRider365, 2020-07-18:
Pygame now installed.  Everything in requirements.txt now seems to be installed.

I am getting the same error from several scripts (fortiusant.py, antdongle.py, fortiusantcommand.py and usbtrainer.py) No module named 'fxload'

I have not found a solution to install whatever to install fxload

Any suggestions?  Thanks

EvoxRider365, 2020-07-18:
I found this:
https://github.com/kolontsov/homebrew-fxload

But I have to manually install an older version of xcode to run the installer successfully or so it seems.
Working on that now...

EvoxRider365, 2020-07-19:
It literally took a days worth of effort to install the 11.3.1 version of xcode in macos 10.14 mojave.  Fxload installed ok but I am still getting the error that ther is no module named fxload.  I am getting a few other errors too.  I am just about ready to give up and buy another copy of windows or a new trainer...any suggestions?

EvoxRider365, 2020-07-19:
Traceback (most recent call last):
File "fortiusant.py", line 25, in <module>
import antDongle          as ant
File "/Users/User/Desktop/fortiusant/FortiusAntCommand.py", line 180, in <module>
import usbTrainer
File "/Users/User/Desktop/fortiusant/usbTrainer.py",  line 104, in <module>
import fxload
ModuleNotFoundError: No module named 'fxload'

WouterJD, 2020-07-20:
fxload is a module in pythoncode.

EvoxRider365, 2020-07-20:
I figured that out but it is not installed by default in homebrew and the installation that I found for mac does not seem to be working.

https://github.com/kolontsov/homebrew-fxload

I seem to be at a deadend...but other mac users apparently got it working...maybe I am doing something else wrong.

EvoxRider365, 2020-07-23:
Another thought...Any reason why fortiusant would not work with the new TACX app for Windows?

EvoxRider365, 2020-07-29:
Well just for fun or out of desperation I experimented a little...
Windows 10 running in Parallels on a MacBook Pro running Mojave 10.14

I changed my USB cord to the Fortius.
I installed Wahoo RPM dual ant and bluetooth Speed and Cadence sensors and because I had it lying around a Runtastic bluetooth dual Speed/Cadence sensor (BLE in one of the pictures I think).  The runtastic only seems to be picking up speed not cadence though because the fortius sensor prevents me from putting it close enough to the pedal arm.

I added a Startech Bluetooth 4.0 as it seemed that the new Tacx desktop app was not recognizing ANT.
I then connected wahoo heartrate monitor (dual ant and bluetooth), and wahoo rpm speed and cadence sensors (dual ant and bluetooth).  My hope was that somehow the Tacx would pick up on the additional sensors.
I fired up Fortius ant and there seemed to be no errors although I did get a calibration aborted error the first time.

I then tried the new Tacx App.
First time the fortiusant seemed to be recognized as a Neo 2T!
but nothing worked...no speed, no cadence, only heart rate
(I wondered what was ant speed cadence sensor 57595 was)

Then I rebooted and the fortiusant was then recognized as an FE-C trainer 57591
For a brief moment I saw speed and cadence and heart, then the cadence fell away.  But I still saw speed and heart rate.

When I tried to play a video I either got a blank screen or the Tacx app crashed (I suspect my wifi or network was not fast enough to stream).
I did get as far as the warm up though but then no speed, cadence or heart was displayed.

Tacx only allows a 7 day trial so my experiments are running out of time.  
Is there a chance this might work with Tacx?
What do you think?

I also tried Rouvy AR but they don't support ANT yet or speed sensors so without a power source or speed sensor or controllable it won't work.

I am waiting for a Trainer Road licence to give that a try.  So far the only success I had was with Zwift under Windows 10 where no TTS 4 had been installed.
[image]
[image]
[image]
[image]
[image]
[image]

WouterJD, 2020-07-30:
Well @EvoxRider365 there is a lot of communication here.
Since it's summer time and holidays over here I have left FortiusANT for a moment and do some movements outside.
Nevertheless, I want to respond.

You are doing a lot of things at the same time
- Check that TTS and FortiusANT run alternatingly on the same system (both using USB)
- Implement FortiusANT on a Windows/MAC environment
- Let TTS communicate with FortiusANT

Note that:
- Tacx deliberately tries (tried) to lock-in using proprietary protocols, be it USB or ANT
- FortiusANT creates a series of (master) devices, see antDongle.py:
```
DeviceNumber_FE     = 57591    #       These are the device-numbers FortiusANT uses and
DeviceNumber_HRM    = 57592    #       slaves (TrainerRoad, Zwift, ExplorANT) will find.
DeviceNumber_VTX    = 57593    #
DeviceNumber_VHU    = 57594    #
DeviceNumber_SCS    = 57595    #
DeviceNumber_PWR    = 57596    #
```
where not all devices are always active.
Rouvy/Trainer Road have enough using the FE-C master device.
If you have an old HRM linked to your FortiusANT (as it used to be) the HRM device transmits that heartrate as ANT+; if you use an ANT+ HRM Rouvy/TrainerRoad will detect that HRM themselves and you will use the command line option to listen to that device by FortiusANT.
SpeedAndCadence Sensor and PowerSensor are transmitted so you can use an Edge to display what your Fortius does.
VTX and VHU are only when an iVortex is used.

You WOULD expect that Tacx would recognize the FE-C device and nicely pair, especially because the device is "faking" Tacx/NEO. But it does not. I already found out that that is the case and did not spend time for further investigation. Reason: having a Rouvy subscription, why pay more for the Tacx software where they try to lock me out?

What I would do in your case:
1- Focus what you want to achieve
2- Make software work "as designed" first: FortiusANT -> USB -> Fortius
3- Do experiments whether you get FortiusANT and TTS work on the same system, each using the USB_interface as designed but with other drives
4- And as soon as that all works, try to pair Tacx software to a non-tacx device (which FortiusANT is) figuring out what specials Tacx built into the interface to lock others out.

Good luck!

EvoxRider365, 2020-08-04:
I have followed your suggestions.  I am about to test fortiusant with Trainer Road.  RouvyAR won't work yet and Zwift is too virtual and race like for me.  I am more of a touring rider.  I have a one year subscription to TACX app so I was hoping that would work for me.  I know that TTS and FortiusAnt did not coexist well on my windows 10 computer.

I have learned from Tacx support in recent days that non-Tacx trainers are limited to bluetooth (not ant) and only work with their app on computers (windows or Mac)

**There is a new product I found that may extend the utility of fortius ant from just ant to bluetooth.**

https://shop.4iiii.com/collections/heart-rate-monitors

A Canadian invention...a combo heart rate monitor/smart phone app that can convert ant to bluetooth in real time.  Maybe that might make the Tacx app work with fortius ant.  I will order one soon.  (availability seems to be limited)

Take a look at the 4iiii trainer as well (contactless)
https://shop.4iiii.com/collections/smart-trainer

dc rainmaker has a review here:
https://www.dcrainmaker.com/2019/12/4iiii-fliiiight-smart-trainer-in-depth-review.html

I have in recent days confirmed with other Canadian users that both Kinetic and 4iiii trainers work with the Tacx app on a Mac using bluetooth. (not on smartphones, not using ant)

WouterJD, 2020-08-04:
Regarding Rouvy, check #52, which would appeal as a touring rider.

I will check the links you have posted and I'm curious about hearing your experiences.

WouterJD, 2020-08-04:
To be honest I'm confused.

FortiusANT connects to a USB-connected-tacx-trainer (or i-Vortex) at one side and communicates with TR/Zwift/Rouvy at the other side thrugh ANT.

4iiii is another smasrt trainer, replacing the "old" tacx trainer.
Of course a good idea, but I miss the relation to FortiusANT, please explain.

Ciao!

EvoxRider365, 2020-08-05:
I was trying to explain that the Tacx App (on Windows) sees the FortiusAnt as a FE-C trainer in ANT but will not connect to it.  Tacx support has explained to me that only bluetooth connections are possible on computers (windows or Mac). And they also said their smartphone app (iOS or android) will only work with their trainers-no others.  So the only option to connect is connect to the Tacx app via bluetooth with a computer.

4iiii has a product that may bridge the ANT presentation (heart rate monitor/app) of the FortiusAnt to bluetooth through a smart phone, which just might work with a computer running the Tacx app.  Because I have a year subscription with Tacx to use up, I am eager to find a way to try it.  And because Zwift is race oriented and virtual, Rouvy AR does not seem to work with FortiusAnt, only Trainer Road is an alternative to Tacx.  So I will give it a month trial with fortiusAnt but even better if I can find a way to use the fortius with the new tax app.  It is more familiar to me because of TTS 4.

I also have three bikes.  26" mountain, 24" recumbent, 20" catrike.  If I have to buy a new trainer I would prefer to get one that can work with all three sizes.  Only Kinetic with a small wheel adapter seems compatible with the Tacx app and will support all three wheel sizes.

The 4iiii trainer officially supports 26-27" bikes, but I am asking if it can support smaller wheels.  A contactless design is attractive seems like it might be configurable for smaller  wheels.  If not, it could give me a way to at least ride the mountain bike and it is compatible with the Tacx App through bluetooth on a Mac (likely pc as well).  I just thought it was a cool trainer.

I know wahoo also works with Tacx app but it only supports 26" and 700 wheels.

CycleOps/Saris may also be compatible but I have not found anyone to confirm that and the small wheel adapter only goes as far as 24" I believe.

And so my options seem to be:
1. Use FortiusAnt with TTS4 (limited library)
2. Use FortiusAnt with Trainer Road (ant)
3. Use FortiusAnt with Tacx New App (using 4iiii HR monitor as bridge to bluetooth connecting to a computer.
4. Get a new trainer (Kinetic or 4iiii)

EvoxRider365, 2020-08-05:
I just looked at your #52 ticket and I will try to download Rouvy Workouts tomorrow and try it too.  Thanks

WouterJD, 2020-08-05:
> ... the Tacx App (on Windows) sees the FortiusAnt as a FE-C trainer in ANT ...  only bluetooth connections are possible ...

So it looks like that 
- ANT is only used for non-FE-C devices (like HRM) and that ANT/FE-C only displays the device for information purpose as the helpdesk suggests
- that Tacx only pairs with devices they made themselves and have an undocumented interface-part to achieve that lock-out
- it's faulty software (why implement ANT and not use it)

If the first option is the case then the bridge may work, if the second is the case (not unlikely, since Tacx created a special ANT-interface for i-Vortex to hide it for other software) then bluetooth may fail as well.

A good friend of mine suggested that Bluetooth is getting prio for software-to-FE-C interfaces over ANT and that ANT remains the on-bike network. Not unlikely, since computers (celphones) are more bluetooth than ANT oriented and it removes the dongle issue (doubtlessly introducing other issues). This direction is supported by the fact that Rouvy evidently has no ANT implementation yet.

So, an option would be to implement Bluetooth in FortiusANT.
If there is anybody following this thread: who has some starting points since I did not do anything in that area yet.

EvoxRider365, 2020-08-05:
The direction the trainer industry seems to have taken is bluetooth over ant.  Maybe because of the popularity of tablets, smartphones and Apple TV 4 & 4K.  You can connect bluetooth to Apple TV and ride Zwift now, maybe Trainer Road (haven't tried it yet) and Rouvy AR when they support a speed sensor.  Tacx points out that you can use their new app to Airplay the video to an Apple TV. 

While Mac's have had Bluetooth 4 for years, most pc's do not have on board bluetooth, hence the need to still support ant.  Ant is also the standard for other fitness devices (fe-c) as I understand.  There are many bluetooth adapters for PC's but I have seen some discussion that only one will work with trainers...some users have pointed to this:

https://www.cycleops.com/product/bluegiga-usb-dongle

That is not the bluetooth adapter I have at present.  I have a StarTech bluetooth adapter.  It seems to connect to wahoo speed and cadence sensors but the data was not passing reliably to the Tacx app.  I have yet to try it with Trainer Road or Rouvy Classic.  Maybe this evening.

Another option I failed to list in my previous post was getting a new Tacx on-wheel trainer.  I have been advised by Tacx that the Genius and Vortex have been discontinued.  I expect they will continue to be supported by the Tacx app for a while though.  Tacx flow is now the only on-wheel trainer it would seem.

Tacx does emphasize bluetooth as a requirement even for their own trainers with the new app so I am not sure how much they use ant at this point.
https://tacx.com/product/tacx-software-features/

Frenske8, 2020-08-05:
> I was trying to explain that the Tacx App (on Windows) sees the FortiusAnt as a FE-C trainer in ANT but will not connect to it. Tacx support has explained to me that only bluetooth connections are [...]

Hi, I have been using FortiusAnt for 2 months now in combination with the Tacx Desktop App (Windows 10). It works fine! The Tacx Desktop App connects via ANT with ANT-ID 57591 (identified as Tacx Neo2T !). Incidentally, I also own the 4iii Viiiiva HRM with ANT to BT bridge, but I expect the bridge to work in one direction only, so not from BT to ANT. So using BT via the 4iii HRM is unlikely to provide a solution, I failed. However, this is not necessary, because FortiusAnt can work directly via ANT with the Tacx Desktop App.
By the way: I’m using FortiusAnt and Tacx App on two different laptops, because I sometimes want to use Tacx TTS4 (installed on the same laptop as Tacx Desktop App). I failed to get FortiusAnt and TTS4 to work on the same laptop.

[image]

EvoxRider365, 2020-08-05:
Thanks for the reply.  That is encouraging.  With my fortiusAnt setup I got a brief connection as a Neo2T 57591 but it later became Fe-c trainer 57591.  In both cases, I got no speed or cadence and no warmup before a video.  The only connection available was ANT not bluetooth.

I wonder why it is not working for me...

Given the comments by Tacx support I was hoping the 4iiii heart rate monitor would convert from ANT to Bluetooth, enabling fortiusAnt to communicate with the Tacx app through bluetooth (not the other way around).  

You are indicating that you can use the Tacx app with fortiusAnt via Ant.  May I ask which ant USB adapters you are using?  Maybe that is playing a role.

With the installation of rouvy ar and trainer road I think I also have their usb drivers installed as part of their app setups.  Maybe there is a conflict there as well.

Can you tell me any more about your setup?  Thanks

EvoxRider365, 2020-08-06:
I had more success tonight.  I connected and tried Trainer Road, Rouvy classic, and even the Tacx app!  I think I have some calibration issues though as the speed did not seem realistic in Rouvy or Tacx.  In Tacx, on the flat my wheel was racing at 20KMH.  Have to figure out what to do, advice on calibration in fortiusAnt and in the apps appreciated.  

the screenshot that Frenske8 sent was very helpful.  It confirmed he was using the Neo2T device, as ANT, and power, speed and cadence were all selected and connected.

My mistake before seems to be that I also connected the power sensor and speed and cadence sensor at the same time as the Neo2T, then chose the power, speed and cadence from those devices instead of the Neo2T.

Thanks for all the help and encouragement.  I may finally delete TTS from my windows computer and may even run all of this in Mac instead of windows if I can get python on Mac to run the fortiusAnt scripts.

I liked what I saw in Rouvy and I felt at home again in Tacx.  I hope they get Rouvy AR working soon.  I will reach out for rides when I can keep up with you in the future...

WouterJD, 2020-08-06:
> Hi, I have been using FortiusAnt for 2 months now in combination with the Tacx Desktop App (Windows 10). It works fine! 

@Frenske thanks for updating us on your experiences!

WouterJD, 2020-08-06:
> Thanks for the reply. That is encouraging. With my fortiusAnt setup I got a brief connection as a Neo2T 57591 but it later became Fe-c trainer 57591. 

Note that, FortiusANT is NOT a Tacx NEO 2T, I use Manufacturer and ModelNumber of the Tacx Neo but perhaps Tacx software detects it's incorrect and then puts the generic name FE-C.
See antDongle.py for comment and constants.

WouterJD, 2020-08-06:
> With the installation of rouvy ar and trainer road I think I also have their usb drivers installed as part of their app setups. Maybe there is a conflict there as well.

Note that Rouvy and Trainer Road communicate through ANT and expect an ANT Dongle to be present. No special device drivers are needed for that and I do not expect them to install device drivers.

The USB device driver issue is the connection to the legacy Tacx Fortius, which is unknown to Rouvy and TR.

WouterJD, 2020-08-06:
> I failed to get FortiusAnt and TTS4 to work on the same laptop.

Two questions,, @Frenske8 
- so TTS4 and "Tacx Desktop App (Windows 10)"  are two different applications (just to avoid confusion)
- did you get TTS4 work with FortiusANT on two laptops?

Frenske8, 2020-08-06:
Yes, indeed, TTS4 and Tacx Desktop App are 2 different applications. TTS 4 is outdated software, but similar to the Tacx Desktop App. A few years ago, Tacx replaced TTS4 with the Desktop App and they changed the revenue model: with the Desktop App they introduced the subscription model (monthly / annual fee). TTS4 uses real life videos that had to be purchased separately. FortiusAnt also works with TTS4 (with 2 laptops), but that is not really necessary, because TTS 4 also works via the USB connection of the Fortius trainer (with the Jungo USB driver that is installed by TTS4). It is tricky to uninstall the jungo driver (to try to run TTS4 and FortiusAnt on 1 laptop), because Tacx has ended the support of TTS4 and reinstallation of TTS4 is therefore no longer easily possible (becauce Tacx servers for authentication are out of operation).

[11 more replies]

closed 2020-10-01 by WouterJD