WouterJD/FortiusANT #101: Genius T2021 Brake Support

WouterJD, 2020-08-01:
Hi Wouter, from Melbourne, Australia 
- I have a Genius T2021 Brake (Non-smart TACX proprietary ANT+), and an original Tacx 2018 Dongle. I am currently running TTS 4.22.5.
I am interested in liberating my trainer a little to experiment with some of the newer software around. Surely if the Vortex can work, then the Genius must be possible too at some point. I would be happy to try and test any ideas you have. 
Interestingly, I decided to jump in and uninstall the "Jungo" TACX USB drivers, and install the "libusb-win32-1.2.6" drivers instead. TTS4 still works for me! (Windows 10)

FortiusANT is not registering the trainer yet though.
Any tips to try and get it working?

Thanks

_Originally posted by @Sigfreid74 in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-667527513_

WouterJD, 2020-08-01:
To start with, please send some photos of the essentials, like headunit and connectivity.

I have the impression you have a USB-connected trainer...

Sigfreid74, 2020-08-02:
Thanks for the quick reply Wouter, I'll try and help out where I can - 

The "T2020 i-Genius" brake is one of the pre FE-C ANT connected trainers using the TACX proprietary protocols - (like the i-Vortex and the Bushido). No direct USB cable connection like the Fortius. I am using the TACX - supplied T2018 USB ANT dongle.

There is a wireless ANT headunit which is purely optional for use in TTS4, it is only used to control the cursor on the screen while you are on the bike. 

There is some information about the Genius ANT interface here:
https://github.com/fluxoid-org/CyclismoProject/wiki/Tacx-Genius
specifically: Device type: 0xd3 (sometimes referred to as "0×53")

I have installed the libusb-win32 drivers and TTS4 can still communicate with the trainer and other ANT devices (like the speed/cadence sensor and HRM). I uninstalled the Jungo USB ANT drivers that came from TACX.

I have installed Python, upgraded PIP and then installed the requirements.txt dependencies. Then I ran ExplorANT.bat and it was not detecting the Genius Brake, but it was detecting the HRM and Garmin Speed/Cadence sensors. I'll get some photos later on for reference.

WouterJD, 2020-08-02:
My first thought: you do not need libusb at all.
Try the -t i-Vortex flag which causes FortiusANT not to use USB but pair with i-Vortex.

If the pre FE-C devices use the same protocol it should work and in that case read for 'i-Vortex' "Tacx proprietary pre FE-C device".

After your confirmation I will come up with some better name than that.

WouterJD, 2020-08-02:
PS I see that -t is not yet wiki-documented😝

Sigfreid74, 2020-08-05:
Hi Wouter - sorry for the delay - I thought I replied yesterday but something must have gone amok...
I tried the -t I-Vortex tag for ExplorANT and FortiusANT and no response so far. The Ant stick is working as ExplorANT picking up the HRM and Garmin speed/Cadence Sensors.
I was wondering if you were using code from the ANT.h file from GoldenCheetah to find the i-Vortex.? it specifies the Ant TYPE as "0x3d" or "61", the frequency is  0x42 (66)  and the period is 0x2000 (8192)
According to the TotalReverse analysis of the ANT protocol for Genius - the ANT Device Type is "0x53" or "83", the frequency is 0x3c (60) and the Period is 4096 (0x1000). Can we try changing these values in your code?

WouterJD, 2020-08-05:
Sure, use the .py and .try😊
The numbers in the code have been tested for i-vortex and work.

Sigfreid74, 2020-08-06:
My first time doing this - I created a fork and started to change ANTDongle.py. That should effect ExplorANT? I will check the results tonight.

WouterJD, 2020-08-06:
Succes!
I'm curious whether you will get things work; if not we can ask TotalReverse for assistence!

Regarding constants; I tend to define things once - so it affects ExplorANT as well.

Sigfreid74, 2020-08-06:
No luck so far. I tried changing the lines in AntDongle.py that referred to DeviceType, Frequency and Period to the Genius values - no detection of the genius. Still picking up the Speed/Cadence meter though. I think we will need a higher power to help out. TotalReverse is seem to be needed on the case.  

I tried to figure out how to get Wireshark working with Ant using this 
https://github.com/jovial/wireshark-ant and also using a program called USBPCAP with wireshark
but I don't really know what I am doing: I will need to try some more with this. Thee purpose was to try and capture some of the traffic the genius produces - I think this is how the reverse engineering experts do it?

WouterJD, 2020-08-06:
Do you have a non-Tacx application that pairs with the Genius?

Sigfreid74, 2020-08-06:
No only TTS4

Sigfreid74, 2020-08-10:
I have just registered with ANT to log into the website so I can download ANTWARE . Don't know if it will help, but I might as well try.

WouterJD, 2020-08-10:
I do not know. Mind that Tacx has deliberately scrambled the interface, so it's a matter of searching.

WouterJD, 2020-08-10:
> According to the TotalReverse analysis of the ANT protocol for Genius - the ANT Device Type is "0x53" or "83", the frequency is 0x3c (60) and the Period is 4096 (0x1000). Can we try changing these values in your code?

Did you try to implement this in the .py code?
What does "Still picking up the Speed/Cadence meter though." mean?

Perhaps a good idea to send the -d127 tracefile, I will then assist to interpret

Sigfreid74, 2020-08-12:
> Did you try to implement this in the .py code?
I did try to change the values in AntDongle.py that referred to DeviceType, Frequency and Period to the Genius values
So far - no luck detecting the Genius with ExplorANT

> What does "Still picking up the Speed/Cadence meter though." mean?

ExplorANT finds the Garmin Speed/Cadence Sensor on my bike.

I will use the -d127 option and send the file through.

Sigfreid74, 2020-08-12:
https://docs.google.com/document/d/1b9eZ14EIToSmQjW7QAOM2u5T3qWBSqYA1zt9hcuhGyM/edit?usp=sharing

WouterJD, 2020-08-13:
The logfile appears to set the correct parameters
```
11:04:07,594: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
11:04:07,594: SlaveVTX_ChannelConfig()
```
**according to** the the quoted values from TotalReverse as you mention above (PS: what is the exact source?)

Searching for "ch=4" gives no activity at all.

You work the same way as I did to get the i-Vortex active; set the parameters and see whether I received some data.
Only when the correct figures were all set I received response.
I have no sniffers and/or network systems to see what is "in the air".

Try to drop a question to TotalReverse on this subject, perhaps (s)he can assist?

I'll be offline next 10 days.

Sigfreid74, 2020-08-13:
Thanks Wouter, will do.

mikeymclellan, 2020-09-28:
Just chiming in on this as I've just bought one of these second hand and would like to get it working with Zwift. 

I think I've changed the correct parameters for the Genius: https://github.com/mikeymclellan/FortiusANT/commit/b783dfc8153a94b827157e9e5c42c0b31f3a4eac

And this is the debug output from `python3 ./FortiusAnt.py -t i-Vortex -d127`

```% python3 ./FortiusAnt.py -t i-Vortex -d127
23:41:26,693: FortiusANT started
23:41:26,693: -d 127 (0b1111111)
23:41:26,693: -H None
23:41:26,693: -p 1.0
23:41:26,693: -t i-Vortex
23:41:26,693: ------------------
23:41:26,717: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
23:41:27,248: Dongle - Using Dynastream Innovations dongle
23:41:27,248: Tacx   - Pair with Tacx i-Vortex and Headunit
23:41:27,248: AntHRM - Heartrate expected from Tacx Trainer
23:41:28,356: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
23:41:28,506: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
23:41:28,657: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
23:41:28,839: FortiusANT receives data from an ANT+ Tacx i-Vortex Headunit (VHU Controller), any device
23:41:29,021: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
23:41:29,173: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
23:41:29,344: Ctrl-C to exit
23:41:29,344: Tacx   - Pair with Tacx i-Vortex and Headunit
23:41:29,345: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:30,359: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:31,371: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:32,380: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:33,394: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:34,405: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:35,419: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:36,429: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:37,442: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:38,455: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:39,463: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:40,471: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:41,481: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:42,493: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:43,506: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:44,516: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:41:45,530: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
^C23:41:45,605: Stopped
23:41:46,106: FortiusAnt ended
```

I've attached the logfile [FortiusANT.2020-09-28 23-41-26.log](https://github.com/WouterJD/FortiusANT/files/5295574/FortiusANT.2020-09-28.23-41-26.log)

Is what I'm attempting to do correct? I'm happy to get more involved and read through the code more, just thought I'd check I'm on the right path.

mikeymclellan, 2020-09-29:
There's a little more explanation of the different device ids in the [TotalReverse Wiki](https://github.com/totalreverse/ttyT1941/wiki#Protocol-of-older-wireless-ant-tacx-trainers), it relates to whether you include the pairing bit or not. I'm not really sure what this means, but either way I don't seem to have any success.

mikeymclellan, 2020-09-29:
BTW here's a pcap file of the USB data when starting and connecting to the brake using TTS4 captured with Wireshark
[tacx1.pcapng.zip](https://github.com/WouterJD/FortiusANT/files/5296214/tacx1.pcapng.zip)
.

mikeymclellan, 2020-09-29:
Somewhere along the way I managed to get it to output _something_:

```% python3 ./FortiusAnt.py -t i-Vortex -d127
04:11:20,941: FortiusANT started
04:11:20,941: -d 127 (0b1111111)
04:11:20,941: -H None
04:11:20,941: -p 1.0
04:11:20,942: -t i-Vortex
04:11:20,942: ------------------
04:11:20,958: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product=ANT USB-m Stick, vendor= 0xfcf, product=0x1009(4105)
04:11:21,488: Dongle - Using Dynastream Innovations dongle
04:11:21,488: Tacx   - Pair with Tacx i-Vortex and Headunit
04:11:21,488: AntHRM - Heartrate expected from Tacx Trainer
04:11:22,594: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
04:11:22,743: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
04:11:22,893: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
04:11:23,082: FortiusANT receives data from an ANT+ Tacx i-Vortex Headunit (VHU Controller), any device
04:11:23,284: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
04:11:23,440: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
04:11:23,607: Ctrl-C to exit
04:11:23,607: Tacx   - Pair with Tacx i-Vortex and Headunit
04:11:23,607: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
04:11:24,614: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
04:11:25,630: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
04:11:26,639: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
04:11:27,647: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:28,660: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:29,664: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:30,672: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:31,678: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:32,684: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:33,698: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:34,708: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:35,713: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:36,721: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:37,729: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:38,734: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:39,743: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:40,757: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:41,769: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:42,773: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:43,780: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:44,788: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:45,793: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:46,796: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:47,809: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:48,820: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:49,832: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:50,842: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:51,853: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:52,873: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:53,889: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:54,906: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:55,916: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:56,919: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:57,931: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:58,944: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:11:59,953: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:00,963: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:01,973: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:02,985: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:03,991: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:04,997: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
04:12:06,000: Target=100W Speed=39.8kmh hr=  0 Current=229W Cad=163 r= 100 T= 15
^C04:12:06,418: Stopped
04:12:06,923: FortiusAnt ended```

[FortiusANT.2020-09-29 04-11-20.log](https://github.com/WouterJD/FortiusANT/files/5296419/FortiusANT.2020-09-29.04-11-20.log)

Sigfreid74, 2020-09-29:
Wow you got a lot further than me Mikey, I tried your changes to my files
and Genius - I got the same output you did the first time round -
nothing..... What were you doing when you got that output speed etc...?
Thanks

Matthew

WouterJD, 2020-09-29:
Hey guys. Sorry for not digging into the issue earlier. @mikeymclellan you seem to be on the right track.
Since I do not have the T2021 I cannot assist very much.
As soon as you have questions, please drop a note.

WouterJD, 2020-10-06:
@mikeymclellan @Sigfreid74 How are things proceeding?
If you have a working model, just let me know and I will integrate into FortiusANT

WouterJD, 2020-10-06:
Refer to next response from TotalReverse, perhaps it gives additional information:
https://github.com/totalreverse/ttyT1941/issues/15#issuecomment-703829537

m472, 2020-10-20:
I am in the same situation - just bought a Tacx Genius T2021 last week and want to connect it to my PC to ride on zwift. I don't have TTS4 available and am trying to connect it using FortiusAnt and my T2028 antenna.

Tried the fork by @mikeymclellan but did not have any luck so far.

WouterJD, 2020-10-21:
@m472 thanks for using FortiusANT, please introduce yourself; I like to know where FortiusANT landed see #14 

@m472 @mikeymclellan @Sigfreid74 I have added a short section in the manual (see [wiki](https://github.com/WouterJD/FortiusANT/wiki)) and hope that is helpful.

Please keep me posted on progress and/or developments; when I can integrate your work it is helpful for others and -of course- I am curious whether the Genius can be connected as well.

m472, 2020-10-23:
I have now had the chance to do some more testing with my Tacx Genius. 

In a first step I have printed all received ANT+ messages to the console. This way I was able to see some communication between the Genius and my computer. Therefor I can confirm the device type 0x53, frequency 50 (2460 MHz) and Period 4096 are correct.

I then checked the different data page numbers that are transmitted. It seems the Genius brake sends three data pages on a regular base: 1, 81, 94 and 221

I don't know anything about ANT+ data pages but it seems the information in pages 1, 81 and 94 is not dependent on weather I turn the cranks or not. The information in data page 221 however definitely changes when the cranks are turned.

The first byte in data page 221 seems to be some kind of sub-index. It takes the values {1, 2, 3}. I then logged all the information for data page 221 during a sprint and looked at the values by adding a print statement in the DecomposeMessage function. 

I was able to identify some values as high and low bytes of an integer. Others are always zero (maybe my power output is just too low...). So now I have multiple values that seem to make some sense but I have no idea which signal is which. I hope someone is a bit more familiar with power and speed curves on an indoor trainer and can help me interpret thes curves.

The values val_1 to val_4 are from data page 221 subindex 1,
the value val_5 is from data page 221 subindex 2
and the values val_6 and val_7 are from data page 221 subindex 3.

[image]

[image]

[image]

Any help is appreciated, I will share my code if anyone is interested in it.

WouterJD, 2020-10-23:
Hi Mathias great work.
Quite busy right now but I will search a timeslot to look what you did.
I will get back asap

WouterJD, 2020-10-23:
`In a first step I have printed all received ANT+ messages to the console. This way I was able to see some communication between the Genius and my computer. Therefor I can confirm the device type 0x53, frequency 50 (2460 MHz) and Period 4096 are correct.`

Please describe for me how you did this. The possibility is new to me🤔

m472, 2020-10-23:
I just added 
```python3
if Channel == 4: 
    print(d)
```
in the `DecomposeMessage` function. That way I was able to see that messages arrive only if the brake is switched on and the correct parameters for the communication are set.

I later discovered that the `HandleANTmessage` function in the `clsTacxAntVortexTrainer` would be a better place to add such a print statement.

m472, 2020-10-23:
What I found out about the values:

`val_1` is the velocity in km/h * 10
`val_2` is some kind of power
`val_3` is (estimated) cadence in rpm
`val_4` I have no idea what this could be
`val_5` is the distance (probably since startup) in m (probably 32 bits)
`val_6` I have no idea
`val_7` some kind of power or something (kind of) proportional to power

I have hacked the decoding of data page 221 subindex 1 into FortiusANT so that at least the console output looks good now https://github.com/m472/FortiusANT/commit/c2abcd4173e2913ff8de199ce3f5826a72ec8332

some sample output: 
```
23:01:36,361: Target=100W Speed= 0.0kmh hr=  0 Current=  1W Cad=  0 r= 100 T= 15
23:01:37,363: Target=100W Speed= 0.0kmh hr=  0 Current=  1W Cad=  0 r= 100 T= 15
23:01:38,364: Target=100W Speed=13.4kmh hr=  0 Current=  5W Cad=  0 r= 100 T= 15
23:01:39,366: Target=100W Speed=16.2kmh hr=  0 Current=  8W Cad=  0 r= 100 T= 15
23:01:40,367: Target=100W Speed=23.3kmh hr=  0 Current= 20W Cad=  0 r= 100 T= 15
23:01:41,369: Target=100W Speed=27.1kmh hr=  0 Current= 35W Cad=  0 r= 100 T= 15
23:01:42,371: Target=100W Speed=27.0kmh hr=  0 Current= 38W Cad=  0 r= 100 T= 15
23:01:43,373: Target=100W Speed=26.8kmh hr=  0 Current= 41W Cad= 27 r= 100 T= 15
23:01:44,375: Target=100W Speed=26.8kmh hr=  0 Current= 41W Cad= 27 r= 100 T= 15
23:01:45,376: Target=100W Speed=33.2kmh hr=  0 Current= 61W Cad= 38 r= 100 T= 15
23:01:46,380: Target=100W Speed=32.8kmh hr=  0 Current= 58W Cad= 29 r= 100 T= 15
23:01:47,630: Target=100W Speed=32.4kmh hr=  0 Current= 58W Cad= 37 r= 100 T= 15
23:01:48,632: Target=100W Speed=32.2kmh hr=  0 Current= 60W Cad= 44 r= 100 T= 15
23:01:49,634: Target=100W Speed=31.8kmh hr=  0 Current= 64W Cad= 48 r= 100 T= 15
23:01:50,636: Target=100W Speed=31.9kmh hr=  0 Current= 71W Cad= 52 r= 100 T= 15
23:01:51,638: Target=100W Speed=31.6kmh hr=  0 Current= 72W Cad= 53 r= 100 T= 15
23:01:52,640: Target=100W Speed=31.4kmh hr=  0 Current= 74W Cad= 53 r= 100 T= 15
23:01:53,642: Target=100W Speed=31.2kmh hr=  0 Current= 77W Cad= 54 r= 100 T= 15
23:01:54,643: Target=100W Speed=31.0kmh hr=  0 Current= 78W Cad= 53 r= 100 T= 15
23:01:55,645: Target=100W Speed=30.7kmh hr=  0 Current= 75W Cad= 53 r= 100 T= 15
23:01:56,647: Target=100W Speed=30.5kmh hr=  0 Current= 74W Cad= 53 r= 100 T= 15
23:01:57,649: Target=100W Speed=30.3kmh hr=  0 Current= 76W Cad= 52 r= 100 T= 15
23:01:58,651: Target=100W Speed=30.1kmh hr=  0 Current= 76W Cad= 52 r= 100 T= 15
23:01:59,652: Target=100W Speed=29.9kmh hr=  0 Current= 75W Cad= 52 r= 100 T= 15
23:02:00,654: Target=100W Speed=29.7kmh hr=  0 Current= 74W Cad= 51 r= 100 T= 15
23:02:01,656: Target=100W Speed=29.4kmh hr=  0 Current= 71W Cad= 51 r= 100 T= 15
23:02:02,658: Target=100W Speed=29.4kmh hr=  0 Current= 73W Cad= 51 r= 100 T= 15
23:02:03,660: Target=100W Speed=29.3kmh hr=  0 Current= 75W Cad= 50 r= 100 T= 15
23:02:04,662: Target=100W Speed=29.1kmh hr=  0 Current= 74W Cad= 50 r= 100 T= 15
23:02:05,664: Target=100W Speed=28.9kmh hr=  0 Current= 70W Cad= 50 r= 100 T= 15
23:02:06,666: Target=100W Speed=28.6kmh hr=  0 Current= 66W Cad= 49 r= 100 T= 15
23:02:07,668: Target=100W Speed=28.4kmh hr=  0 Current= 65W Cad= 48 r= 100 T= 15
23:02:08,670: Target=100W Speed=28.6kmh hr=  0 Current= 69W Cad= 49 r= 100 T= 15
23:02:09,672: Target=100W Speed=32.3kmh hr=  0 Current= 77W Cad= 51 r= 100 T= 15
23:02:10,673: Target=100W Speed=32.0kmh hr=  0 Current= 72W Cad= 41 r= 100 T= 15
23:02:11,676: Target=100W Speed=31.7kmh hr=  0 Current= 68W Cad= 41 r= 100 T= 15
23:02:12,678: Target=100W Speed=31.2kmh hr=  0 Current= 62W Cad= 36 r= 100 T= 15
23:02:13,680: Target=100W Speed=30.8kmh hr=  0 Current= 56W Cad= 29 r= 100 T= 15
23:02:14,682: Target=100W Speed=30.4kmh hr=  0 Current= 49W Cad= 21 r= 100 T= 15
23:02:15,683: Target=100W Speed=28.2kmh hr=  0 Current= 45W Cad=  0 r= 100 T= 15
23:02:16,685: Target=100W Speed=25.0kmh hr=  0 Current= 43W Cad=  0 r= 100 T= 15
23:02:17,688: Target=100W Speed=20.0kmh hr=  0 Current= 38W Cad=  0 r= 100 T= 15
23:02:18,690: Target=100W Speed=15.1kmh hr=  0 Current= 34W Cad=  0 r= 100 T= 15
23:02:19,692: Target=100W Speed=11.9kmh hr=  0 Current= 30W Cad=  0 r= 100 T= 15
23:02:20,694: Target=100W Speed= 6.9kmh hr=  0 Current= 25W Cad=  0 r= 100 T= 15
```

mikeymclellan, 2020-10-24:
Great work @m472 👍

totalreverse, 2020-10-24:
Looks a little like the communication described here:

https://github.com/fluxoid-org/CyclismoProject/wiki/Tacx-Bushido-Headunit-protocol

Maybe the Bushido and the Genius are not this different.

m472, 2020-10-25:
Thanks @totalreverse, I had a look at the Bushido protocol. The power, speed and cadence messages indeed seems to be pretty identical.

However, trying to set the slope I had no success yet. It seems the protocol here should be the same as with the Bushido:

From Bushido Protocol:
```
Tx: 0xDC01000000WW0000 Set no slope on the brake
WW = User weight = 0x50 = 80kg
```

I have been able to find this sequence in the TTS4 wireshark dump by @mikeymclellan:
0000   01 01 28 01 0f 00 00 00 00 00 00 00 00 00 00 00
0010   36 18 00 00 00 00 00 00 00 00 21 14 01 0b 01 02
0020   ff 00 00 00 cf 0f 09 10 **a4 09 4e 01 dc 01 00 00**
0030   **00 50 00 00 6f 00 00**

I was also able to replicate this message in FortiusANT:
```
21:57:13,558: Dongle    send   : synch=0xa4, len= 9, id=0x4e Broadcast Data       , check=0x6f, info="01 dc 01 00 00 00 50 00 00" (ch=1 p=220(??))
21:57:13,558: devAntDongle.write(0x01,"a4 09 4e 01 dc 01 00 00 00 50 00 00 6f")
```

Unfortunately changing the slope according to the Bushido protocol between 255 and -255 has no perceivable impact on pedaling resistance.

WouterJD, 2020-10-26:
Well guys, how could I add to this?
I have no Genius or Bushido so cannot do practical work; as soon as you are further - I'm happy to assist to integrate into FortiusANT.

m472, 2020-10-26:
I have now found the mistake in my attempt on setting the power yesterday. I tried to send exactly the same information as the TTS software did in the capture by @mikeymclellan but TTS sets the channels up differently. Settings slopes works now in my messing-around-branch https://github.com/m472/FortiusANT/commit/a420b4b382e7c8bda15e1c296827ab12dd305d35 (to change the resistance the hardcoded message has to be changed)

For me the question is now the same as for Wouter. How could I add this. I know now how the messages that are sent by the computer have to look and how to interpret the messages sent by the brake. What I am not sure about is, how I have to integrate this information into the existing software architecture.

My guess is the best approach would be to create a new `clsTacxAntGeniusTrainer` class. Paging and Unpaging of the messages would be added in `antDongle.py`. 

Open questions I have at the moment:

- The Genius brake expects a slope. As far as I understand the existing code, for the iVortex brake a power in watts is transmitted. Is this correct? And how could this be converted?
- The Genius slope command also transmits the weight of the user. How would this best be implemented?
- Is there a better way than just copying the `clsTacxAntVortexTrainer` code to avoid some of the code duplication? Maybe a mutual base class?

WouterJD, 2020-10-27:
Hi Mathias

`My guess is the best approach would be to create a new clsTacxAntGeniusTrainer class. Paging and Unpaging of the messages would be added in antDongle.py.` --> Agree

`The Genius brake expects a slope. ` you could choose to send power to the brake, but since the Genius supports a slope I would certainly do that. This means that in "Slopemode" another message is sent to the Genius than in "PowerMode". Should be doable in the new class.

`transmits the weight of the user` the weight of the user is available in the class. It's either the default value or the value as provided by Zwift (or so). field: UserAndBikeWeight.

`code duplication` I've been thinking of this and would do the following: first make a copy and develop the new class as suggested. When everything works, it's always possible to make a common class to de-duplicate. 

Other things to do:
- `FortiusAntCommand.py` : add `Tacx_Genius` like `Tacx_iVortex`
- `usbTrainer.py` : check for `clv.Tacx_iVortex` to create the new trainer type

Github is now updated with my last modification (removing minimum resistance for low FTP users).
I'm curious to see your code-suggestions.

m472, 2020-11-01:
I have now tried to implement the changes. See https://github.com/m472/FortiusANT/commit/db62a843ce183c099a494740a286fdcbc22099f2. I have had two successful rides on Zwift but not everything works as expected yet.
I am not quite sure weather the slope simulation works correctly because I am also new to zwift and was not really able to go to the hills intentionally.

Also connecting the brake to the computer did not work properly and I don't know how the connection starts to work. But once it has been connected and the brake is not restarted a restart of FortiusAnt seems not to cause any problems.

WouterJD, 2020-11-02:
@m472 great work and keep me updated.
If you have any questions or want me to investigate logfile (-d flag) let me know

mikeymclellan, 2020-11-02:
Nice work @m472 👍 

I am keen to do some testing on Zwift as well. Curious on your setup. I'd imagine you require three ANT+ dongles to achieve this? One to receive from the Genius, one to transmit  FE-C ANT from FortiusANT, and finally one for Zwift to receive on. 

Or can FortiusANT and Zwift use the same dongle for receiving?

WouterJD, 2020-11-03:
@mikeymclellan you need one dongle per application using it
FortiusANT will communicate with the trainer and with Zwift through the same dongle on different channels; one dongle supports up to 8 channels.

FortiusANT and Zwift cannot share a dongle.

Hence two dongles are required.

WouterJD, 2020-11-03:
@mikeymclellan what Tacx do you use?

mikeymclellan, 2020-11-03:
> Hence two dongles are required.

Makes sense, thanks @WouterJD 

> what Tacx do you use?

I've got the T2021 Genius

m472, 2020-11-04:
On monday I had a successful ride on zwift. Slope simulation worked as expected and I had no connection issues. Overall I was happy with that result.

Today I expected to have the same experience but was disappointed. I have not figured out what the reason for the connection problems was, but it worked after a reboot of the computer running FortiusANT. The second problem was, that the slope simulation did not work. I had wheel speeds of nearly 60 km/h on a 10% ascend. I attached the logfile of that run: [FortiusANT.2020-11-04 20-57-33.log](https://github.com/WouterJD/FortiusANT/files/5490697/FortiusANT.2020-11-04.20-57-33.log)

I have now started a debugging ride and stopped on a 10% incline but FortiusANT only sets the target slope to 1.0% (see picture). As the printed messages have nothing to do with the Genius specific protocol I guess the problem could be in the communication between FortiusANT and zwift. @WouterJD do you have an idea what could cause this problem?
[image]

@mikeymclellan do you have the head unit for the genius? I don't have it but would be interested if it works with the current implementation.

mikeymclellan, 2020-11-04:
> do you have the head unit for the genius? I don't have it but would be interested if it works with the current implementation.

@m472 Yes I do. I also bought another ANT dongle yesterday so I could do some testing. I got as far as installing everything on my Zwifting computer and ran out of time. I'll have another go tonight. 

What command line parameters are you using when riding with Zwift?

m472, 2020-11-04:
I start FortiusAnt using the following command:
```bash
python3 pythoncode/FortiusAnt.py -d 127 -H -1 -t i-Genius
```

mikeymclellan, 2020-11-04:
Great work @m472 🎉 I just did a quick test with my Genius and confirm that it seems to be communicating and showing up in Zwift. I haven't been for a ride yet but will try it properly later today. 

I pressed some buttons on the headunit but didn't see any related output. Is there something else I should try?

mikeymclellan, 2020-11-05:
Just took it for a 30 minute ride and everything works perfectly. 

However, I'm wondering if the reported power is higher than it should be. I also have a Kikr Snap trainer and on that I average about 50 watts less 🤔 

https://www.strava.com/activities/4287470911/overview

WouterJD, 2020-11-05:
@m472 I see that cadence = 0 in the logfile. This may be due to lose contact/no magnet etc. Tacx Fortius does not work well when there is no cadence.

So current status for the i-Genius development is that it conceptually works, but is not yet stable for @m472 ? Correct?

@mikeymclellan the "old" tacx trainers are not very precise nor repeatable in the power-reporting. It's also quite dependant of the tire-temperature. My yesterday's training was tenths of Watts off (realized versus requested power) but after half an hour is stabilized. This has to do with the calibration, which obviously moves during the training. I already experimented with a dynamic calibration (realized power must be equal to requested power) but did not activate it in the software because it seemed superfluous; where yesterday I had differences again. Yesterday the room was colder than some time ago ==> colder tire ==> longer warming-up time.

WouterJD, 2020-11-05:
@m472 @mikeymclellan could you share the current status of the code modifications you made?

m472, 2020-11-07:
@WouterJD Thank you for the hint with the cadence, I have not had any problems with the slope simulation if the cadence was measured correctly.

```
So current status for the i-Genius development is that it conceptually works, but is not yet stable for @m472 ? Correct?
```

Yes, I would agree. It works well if it works but I have the following issues with the code:

- I don't understand some of the code I copied and therefor I do not know if all the code makes sense.
  For example the ```TacxMode``` parameter in ```SendToTrainer```: is ```Tacx-Mode == stop``` needed for the Genius or not?
  I think for my setup it would not need to be used, but I don't know about the head unit.
- Is ```TargetPowerToResistance``` needed and if so, should ```SendToTrainer``` always send a grade instead of a power target?
  I first had it implemented like that but ERG mode works better in Zwift if I send target power instead (see latest commit: https://github.com/m472/FortiusANT/commit/e3a0c2658585d1bee6467dfb96d49d4882ecc017).
- As I don't have the head unit for the Genius, I am not quite sure what to do with the head unit code I copied from ```clsAntTacxVortexTrainer```
- I have only tested it in zwift for about 2-3 hours

Should I create a pull request to try to integrate the code or continue testing and try to find out how to implement the head unit stuff correctly (probably implement according to reverse engineered Bushido protocol and see if it works with @mikeymclellan)?

WouterJD, 2020-11-07:
hi @m472 for a starter I do not know how the Genius works, so it's difficult to answer questions on technical level.
A quick reaction now to assist to forward, I would need more time for an in-depth verification.

`Tacx-Mode == stop`
perhaps leave as is; merely needed for dongles that disconnect I think.

```
def TargetPower2Resistance(self):
    self.TargetResistance = self.TargetPower
```
If the Genius supports setting TargetPower or TargetGrade, that must be implemented
From what I understand is that the Vortex supports setting TargetPower only (at least: that is what is implemented)

I do not have Genius specs, so cannot help you here

Pull request:
It's most usefull that you submit the pull-request when the implementation is fine for you.

If you want me to do code-review in the meantime; yes submit it.

Success guys! Keep me posted.

WouterJD, 2020-11-08:
> * As I don't have the head unit for the Genius, I am not quite sure what to do with the head unit code I copied from `clsAntTacxVortexTrainer`

The i-Vortex consists of two devices; the brake and the headunit.
Your system works without the headunit, so the thing is "only" to provide buttons and manual control.

You could choose:
- implement brake-only
- implement brake and head-unit , but leave head-unit untested.

In clsTacxAntGeniusTrainer
- (pairing can take a minute) will remain visible since you do not have a head-unit, which is not nice. I think I would add a flag to ignore absense of head-unit and leave code complete (to be prepared for later extension).

In FortiusAntBody.py I would also leave the headunit code present

I do not know how you implemented the Genius channel info; did you add new variables or simply change the Vortex variables?

m472, 2020-11-09:
> You could choose:
> 
> - implement brake-only
> - implement brake and head-unit , but leave head-unit untested.

Decided to go for the second. Actually the button protocol of the Bushido and the Vortex is the same, so I guess its probably the same for the Genius as well.

> In clsTacxAntGeniusTrainer
> 
> - (pairing can take a minute) will remain visible since you do not have a head-unit, which is not nice. I think I would add a flag to ignore absense of head-unit and leave code complete (to be prepared for later extension).

I added a flag to the ```__init__``` method and the ```--no-headunit``` flag to the command line arguments.

> I do not know how you implemented the Genius channel info; did you add new variables or simply change the Vortex variables?

I guess you mean the variables ```DeviceTypeID_VTX```, ```RfFrequency_2466Mhz``` etc. I did create new variables for these in order not to affect the Vortex functionality

Sigfreid74, 2020-11-17:
This is amazing work people! I have now been testing the new additions made by m472 and by gosh it works! I have successfully been using my non-smart Genius with Golden Cheetah. 

I have a head unit; is there a way I can help you test the code out for that? So far it is not seeming to recognise the head unit. 

Also is there a calibration routine Genius users can use reliably now that we are free of TTS4? Ideally the wheel should be spun up to 30km/h for about 6-7 minutes in my experience with TTS4. I think the heating of the tyre changes the resistance or something.

WouterJD, 2020-11-17:
The title is changed, because the head-unit is the significant device; the head unit hides details from the software

WouterJD, 2020-11-17:
> I have a head unit; is there a way I can help you test the code out for that? So far it is not seeming to recognise the head unit.

Yes you sure could.
First question: are you able to program in python?

Sigfreid74, 2020-11-17:
Sorry Wouter, I am not able to program in Python (or anything else properly - I am very limited to just trying to change a few numbers here and there!). I just have the hardware ready to test it out and provide some logs.

[104 more replies]