WouterJD/FortiusANT #102: Strange power readings with TacX Flow

SwitchMcBlade, 2020-09-03:
I've got an old TacX Flow that I dug out to use over lockdown only to be disappointed that TacX have stopped me being able to re-install the TTS 3.5 software in order to use it again.  Talking to a friend about it he found your project and pointed me in this direction and I've been able to use it to successfully get everything working and be able to ride in Zwift.  I noticed that on the flats the system reports me as doing ~40w, I've no idea what my normal riding power was before and just accepted it as a thing and me not being bike fit after years of not doing much exercise apart from some light running.

Anyway, at the end of a zwift workout the other day I noticed something strange, as the trainer removed the resistance for the cooldown and I continued somewhat exhausted at a slow cadence of ~60RPM it was reporting a power output of around 75-80w, I started to pick up the cadence without changing gear and saw that as my cadence reached around 75RPM the power output dropped to showing ~40W.  

I'm not sure if I've missed a step with configuring the unit, I tried entering my tyre circumference and chainring/chainset sizes with the -b flag as mentioned in the wiki about the power curves in case that was needed but discovered that the -b flag isn't accepted now.  Using version 3.2.1 on Windows with FortiusANT running on the same machine as Zwift using 2 ANT+ dongles.

SwitchMcBlade, 2020-09-03:
Having a bit more of a play this evening it seems to be just when the gradient is 0 or when it is below 0.  Once the unit adds resistance from an increase in gradient everything operates as expected.  If I use the up arrow on the head unit to manually add resistance it seems to respond normally as well.  

Checking the command line when FortiusANT starts up I can see it says: "Calibration stopped because of unexpected resistance value"

For reference the brake has a label on it identifying it as a "FLOW Multiplayer Type T1901"
The head unit has a label on it saying "T1932 Fortius"

I've attached a photo of the brake unit and head unit.
[image]
[image]

WouterJD, 2020-09-03:
Will check lateron; in the meantime check wiki for commandline; has been changed since version 3
    	
    	

    	Met vriendelijke groeten Wouter Dubbeldam

WouterJD, 2020-09-17:
Did not forget you, will look into it. Waiting for wintertime for more inside-time :-)

SwitchMcBlade, 2020-09-20:
No problems.  As I've used it more I've started using the up and down arrows to control the resistance (especially when the gradient is 0 or lower in Zwift) and it's made it feel far more usable.

It seems that at a 0 gradient in Zwift or lower that even going flat out in the highest gear doesn't seem to report the power I'm putting out ( around 40w-60w and I'm struggling to hold that), yet when I tap up arrow to increase the resistance a bit I can change down a few gears, ride comfortably and be reading 80w that I can hold for a while.  Admittedly here I'm not very bike fit or all that fit at all.

WouterJD, 2020-09-24:
hi @SwitchMcBlade 

Zwift provides a target Grade. The target Grade (using current speed and bike weight) is translated to TargetPower.
At it's turn the TargetPower is translated to the required resistance.

I do not see a special around Grade=0

Please run a test, using command-line -d127 to produce a full logfile and provide that. 
Indicate at what log-time your experience is that the grade is incorrect

I will see what I can figure out

WouterJD, 2020-09-25:
@SwitchMcBlade @LitaliaFC @Stedec @fireone; see also #106
You're all (i-)Flow users; you might exchange info to the the Flow working.

To avoid that I have to study it all: (1) What is the difference between Flow and i-Flow and (2) how do you connect (USB or ANT).

SwitchMcBlade, 2020-09-25:
It seems there were two versions of the original Flow unit, the Flow (with a dumb control unit where you pressed buttons to manually change resistance) and Flow multiplayer which came with a head unit that connected to the computer via USB to work with the Tacx TTS software.

From what I can see the i-Flow was a newer model of the Flow multiplayer, there was an upgrade kit from Flow to i-Flow and all that is mentioned in the advert I found for that was an upgraded headunit with wireless support and an upgrade to TTS4 (the older Flow that I have came with TTS3.5). ( [Link to the upgrade kit](https://www.probikekit.co.uk/turbo-trainers-bicycle-rollers/tacx-flow-to-i-flow-turbo-trainer-upgrade-pack/10774122.html) 

All three units seem to share the same brake unit which, in case you don't know, is a flywheel with magnetic brake system.  So it is only capable of simulating uphill resistance, not downhill acceleration.  By memory I think it was advertised to simulate a maximum of 8% incline, but I'm struggling to find any information on that now.

WouterJD, 2020-09-25:
OK good information. Could you copy what FortiusANT displays as connected tacx, (The Txxxx number displayed).
I suspect it will be the headunit, so does not reveal what brake system is behind it.
Let's investigate together to get your machine working fine as well.

SwitchMcBlade, 2020-09-25:
The trainer shows as a T1932 in FortiusANT.

Attached should be a lap of the short Innsbruckring on Zwift.

From how it feels to me, when the gradient is 0 or lower it "feels" like the effort I'm putting in doesn't match the output power in the way it does when the gradient is 1 or higher.  

Timestamps I noted while riding:

17:57:50 - at this point the effort I was putting in felt the same as before but the reported power output went down.

16:00:50 - this time I noted that the effort feels the same but I get a higher output.

16:04:00 - I noted that again I felt like I was putting the same effort in but the power reported was lower.

16:17:30 - This is near the start of the downhill, at this point I felt I'd got to the point of doing similar effort to what I was near the top of the climb when it was 8%, yet the power reading is much lower. 

[FortiusANT.2020-09-25 15-56-16.log](https://github.com/WouterJD/FortiusANT/files/5284475/FortiusANT.2020-09-25.15-56-16.log)

I realise those reports are vague and relying on leg feel isn't very exact.  I don't have a power meter so can't give any more meaningful comparison information I'm afraid.

WouterJD, 2020-09-28:
Text from @fireone

Now Im struggling with the power readings. I get very strange power readings when the slope is 0% and it feels like overall my power readings are low compared to when running TTS. When I have the time I will try to make a comparison between power readings on TTS and FortiusAnt (try different slopes and speeds and maintain a steady cadence).

Probably related to the power issue is the fact that I cant calibrate the system. When I looked in the python code I saw a comment about it in FortiusAntBody.py "# iFlow 1932 is expected to support calibration but does not.". I guess this might be related to the power problems Im having. When running TTS the power readings and the brake always feels wrong for the first minutes and after a while it feels normal, I always assumed that the system does some sort of automatic calibration but of course I really have no idea :).

It might be interesting to snoop the USB traffic when running the TTS to see if there is some TACX magic going on initially when I feel that the trainer is not operating normally. This is on my to do list :).

WouterJD, 2020-09-28:
There are two ways to calibrate
- do the runoff
- automatic calibration, which is supported by the Tacx Fortius (New "white blue" Fortius headunit (firmware inside))

All the software knows is the headunit (the type of trainer should be provided through a command-line variable).
During the automatic calibration, a negative resistance is returned.
Please confirm that in your case message "Calibration stopped because of unexpected resistance value" is given.

Even that should not be an issue; standard calibration will be used. calibrate = 0 and in usbTrainer the default formula is used (Calibrate   = ( Calibrate + 8 ) * 130).

When reading onwards, note that PowerMode and GradeMode for the trainer both are ResistanceMode!

In PowerMode, the resistance is minimized to the Calibrate value (PowerMode would expect a positive value and it'd be strange when the motor starts rotating if you specify 10 watt.

In GradeMode, grade can be negative ... 0 ... positive (down/flat/up) and you would expect the resistance to be accordingly, but that's where the calibrate value realizes a shift.
So for an experiment on your trainer, check SendToTrainer() and find "Calibrate   = ( Calibrate + 8 ) * 130" which defines the zero-point. 

Experiment here what happens if you specify another value for the Calibrate value; e.g.0x0820.

And of course if you can reverse engineer the USB-interface using TTS, would be great. 
Check https://github.com/totalreverse/ttyT1941/wiki which could be helpfull.

fireone, 2020-09-28:
Ok, I wasnt aware that automatic calibration was possible with FortiusAnt. I have the T1932 which I believe is the new "white blue" Fortius headunit with firmware inside. Do I need to enable it in any way? (the automatic calibration)

"Please confirm that in your case message "Calibration stopped because of unexpected resistance value" is given."

Yes, this is given when I try the runoff calibration.

Ill try the experiments later tonight. 

You mention that I should provide what type of trainer I have but this was news to me. I start with (at least I think so, Im at work so I cant test it):

FortiusAnt.py -d127 -a -n -H -1

Is that a correct way to start it?

-n was added since I couldnt finish the runoff.

WouterJD, 2020-09-29:
The command line is correct, not starting the GUI

SwitchMcBlade, 2020-09-29:
Hi Wouter, 

I attempted to do the runoff test today but either the default resistance when starting FortiusANT is really high or I'm really unfit, as I could only get up to about 20-24Km/h and I was pedalling squares in my top gear.

roggenbrot, 2020-09-30:
Same here. In manual mode 40 km/h is no problem, but in runoff it's impossible (for me). What I also recognized yesterday when I was riding a hour in manual mode is that the resistance (power and grade mode) just seems to change for the first 3 config steps, an keeps constant after that.

I will try to apply a factor tonight and see if this changes something

WouterJD, 2020-09-30:
You are absolutely right.
In release 3.0 this problem has been introduced and is now resolved in FortiusAntBody.py; modeCalibrate was set which must be modeResistance.

So please retry:
- Runoff is intended to warm-up your tire; first cycle untill passing the 40 km/hr speed 
- Then stop pedalling; the wheel should stop in 7 seconds

When stopping faster, the role may be too tight / when rundown is taking longer the role may be too loose.
This procedure originates from antifier, I have no evidence on the figures; your feedback is welcome.

Sorry for slow response on this issue which has been around some time; I cooled down development during the summer months.

WouterJD, 2020-09-30:
If anyone needs the .exe let me know. I plan to make when current actions result in stable version, if you need will do sooner.

Timmenem, 2020-10-10:
Hey guys,

First of all, great Work @WouterJD, thanks for putting that much effort into this! I was using Antifier before, but since it crashes
like every second time I use it, I also wanted to give FortiusANT a try. I am also using a T1932 (white/blue) Headunit with a Flow,
in my case it's the white/orange "Flow i-Magic", but since it's also labeled T1901, I guess it's the same break as above.

When I start FortiusANT via command line and try to do the calibration, I also can't finish the procedure because even in highest gear I can't reach 40km/h. I noticed however, that whilst doing the calibration, current power and resistance are always negative and increase in value when I go faster. For example, at 30km/h I get -328W and a resistance of -4677. So, to me it seems as if the software was trying to decrease resistance in order for me to be able to reach the targeted 40km/h, but this isn't working. 

Is this even related to the issue above or is it just a display-issue with the sign?

SwitchMcBlade, 2020-10-10:
Hmms, I wonder if that is related to the behaviour I've noted a few times now.

When I use the up/down arrows to allow for some resistance on the flat such that I'm out-putting 100-140W, when I hit a steep hill in Zwift (8% or higher) it's as though the resistance unit may have "rolled over" and suddenly I'm pedalling to no resistance and the reading shows the usual ~40W.  I then have to down arrow frantically until I get resistance which will then come in in a big hit.

Timmenem, 2020-10-10:
I must apologize, I didn't have Git installed on my Windows setup and just used the "Download ZIP" button -
turns out I don't get the latest code through that. Just installed Git and now I also get the "Calibration stopped because of unexpected resistance value" when running from commandline, so my observations above are pretty much useless/outdated.

WouterJD, 2020-10-13:
I invited Matt to join ...
https://forums.zwift.com/t/now-zwifting-with-tacx-fortius/210567/41

WouterJD, 2020-10-15:
Hi guys, sorry for keeping you waiting.

From the software, all I know is the headunit (1932) and the software is texted with a Tacx Fortius with a T1932 headunit.
It seems that the configuration reacts differently, for example because the resistance would be maximized.

@Timmenem I understand that you try to calibrate using the Runoff option.
Non-fortius systems appear not support calibration, so use the -n command line flag to skip that step.

Next step is to try manual mode; -m = manual power.
See what happens; start is 100Watt, go cycling and try some wheelspeeds.
Then add power with the UP button and see what happens at 200W, 300W, 400W --- within your abilities.
And reduce power with DOWN button for -100W, -200W etc.

The means you are in ERGmode but decide the power yourself, no Trainer Rod, Zwift or Rouvy needed.

To create a logfile, use the -d127 flag.

Timmenem, 2020-10-20:
Hi Wouter,

I observe the same behaviour in manual mode, i.e. as long as I pedal with low rpm, the output power matches the target power (even though the output power seems to be less than it actually is), but as soon as I start pedaling at around 80-90rpm, the resistance drops so much that I get way less than the target power i set.

Are there more parameters I can play with than the `Calibrate` parameter from `SendToTrainer()`?
I attached a log file, maybe it can help.

[FortiusANT.2020-10-20 13-44-01.log](https://github.com/WouterJD/FortiusANT/files/5409123/FortiusANT.2020-10-20.13-44-01.log)

WouterJD, 2020-10-22:
See usbTrainer.py line 
```
    def TargetPower2Resistance(self):
        rtn        = 0
        if self.WheelSpeed > 0:
            rtn = self.TargetPower * self.PowerResistanceFactor / self.WheelSpeed

```
Note that Resistance is lineair with TargetPower and reversed linear with WheelSpeed.

I have checked you logfile, see attachment. [#102 FortiusANT.2020-10-20.13-44-01.txt](https://github.com/WouterJD/FortiusANT/files/5421886/102.FortiusANT.2020-10-20.13-44-01.txt)
The resistance seems normal (as designed) but it may have an incorrect effect for the Flow.

But note the following: for some time, the trainer returns 1559 as CurrentResistance; which is probably what you experience as the power drop.

I would analyse that first

```
13:44:15,378: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=14.2 TargetRes=3099 CurrentRes=4027 CurrentPower=133, pe=1 Connected to Tacx Trainer T1932
13:44:15,628: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=15.8 TargetRes=3012 CurrentRes=3118 CurrentPower=114, pe=0 Connected to Tacx Trainer T1932
13:44:15,880: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=16.2 TargetRes=2711 CurrentRes=3118 CurrentPower=117, pe=0 Connected to Tacx Trainer T1932
13:44:16,130: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=18.2 TargetRes=2645 CurrentRes=2598 CurrentPower=110, pe=0 Connected to Tacx Trainer T1932
13:44:16,382: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=18.9 TargetRes=2358 CurrentRes=2598 CurrentPower=114, pe=0 Connected to Tacx Trainer T1932
13:44:16,632: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=21.3 TargetRes=2260 CurrentRes=2078 CurrentPower=103, pe=0 Connected to Tacx Trainer T1932
13:44:16,884: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=54 Speed=22.5 TargetRes=2008 CurrentRes=1819 CurrentPower=95, pe=0 Connected to Tacx Trainer T1932

Cadence = 88; resistance goes down when wheelspeed goes up (that is expected behaviour)
Strange is that the current resistance (as returned by the trainer) is suddenly fixed at 1559
13:44:17,135: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.5 TargetRes=1903 CurrentRes=1559 CurrentPower=81, pe=0 Connected to Tacx Trainer T1932
13:44:17,385: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.9 TargetRes=1903 CurrentRes=1559 CurrentPower=83, pe=0 Connected to Tacx Trainer T1932
13:44:17,635: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.9 TargetRes=1870 CurrentRes=1559 CurrentPower=83, pe=1 Connected to Tacx Trainer T1932
13:44:17,887: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=23.3 TargetRes=1870 CurrentRes=1559 CurrentPower=84, pe=0 Connected to Tacx Trainer T1932
13:44:18,138: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=23.3 TargetRes=1838 CurrentRes=1559 CurrentPower=84, pe=0 Connected to Tacx Trainer T1932
13:44:18,388: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.9 TargetRes=1838 CurrentRes=1559 CurrentPower=83, pe=0 Connected to Tacx Trainer T1932
13:44:18,638: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.9 TargetRes=1870 CurrentRes=1559 CurrentPower=83, pe=0 Connected to Tacx Trainer T1932
13:44:18,890: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=88 Speed=22.9 TargetRes=1870 CurrentRes=1559 CurrentPower=83, pe=1 Connected to Tacx Trainer T1932
13:44:19,141: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=93 Speed=22.9 TargetRes=1870 CurrentRes=1559 CurrentPower=83, pe=0 Connected to Tacx Trainer T1932

```

Lorangaw, 2020-10-23:
I would claim the same problems on my I-magic. I can't generat enough speed for the run off and in some settings can't get more power than 50-60 watt while I know it should be 250 w +. It seems that while riding routes in Zwift it works fine but while following exercises, it fails when trainer should adjust power. in recoveries were it is "free-ride" it acts correct again.

WouterJD, 2020-10-26:
@Lorangaw Please check other posts and submit logfile on your system.
If such data is provided it must be possible to find out what is special about your systems

WouterJD, 2020-11-05:
I understand Tacx Flow works fine for some, there are questions for others.
Please provide more info, using a structured test (manual mode, 0 - 50 - 100 - 150 ... Watt + logfile).
Issue is candidate for closure now

e7andy, 2020-11-09:
I got the exact same setup as SwitchMcBlade.

When I don't use the Fortius unit (T1932), but the head unit that came with my trainer with a display the power readings on that unit and the values from my power pedals are pretty much the same. Good enough.

When I switch head unit to T1932 and use FortiusANT I get 50-70 watt and at the same time my power pedals show 100-200 watt. The power reading increases when I increase the value in manual mode, but is always lower. When the value is increased to 250 W I get almost the same value. 210 W from FortiusANT and 250 W from power pedals.

I did a few debug runs today.
Start parameters was
-n -a -A -H0 -P -d127 -m
or 
-n -a -A -H0 -d127 -m

I increased the value up to 300 W and then down to -300 W.

I just noticed that the cadence wasn't measured. I guess the sensor was too far away from the magnet.

[FortiusANT.2020-11-09 12-57-21.log](https://github.com/WouterJD/FortiusANT/files/5510909/FortiusANT.2020-11-09.12-57-21.log)
[FortiusANT.2020-11-09 13-03-03.log](https://github.com/WouterJD/FortiusANT/files/5510910/FortiusANT.2020-11-09.13-03-03.log)
[FortiusANT.2020-11-09 13-06-13.log](https://github.com/WouterJD/FortiusANT/files/5510911/FortiusANT.2020-11-09.13-06-13.log)
[FortiusANT.2020-11-09 13-08-45.log](https://github.com/WouterJD/FortiusANT/files/5510912/FortiusANT.2020-11-09.13-08-45.log)
[FortiusANT.2020-11-09 13-13-28.log](https://github.com/WouterJD/FortiusANT/files/5510913/FortiusANT.2020-11-09.13-13-28.log)

WouterJD, 2020-11-10:
Hi @e7andy I would suggest to make a power analysis as described in the manual; 
```
2.6	Power curve validation	12
2.6.1	General	12
2.6.2	Test for Tacx Fortius (T1932)	13
2.6.3	Test for i-Magic (T1902)	14

```
What is the headunit that DOES work?

What I know is that, Fortius/T1932 does not work (well) without cadence sensor.

Please understand I leave analysis of the logfiles to you (because of time effort), if you need assistence with that just let me know.

e7andy, 2020-11-15:
@WouterJD The head unit that I get pretty good values from is the one that came with my trainer.
Image that I found on the web:
[image]

I will try to do the power curve validation described in chapter 2.6 and fix the issue with the cadence measurement. Better to do the analysis of log files when I have all the data in them.

WouterJD, 2020-11-15:
Hi all. Version 3.6 has a new logging feature -d32 to create a JSON file for this kind of analysis.
Please let me know whether that helps to investigate.

WouterJD, 2020-11-17:
I'm digging into this "Power curve" subject and get the following impression: headunit/brake combinations T1932/T1941 and T1902/T1901 work, but the others don't.

Please respond to this post: what bundle did you buy, and what brake and what head unit do you use?

Bundle | Brake | Head unit | Remarks
-- | -- | -- | --
T1930 Tacx Fortius | T1941 Motor | T1932 | USB interface tested, see manual 2.6.2
T1900 Tacx i-Magic | T1901 Magnetic | T1902 | Legacy USB interface tested, see manual 2.6.3
T???? Tacx Flow | T1901 Magnetic | T1932 | Issues: #102 #143
T???? Tacx i-Magic | ? | T1904 | Issues: #128

e7andy, 2020-11-17:
I got bundle "T2200 Taxc Flow" with T1901 brake and T1932 head unit.

SwitchMcBlade, 2020-11-17:
I bought the "Tacx Flow Multiplayer VR Turbo Trainer" which came with the T1901 Brake, T1932 head unit and the Steering unit (there's no obvious model number on this bit).

I don't know the model number for sure (it's not on the invoice e-mail), but from googling the bundle was T2220

Timmenem, 2020-11-17:
I am also using a T1901 brake with a T1932 headunit. My brake originally came with a non-smart headunit T1682, I got the "Upgrade i-Flow" package that includes the T1932 HU.

Lorangaw, 2020-11-17:
I have the T1901I-magic break and T1904 head unit.
Br Morgan

WouterJD, 2020-11-18:
This seems the same as #143 and I will proceed from there
The [response from @cyclingflow ](https://github.com/WouterJD/FortiusANT/issues/143#issuecomment-729382076) is promising.

closed 2020-11-18 by WouterJD