WouterJD/FortiusANT #103: Please implement Bluetooth support

WouterJD, 2020-09-07:
It disconnects several times and reconnects, only issue that's causing for now is that it seems to hinder the calibration process...

BLE - what I meant is if it would be possible to broadcast the data using bluetooth instead of ANT+, no dongle needed in this case.

Thanks a lot for this!

_Originally posted by @mruzzon in https://github.com/WouterJD/FortiusANT/issues/61#issuecomment-688177189_

WouterJD, 2020-09-07:
There are two sides to this
- implementing a new interface is a challenge
- but considerably more costlyt (in hours and effort) than bying some ABT-dongles

But it's noted as a feature request and if anyone likes to do a code-suggestion; I will gladly integrate into the code.

aegisdesign, 2020-09-07:
I'll +1 this. Adding Bluetooth support opens up the possibility of using more hardware such as phones, tablets/iPads and things like AppleTV that don't support ANT+ but run Zwift and other apps well.

Lots of work though.

roggenbrot, 2020-09-27:
https://github.com/jedla22/BleTrainerControl/blob/master/How-to%20FE-C%20over%20BLE%20v1_0_0.pdf

On Linux it should be no problem to create a BLE GATT server via bluez, on windows/mac not idea.

Protocol seems to be the same as used for ANT+

aegisdesign, 2020-09-27:
pybluez ?It's multi platform. https://github.com/karulis/pybluez

https://github.com/jedla22/BleTrainerControl/blob/master/How-to%20FE-C%20over%20BLE%20v1_0_0.pdf

roggenbrot, 2020-09-28:
BLE part which is required to implement GATT is linux only and experimental.

Microsoft has a documentation how to implement a GATT server with c#. But this would require a python wrapper which would cause additional effort.

marcoveeneman, 2020-10-29:
I'm interested in this feature since i have a tacx fortius (T1932) trainer laying around but no ant+ dongles (and my macbook only has only 2 usb ports.. 🙄)

Let me see if i can put something together.

WouterJD, 2020-10-29:
Well; two USB-ports would do.

And if not, a USB-hub is a quick fix; avoiding the whole software development traject.
Two USB-dongles (#61) and a [USB-hub](https://www.google.com/search?q=usb+hub) costs € 75,= and you are good to go this weekend :-)

[image]

marcoveeneman, 2020-11-01:
Hi Wouter, I took the challenge and tried to get Bluetooth (BLE) working this weekend. I just rode my first couple of meters in Zwift on my Apple TV by running FortiusANT with BLE support on my MacBook. I'm Zwifting from Nijmegen btw.

Ok, so instead of implementing Bluetooth (BLE) support directly in FortiusANT i chose to handle the BLE functionality as a separate process in NodeJS. At startup, FortiusANT should start the NodeJS server so it can then communicate with it. Instead of writing data to an ANT+ dongle it can now send a message to the NodeJS server which will then handle the BLE stuff.

Would you be interested to include the changes in FortiusANT? If so, i would first propose some serious code refactoring tbh. It was quite hard to wrap my head around the code in the current state and fit the BLE part in.

Just let me know, i'd be happy to help :)

WouterJD, 2020-11-01:
Marco; great work, compliments!👍
And yes i would integrate.

Als je wilt kunnen we wel een conference call organiseren om door te spreken op welke wijze te integreren

WouterJD, 2020-11-01:
PS. Welke Tacx heb jij?

orrmany, 2020-11-02:
Hi @WouterJD congrats for your work. I came over here due to your comment https://www.strava.com/activities/4273307198. :)
Learning Python is on my bucket list, maybe understanding your source code can be a good quest  to start with :)

WouterJD, 2020-11-02:
@marcoveeneman has done investigations; next steps are to engineer and integrate

marcoveeneman, 2020-11-02:
> Marco; great work, compliments!👍
> And yes i would integrate.
> 
> Als je wilt kunnen we wel een conference call organiseren om door te spreken op welke wijze te integreren

@WouterJD Great, then i'll proceed with it. I'm using a Tacx Fortius Multiplayer T1932. I've sent you a message to setup the conference call.

WouterJD, 2020-11-02:
> I've sent you a message to setup the conference call.

Hoe? Doe maar een ping op 0651103725

orrmany, 2020-11-02:
guys, consider to implement the FE-C profile, too, with BLE, see issue #122  for rationale

WouterJD, 2020-11-02:
We're working on it and implement stepwise with purpose a full implementation. I'd be curious to know why zwift does not support ANT/FE-C and Rouvy's ANT runs behind on Bluetooth...

orrmany, 2020-11-03:
Proposal is to transmit _simultaneously_ on BLE and ANT+. I have to note that not all computers have BLE built in, not even laptops.

msjnaessens, 2020-11-03:
Best would be for Zwift to just support ANT+ FE-C on android tho...
https://forums.zwift.com/t/ant-update/195323
If there were enough requests for the feature, they would consider implementing it. Maybe the feature will be available soon and it would be a waste of effort implementing the conversion in FortiusANT.

Another option would be the NPE Cable ANT+ to BLE converter (https://youtu.be/HvrYcOyd7wA) which does support FE-C control. (Often mentioned in zwift forums).

I am certainly not against the implementation in fortiusANT. Now, I have to use my laptop for running zwift, which has to be installed and removed at each session. It is also not ideal to use an expensive laptop when sweating... I would love to run Zwift on my phone, casting zwift over usb to my screen. Either the implementation of the BLE/ANT conversion or a 1155 socket motherboard is needed to change my laptop for an other option...

marcoveeneman, 2020-11-08:
As you can see i just opened a pull request where initial support for Bluetooth is added. It is working fine for me, but i would be very interested in results of other users. Please let me know if it works ok for you too or if you encounter any issues. Thanks!

brauereik, 2020-11-11:
Great work so far Marco. I've just cloned your PR and had issues to connect to 'Virtual Trainer' either through Tacx Training App or RGT Cycling on Android11 - Pixel3a. Pairing the mobile with the trainer directly asked for the pin connects and disconnects right away. FortiusAnt was started on OSX 10.14.6 > python3 FortiusAnt.py -b -t i-Vortex -g -a 

Any logs you like to see? BTW, I'm from Germany

WouterJD, 2020-11-11:
Hi @brauereik welcome in the FortiusANT community!

marcoveeneman, 2020-11-11:
> Great work so far Marco. I've just cloned your PR and had issues to connect to 'Virtual Trainer' either through Tacx Training App or RGT Cycling on Android11 - Pixel3a. Pairing the mobile with the [...]

Hi @brauereik, thanks for taking the effort in testing my PR! So if i understand correctly FortiusANT did actually start the BLE server and started advertising, nice, good to know that works for someone else as well.

To better understand why the connection was dropped there might be some information in a file called ble.log, which should be located in the pythoncode folder. Note that this file is currently overwritten each time you start FortiusANT. Can you provide the log file so i can have a look at it?

marcoveeneman, 2020-11-11:
I installed both Tacx Training and RGT on my iPhone and both are connecting with the Virtual Trainer (I am thinking to rename this to FortiusANT for clarity). I guess this is an Android vs iOS issue. I'll have a look if there are some special requirements for Android to connect.

marcoveeneman, 2020-11-11:
@brauereik I forgot to ask, what do you mean exactly by pin connects?

brauereik, 2020-11-12:
[ble.log](https://github.com/WouterJD/FortiusANT/files/5530021/ble.log)

I meant Bluetooth pairing code.

brauereik, 2020-11-12:
It's odd: sometimes the BT device found on Android is called mini as in the OSX BT-settings ans sometimes VirtualTrainer.

marcoveeneman, 2020-11-12:
Thanks! I had a look at the file but unfortunately i cannot deduce the issue from the log file at the moment. I'll look into how to improve logging there.

Regarding the pairing code: interesting, there are no secure characteristics which would require secure pairing, so i'm a bit surprised Android is asking for a pin here.

brauereik, 2020-11-20:
With the latest Update for Android 11 I can finally report success! I can finally test how it feels to have the trainers slope being controlled by RGT, which is planned for the weekend.

orrmany, 2020-11-20:
Hi, what is the motivation for a NodeJS based BLE support instead of Python based BLE support?

WouterJD, 2020-11-20:
@marcoveeneman I know you have studied the subject, please provide the arguments
@orrmany Would you think a platform indepent solution is available; please suggest

marcoveeneman, 2020-11-21:
@brauereik Awesome! So with Android 10 it didn't seem to work, but with Android 11 it does?

@orrmany Motivation is that there doesn't seem to be a complete, cross-platform BLE server implementation available in python, at leas i couldn't find one yet. The bleno project has all of this, except it is in NodeJS instead of python. However, if you know a python library, that would be awesome and i'd be willing to convert this thing to 100% python.

For now, it is functional, which i think is very important in the first place. If the interface to the BLE functionality is abstracted correctly, we could swap the implementation from NodeJS to Python at a later point without much hassle.

orrmany, 2020-11-21:
> @orrmany Motivation is that there doesn't seem to be a complete, cross-platform BLE server implementation available in python, at leas i couldn't find one yet. The bleno project has all of this, [...]
FYI: I am not a Python programmer, so I have no experience. There are a number of libraries according to Google, say https://github.com/nccgroup/BLESuite, but I trust your judgement here. I was just curious, since in my work experience, mixing different programming technologies within the same project always leads to a maintenance hell.

WouterJD, 2020-11-22:
> mixing different programming technologies within the same project always leads to a maintenance hell

Agree...but I assume it's done because it's the best solution available

What would you suggest?

orrmany, 2020-11-22:
On short term, I can't suggest anything. On long term: learning Python is
on my bucket list. After cleaning up my current hobby project (an ANT+ &
BLE sensordata-logger using an Arduino compatible nRF52840 MCU) it sounds
as a good hobby project for me. Don't hold your breath, it would take at
least a year for me, or more, considering my workload at my employer...

Wouter Dubbeldam <notifications@github.com> ezt írta (időpont: 2020. nov.
22., V, 13:32):

switchabl, 2020-11-23:
I have been looking for a cross-platform BLE library too and haven't found much. For the central role, Bleak looks really good, but unfortunately they decided against implementing the Peripheral role (which is what we really need).
https://github.com/hbldh/bleak/issues/81

So it looks like bleno is probably the quickest way forward. Having it as an optional feature that is only losely coupled through sockets is probably a good thing here. I do worry that people having to install extra depencies themselves may be a support nightmare in addition to the maintenance issue. So having a complete package for people to download on Windows might make sense.

I had a look at the bleno code as well and the way it is built has advantages as well as disadvantages. On Linux and Windows they seem to use libusb and essentially implement their own Bluetooth stack. This is similar to the situation with ANT:
- people have to install another libusb driver
- not all BLE hardware might be compatible
- other applications can't use Bluetooth anymore (and afterwards you have to switch back drivers) unless you have a second Bluetooth adapter
- on the plus side, you CAN potentially run FortiusANT and a software that uses our simulated trainer on the same PC (though most BLE clients will probably be mobile apps on a tablet/phone anyway I guess)
- you can simulate a HID device (like a keyboard), which i *think* the native Windows stack doesn't allow
On macOS it looks like they use a reverse-engineered, internal interface to the Bluetooth daemon instead of the official API. That is bad because it might break at any time.

It would be nice if we could have the profile implementations in Python, so it would be easier to switch to a native backend in the future, but looking at the bleno architecture, that may not be feasible.

In principle, accessing the native BLE APIs from Python should now be relatively smooth:
- Windows 10 now has winrt (https://pypi.org/project/winrt/) bindings for the UWP API
- Linux/BlueZ has a straightforward D-Bus interface (and there are lots of Python packages for that)
- PyObjC (https://pypi.org/project/pyobjc/) should access to Core Blutooth on macOS
But having to write and maintain three backends is not greate either. If BLE support is made sufficiently modular, maybe we can have Node.js as a fall-back option and implement native support for one platform at a time.

WouterJD, 2020-12-04:
I have done some tests with Marco's implementation.

(It crashes somewhere but that could be solved perhaps)

To make it work
- buy an BLE-dongle; builtin bluetooth is not supported

install
- zadig (BLE driver)
- node.js

then run
- npm install
```
gyp ERR! find VS **********************
gyp ERR! find VS You need to install the latest version of Visual Studio
gyp ERR! find VS including the "Desktop development with C++" workload.
gyp ERR! find VS For more information consult the documentation at:
gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
gyp ERR! find VS **********************

```
but node.js on windows requires (I do not know yet which of the two steps is really needed)
- visual studio (or)
- npm install --global windows-build-tools
and (on my system) this causes python 2.7.1 being installed, causing version issues

**I think it is a LOT of work, to get the bluetooth interface work.**
But yeah; you can use an old computer and run Zwift on your mobile device...

There are two approaches:
- It's an optional feature; only for the advanced user
- Do not implement since quite some users already have issues with the libusb installation; but that is mandatory for the Tacx USB device. They most likely will not be able to get through the bluetooth installation. It may cause a lot of issues ("helpdesk load")

Just sharing my thoughts at this point. Now I have the dongle, I would want to see it work.

switchabl, 2020-12-04:
At least on Windows, it is probably a better idea to package all the node.js stuff into a single executable rather than have the user set it up. It seems https://github.com/vercel/pkg should do the job, but someone with more node.js experience might want to weigh in.

The alternative is to go through the native Bluetooth LE API. That would solve a lot of problems: works with all BLE hardware, no zadig driver, can be Python-only. But it adds another big one: separate implementations for Windows, Linux and Mac. So likely would either be Windows-only or a lot of work.

TacxBiker, 2020-12-23:
Hi Wouter,
I'm new to this all, so first of aal thanks for your work and effort.
I can start the bluetooth version without any error. I try to start in sim mode ( -a -n -s -b -d127 -g)
The gui tells there a Bluetooth interface available, so next step should be pair a tablet / bike computer / smartphone
when triyng to do so, none of these devices find a FortiusANT Trainer
I probably miss a step.
[FortiusANT.2020-12-23 12-00-09.log](https://github.com/WouterJD/FortiusANT/files/5734749/FortiusANT.2020-12-23.12-00-09.log)
[FortiusAntGUI.2020-12-23 12-00-10.log](https://github.com/WouterJD/FortiusANT/files/5734750/FortiusAntGUI.2020-12-23.12-00-10.log)

marcoveeneman, 2020-12-23:
> Hi Wouter,
> I'm new to this all, so first of all thanks for your work and effort.
> I can start the bluetooth version without any error. I try to start in sim mode ( -a -n -s -b -d127 -g)
> The gui tells there a Bluetooth interface available, so next step should be pair a tablet / bike computer / smartphone
> when trying to do so, none of these devices find a FortiusANT Trainer
> I probably miss a step.
> [FortiusANT.2020-12-23 12-00-09.log](https://github.com/WouterJD/FortiusANT/files/5734749/FortiusANT.2020-12-23.12-00-09.log)
> [FortiusAntGUI.2020-12-23 12-00-10.log](https://github.com/WouterJD/FortiusANT/files/5734750/FortiusAntGUI.2020-12-23.12-00-10.log)

Hi TacxBiker, I guess you started the version from #179 right? Currently Wouter and i are very close to finish integrating bluetooth. However this is done in two separate PRs. #179 implements the necessary logic inside FortiusANT python code, while #130 implements the specific bluetooth logic separately, you need both to get it working. When the two versions are merged, bluetooth support should work (note the setup required for starting Bluetooth in the doc folder of #130, we currently consider bluetooth an advanced feature since it requires extra setup for the user at the moment.)

WouterJD, 2020-12-24:
Ho @TacxBiker 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

TacxBiker, 2020-12-24:
Hi,

when starting FortiusANT with bluetooth support I get the following error:

13:42:30,448: ... requests.post() error HTTPConnectionPool(host='localhost', port=9999): Max retries exceeded with url: /ant (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x0828A4C0>: Failed to establish a new connection: [WinError 10061] Kan geen verbinding maken omdat de doelcomputer de verbinding actief heeft geweigerd'))

It seems the HTTP server for receiving the data from Fortius ANT isn't online

Is there a way to check that?

TacxBiker, 2020-12-24:
One step further. To start de server simple i simply can use this command in de \node folder

Node server.js

But then unfortunately the next error:

C:\Github\FortiusAnt\FortiusANTBLE\FortiusANT-master\node\node_modules\@abandonware\bluetooth-hci-socket\lib\usb.js:90
    throw new Error('No compatible USB Bluetooth 4.0 device found!');

But there should be a right bluetooth device. 
I changed the driver with the zadig tool.

The bluetooth device I use has the following ID: USB\VID_0A5C&PID_21F1
so that should not be the problem.

TacxBiker, 2020-12-24:
And there we go:
[image]

TacxBiker, 2020-12-24:
I added a new device in the \bluetooth-hci-socket\lib\usb.js file
var VENDOR_DEVICE_LIST = [
  {vid: 0x0CF3, pid: 0xE300 }, // Qualcomm Atheros QCA61x4
  {vid: 0x0a5c, pid: 0x21e8 }, // Broadcom BCM20702A0
  **{vid: 0x0a5c, pid: 0x21f1 }, // Broadcom BCM20702A0**  
  {vid: 0x19ff, pid: 0x0239 }, // Broadcom BCM20702A0
  {vid: 0x413c, pid: 0x8143 }, // Broadcom BCM20702A0
  {vid: 0x0a12, pid: 0x0001 }, // CSR
  {vid: 0x0b05, pid: 0x17cb }, // ASUS BT400
  {vid: 0x8087, pid: 0x07da }, // Intel 6235
  {vid: 0x8087, pid: 0x07dc }, // Intel 7260
  {vid: 0x8087, pid: 0x0a2a }, // Intel 7265
  {vid: 0x8087, pid: 0x0a2b }, // Intel 8265
  {vid: 0x0489, pid: 0xe07a }, // Broadcom BCM20702A1
  {vid: 0x0a5c, pid: 0x6412 }, // Broadcom BCM2045A0
  {vid: 0x050D, pid: 0x065A }, // Belkin BCM20702A0
  {vid: 0x1286, pid: 0x204C }, // Marvell AVASTAR
  {vid: 0x8087, pid: 0x0025 }, // Dell Precision 5530
];

TacxBiker, 2020-12-24:
Connection to an iPad in the Sufferfest app:

[image]

and with all the data:

[image]

WouterJD, 2020-12-24:
@marcoveeneman you evaluare this?

TacxBiker, 2020-12-24:
FortiusANT receives just like it should the power from the CPT.
But when the power goes up by 50W it doesn't feel like so.
Maybe this has something to do with #173

marcoveeneman, 2020-12-25:
Hi @TacxBiker, I see you got the BLE interface up and running eventually, great!
The change you applied is something that would ideally be integrated in https://github.com/abandonware/node-bluetooth-hci-socket/blob/master/lib/usb.js, since this is part of the library we use. Can you open an issue + PR in that repository?

WouterJD, 2020-12-25:
@marcoveeneman perhaps add a note in the documentation as long as not implemented. Merry Christmas🚴‍♀️🚴‍♀️

mtbiker22, 2020-12-26:
Hi, trying to get FortiusANT up and running on a raspberry pi 3..... So far FortiusANT starts, callibration works, but bluetooth is not working.

Can anybody point me out how to setup and use node.js of the BTLE support?

TacxBiker, 2020-12-28:
@marcoveeneman 
Done, see what happens. Thanks

hleidecker, 2021-01-01:
@WouterJD and other contributors, I really appreciate all your work on FortiusAnt. It worked 'out of the box' on Windows with two Ant+ dongles.

I can confirm that Bluetooth is working with Linux - tested today with a T1932, Raspberry Pi 3B, Raspbian and Zwift.

@marcoveeneman one observation - the python script must be executed from the 'pythoncode' directory i.e. 'sudo python3 ./FortiusAnt.py' not 'sudo python3 ./FortiusAnt/pythoncode/FortiusAnt.py' as stated in the user manual.

The python script/application tries to execute the node.js script/application from '../node/' relative to where the python script/application is executed from.

In my case it ended up looking for the node.js script/application in '/home/pi/node/' as I have the FortiusAnt source in '/home/pi/git/FortiusAnt' when using 'sudo python3 ./FortiusAnt/pythoncode/FortiusAnt.py'.

TacxBiker, 2021-01-03:
@hleidecker  Is it possible to post a small guide for bluetooth on linux / Paspberry Pi?
I'm stuggling with this

hleidecker, 2021-01-03:
> @hleidecker Is it possible to post a small guide for bluetooth on linux / Paspberry Pi? Or the steps you have done to get it working
> I'm stuggling with this

@TacxBiker, with the basic FortiusAnt functionality up and running I did the following to get it working with the Raspberry Pi 3B:

1. Install NodeJS
`sudo apt-get install nodejs`

2. Install dependencies: 
`sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev`
`cd node && npm install`

2. Run FortiusAnt with automatic start, BLE support, simulated trainer and any heart rate monitor
`cd FortiusAnt/pytoncode` (must be run from this directory)
`sudo python3 ./FortiusAnt -a -b -s -H0`

Please note that the system had a lot  of stuff already installed ... in my pursuit of upgrading to the latest version of wxPython - I eventually gave up as I never succeeded in building it. 

Let me know if this works for you.

WouterJD, 2021-01-04:
@hleidecker 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

closed 2021-01-09 by WouterJD

StFalagar, 2021-03-03:
Hi @WouterJD 

Thanks... I got this working...  Connected to Zwift and RGT so far.  Also using Golden Cheetah with a usb connection.  Just as a side note (not sure if covered before) I am using a Microsoft Surface Pro 4 and only need one BLE dongle for Fortius Ant.  The Surface's built in Blue Tooth handles the Zwift Connection.  I am also able to connect to it using the companion app...  thanks again for this awesome initiative :)

WouterJD, 2021-03-05:
@StFalagar thanks for your nice words and the update

PR #130 (2020-12-24): Bluetooth support
# Adding Bluetooth support to FortiusANT

This pull-request adds _initial_ and _experimental_ support for BLE (Bluetooth Low Energy) to FortiusANT, as requested in #103. Bluetooth can be enabled by starting FortiusANT on the command line with the `-b` option.

The current implementation has been tested using the following setup:
- Tacx Fortius T1932
- MacBook Pro 2015 running macOS Catalina
- Zwift running on AppleTV4 with tvOS 14.0.2

FortiusANT currently implements the following BLE services to connect to Zwift:
- FTMS: FiTness Machine Service
  - Indoor Bike Data Character [... cut]