WouterJD/FortiusANT #106: Calibreren

LitaliaFC, 2020-09-25:
Hallo Wouter,

Zoals verzocht via Github; ik heb een TacX I-Flow 2230, die ik nu via een CYCPLUS dongle heb aangesloten aan Fortius. Echter, loop ik tegen een aantal zaken aan. Waarschijnlijk heb ik zelf eea verkeerd aangesloten of niet goed gelezen. Maar bij het calibreren lijkt het alsof de resistance te hoog is. Ik kom niet in de buurt van de 40km/uur, en de resistance is niet evenredig, het hapert bij elke omslag. Ook heb ik het idee dat wanneer ik flink kracht zet, de buitenband wat slijt, omdat het dan heel licht naar verbrand rubber ruikt. Heb je een suggestie waar dit aan zou kunnen liggen?

Gr Livio

PS: Voor het gemak heb ik dit in het Nederlands getikt, aangezien ik vermoed dat je ook Nederlands bent. Als dit niet zo is / je het liever in het engels ziet, zal ik het aanpassen.
[image]
[image]

WouterJD, 2020-09-25:
Ik ben inderdaad Nederlander, dus NL-support is geen enkel probleem.

De Flow ziet er net uit als de Tacx Fortius. Ik ga er vanuit dat je deze dus ook met een USB kabel aansluit.
Zoals je in #61 kunt lezen is de CYCPLUS dongle een ding met kuren; maar dat zou voor de calibratie niet uit moeten maken.

Ik calibreer (runoff) zelf nooit omdat de Fortius een automatische calibratie heeft.

Wanneer je rubber ruikt heb je slip op de band en zou je dus de rol strakker op de band moeten zetten / bandenspanning controleren. Op dezelfde manier als wanneer je Tacx TTS gebruikt (werkt dat nog?).

Functioneert alles en twijfel je aan de resistance (zou voor de Flow wel eens anders kunnen zijn als de Fortius) maak dan een logfile met de -d127 command-line parameter.

Succes

LitaliaFC, 2020-09-25:
Bedankt voor je snelle reactie. Zo simpel is het dus, heb de rol strakker op de band gezet, en nu geen slippende band meer en geen rubbergeur. Het is de eerste keer dat ik een TacX gebruik, dus het is even zoeken. Volgens mij krijg ik nu ook een logischere resistance, zonder het wegslippen van de band. Aangaande je opmerking van de CYCplus, wat voor Usb Ant zou je adviseren? 

TTS werkt volgens mij niet meer, ik moest de trainer via libusb verbinden met mn laptop, zodat de driver kon worden geupdated. Voor een niet-ITer al best een opgave. Om met Zwift te verbinden heb ik sowieso een 2e USB Ant nodig toch? Die zal ik nog even moeten kopen, een betere dan CYCplus. Maakt het nog uit welke USB Ant ik met Fortius verbind en welke met Zwift? Als in, is het logisch de kwalitatief betere USB Ant te verbinden met Fortius, of maakt dat niet uit?

WouterJD, 2020-09-25:
Zie #61 eigenlijk lijken alle ANT+ dongles goed, behalve de CYCPLUS.
Als je twee verschillende hebt, probeer dan of de CYCPLUS met zwift werkt.

2e USB-ant; inderdaad - zie wiki.

En ... ik heb inderdaad het plan om een overall document te maken waar e.e.a. wat beter in beschreven staat; vooral de USB-driver is lastig. Maar dat het je al voor elkaar; niet meer aankomen zou ik zeggen.

TTS en FortiusANT samen op één PC? zie #99

Dit issue kan gesloten worden denk ik.,

closed 2020-09-25 by WouterJD