WouterJD/FortiusANT #107: Problem with inital setup of T1932

roggenbrot, 2020-09-26:
After several years i've decided  to give my Tacx Flow another chance (Winter is coming) but i'm struggling on initial setup. 

Before spending money for 2 ant dongles I wanted to try if i'm able to setup `FortusANT` in manual mode, but i'm not able to set anything via GUI (all buttons bisabled)

`FortiusANT.exe -g -a -m -d 127`

[image]

[logs-3.2.1.zip](https://github.com/WouterJD/FortiusANT/files/5286535/logs-3.2.1.zip)

After downloading an older version (2.6) i was able to start the manual mode an set power, but the settings seems to have no effect.

[image]

[logs-2.6.zip](https://github.com/WouterJD/FortiusANT/files/5286536/logs-2.6.zip)

Since the last time I used the trainer was years ago, I'm not sure if everything works or something is broken. 

I even can't remember if the trainer support "manual" mode without a computer (pressing any button on the T1932 panel without plugged USB seems to have no effect)

Would the driver/the application reconize if the brake works or is this kind of a black box for the driver?

At least i'm not able to change power settings anymore if I unplug power from brake/motor, so it seems that there is a least some communication between brake/motor, T1932 and FortiusANT.

Am I missing something?

WouterJD, 2020-09-26:
Hmmm. As you can see, the "Locate HW" button remsins active; operation without ANTdongle is not expected, although manual mode suggests you could do without.

Will check whether this can be modified...

roggenbrot, 2020-09-27:
Meanwhile I found out while manual mode does not work anymore in the latest version. It was removed (accidentally???) in commit b00ad28

I will debug further and see if I'm able re-implement manual mode again

roggenbrot, 2020-09-27:
Ok, think running without ant dongle in manual mode seems to be unsupported

```
# 2020-05-24    i-Vortex adjustments
#               - in manual mode, ANTdongle must be present as well, so that
#                 manual mode works for i-Vortex as well. 
#                 For USB-trainers perhaps not needed, but FortiusANT needs a
#                   dongle - I would say by name and hence definition!
```

Sorry for the ticket

roggenbrot, 2020-09-27:
I've created a fork and added a new command line option   
```
parser.add_argument('-S','--simulateAnt',  help='Simulated ANT+ connectivity to test trainer',required=False, action='store_true')
```
If set initialize if clsAntDongle will set OK to True

```python
    def __init__(self, DeviceID = None, simulate = False):
        self.DeviceID = DeviceID
        self.simulate = simulate
        self.OK       = self.__GetDongle()
        if self.simulate:
            self.OK   = True
```

and all other methods will return immediately, e.g.

```python
    def Write(self, messages, receive=True, drop=True):
        rtn = []
        if self.simulate:
            return rtn
```

With this option I was able to verify that speed and power values are returned and that the trainer seems to work.

If you want I can submit a pull request for these changes, so that other people also have the chance to check if the trainer is working before buying a ANT dongle.

WouterJD, 2020-09-28:
I have seen your update and it sure touches the point.

I have updated FortiusAntBody.py and would like to hear from you whether that is working.
I have updated in the master branch but did not create an executable yet; just for quick response.

I have changed LocateHW() only; if there is no ANT-dongle and you run in manual-mode (but not i-Vortex) then the function is successful and you should be able to test whether the USB-interface works.

roggenbrot, 2020-09-29:
Thx, already patched this in my fork.

I've also verified your new version and it also works to test the usb interface. Thanks a lot. Since my Ant dongles arrived today, I will hopefully be able to climb mount ventoux today :)

closed 2020-09-29 by roggenbrot

WouterJD, 2020-09-29:
Nice, let us see it on strava - ping me there. Thanks
https://www.strava.com/athletes/2885978

roggenbrot, 2020-09-29:
Unfortunately it turned out that the dongles advertised as Antself are cycplus dongles to the connection is very unstable. Also it seems that I'm also running into the same problem then #102