WouterJD/FortiusANT #109: Fortius iMagic and Chromebook

pimdespreeuw, 2020-09-29:
Hi Wouter,

My father-in-law gave me his old Tacx Fortius iMagic to continue cycling during the rainy autumn and winter days. Unfortunately, I do not have a Windows laptop, but a Chromebook. Is there a way to start Zwifting on the Fortius using this as well?

Thanks in advance!

Cheers,
Tim

WouterJD, 2020-09-29:
Q1: How does iMagic c

pimdespreeuw, 2020-09-29:
Via USB

WouterJD, 2020-09-29:
Reason i asked is the i- in i-Magic (What does it mean?)

Now: install python on Chromebook with required modules. Also install libusb. And check whether ANRdongle is supported.
Where are you going to run Zwift?

WouterJD, 2020-09-29:
If it must run all on the Chromebook, you need three USBports

pimdespreeuw, 2020-09-29:
Sorry, indeed, it is I-Magic (not iMagic).

It's getting quite technical, so I'll have to take some time to figure it all out during the weekend.  

W.r.t. your last question: I intend to use my ipad for Zwift. Will that do the trick?

Thanks already for your help! If there is anything else I should take into account, please let me know :)

WouterJD, 2020-09-29:
If the iPad supports ANT ... yes it could work

closed 2020-10-26 by WouterJD