WouterJD/FortiusANT #111: "GetTrainer - Unable to load firmware" on 1942 head unit

bradyap, 2020-10-01:
First of all, thank you for writing this! When starting the program I get this error: "GetTrainer - Unable to load firmware." I am using a Tacx Fortius, 1942 head unit. Attached are screenshots and a log file. 
[image]
[FortiusANT.2020-10-01 21-35-04.log](https://github.com/WouterJD/FortiusANT/files/5314921/FortiusANT.2020-10-01.21-35-04.log)

WouterJD, 2020-10-02:
You have a head unit that requires firmware to be loaded.
It seems the load of the firmware goes well.

After that the firmware is loaded, ForiusANT waits five seconds and then checks whether the head unit is rebooted.
A device with DeviceID 1942 should be present now.
If that is NOT the case, this error is given.

I do not know why this happens (I do not have a 1942) .
In case that the delay is too short and the head units needs more time to reboot, a restart of FortiusANT would help since then the head unit is initialized and the 1942 would be detected immediatly.

I see you use the windows executable, woould you be able to run the python version, that would be easier to experiemnt with the delay.

-----

On your console you see that the load goes as expected
In the logfile you see the 5 seconds delay.

21:35:05,544: Initialising head unit T1942 (Fortius), please wait 5 seconds
21:35:05,544: C:\Users\brady_p\Documents\FortiusANT\FortiusANT-master\pythoncode\tacxfortius_1942_firmware.hex
21:35:10,662: GetTrainer - Unable to load firmware
21:35:10,662: GetTrainer() returns, trainertype=0xe6be

WouterJD, 2020-10-02:
PS @bradyap would you mind introducing yourself? I'm always curious who I discuss with and where you are located! #14

bradyap, 2020-10-02:
Thanks for the quick response! I changed the delay that you were talking about (at least I think I did) on line 357 of usbTrainer.py from 5 secs to 30 secs, it still did not work. When I run the program, the trainer disconnects as vr interface and reconnects as tacx fortius in device manager. If I restart fortiusant while the trainer is connected as tacx fortius, it immediately says "no tacx trainer found." I have to unplug and replug the trainer so that it appears as vr interface in device manager, upon which the program will give "gettrainer - unable to load firmware" again. Any ideas?

WouterJD, 2020-10-02:
Fortunately you have access to the python code.

The deviceID is 0xe6be 
Then software is loaded
Then 0x1942 is expected (dev = usb.core.find(idVendor=idVendor_Tacx, idProduct=hu1942)

It might be that the trainer comes up with another ID, although I would think that the ID is part of the loaded software.

So for a starter I would be curious to know the deviceID.
Could you add a loop listing the deviceID's?

WouterJD, 2020-10-05:
You manage or need assistance?

WouterJD, 2020-10-08:
Anynrws?

yohann87, 2020-10-09:
Bonjour,
J'ai exactement le même soucis.
[image]
Je n'arrive pas à trouver la solution.
Pouvez-vous m'aider ?
Merci.

yohann87, 2020-10-10:
Bonjour, je pense avoir trouvé la solution en modifiant les Power Management Options de libusbK inf wizard.
Mon interface t1942 est reconnue. La led clignote vert. En appuyant sur cancel l'application se ferme.
Par contre les lignes de la console défilent avec ce message : Tacx returns insufficient data. Est-ce normal?
Merci.

[image]
[image]

WouterJD, 2020-10-15:
Tacx Returns insufficient data:
- either software is polling the headunit too fast; do not use the -A flag (pedal stroke analysis); see docu
- either connections are not good which causes faulty buffers; check wiring

The message CAN appear but should not appear regularly.

WouterJD, 2020-10-15:
"Bonjour, je pense avoir trouvé la solution en modifiant les Power Management Options de libusbK inf wizard."
Does this mean you switched OFF that option?

yohann87, 2020-10-15:
Bonjour,
non j'ai activé cette option.
C'est ce qui a fait fonctionner l'unité T1942.

Pour le message : 
J'ai juste connecté l'unité T1942 au pc pour l'installation.
Le home trainer n'est pas encore branché à l'unité.
J'attends également mes clés ANT+.
Peut-être la cause de ce message?

WouterJD, 2020-10-15:
I have documented your settings, using the screen-print as communicated.

It is likely that this is the reason; the headunit does not report whether all connections are fine or not.
What is the reason that you do not connect the complete trainer?

yohann87, 2020-10-15:
Je ne l'ai pas reconnecté suite au premier soucis d'initialisation.
D'un point de vu pratique, pour trouver le problème.
J'ai trouvé plus pratique de déplacer uniquement l'interface plutôt que le trainer entier  :)  (étages)
Effectivement je n'est pas essayé de tout brancher car je n'est pas reçu mes  clés ANT+ pour finir la configuration.

WouterJD, 2020-10-15:
Avant les ANT+ vous pouvez commencer; utilisant les modes manuelle.

A propos; voulez-vous vous presenter; je suis toujours curieux ou FortiusANT est arriveé

yohann87, 2020-10-15:
très bien je vais essayer.
Merci.

https://www.strava.com/athletes/14062930

yohann87, 2020-10-15:
Je viens d'essayer avec le trainer.
Ca démarre bien le moteur au moment ou je pédale, mais la cadence de pédalage ne fonctionne pas. Par contre il ne s'arrête pas à l'arrêt du pédalage.
Le message apparaît uniquement lorsque j'appuis sur l'une des touches de l'interface t1942.
J'ai testé les fichiers commandes mais ca marque qu'il n'y a pas de fichier nommé "numpy".
Je ne comprends pas trop les fichiers de commandes.

WouterJD, 2020-10-15:
suivez chapitre 3.4 installation python
la commande pip install -r requirements.txt devrait installer numpy

Vous ustilisez Windows ou autre?

msjnaessens, 2020-10-15:
Sorry for responding in English, tried to use fortiusant python on an old windows 8.1 system (unfortunately zwift wont run on this system). And had the same issues.

If you don't check the "install python 3.x to PATH" it wont be able to check the pip and python version and won't recognise the pip command.

My advice is to check whether you have installed python with the PATH option checked, and to use cmd to check your python and pip version. If you installed python 3.9, you are also not able to install numpy. I chose to install python version 3.7.9 instead.
Then, when the older version of python is installed, you should be able to install the requirements via commands in cmd. Hope this helps

yohann87, 2020-10-15:
Merci pour cette information.
C:\Users\maath>py -h
Python Launcher for Windows Version 3.9.150.1013
Je vais donc essayer avec la version 3.7.9

yohann87, 2020-10-15:
C:\Users\maath>py --version
Python 3.7.9

C:\Users\maath>pip install -r requirements.txt
'pip' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

msjnaessens, 2020-10-15:
This is on my windows 10 system the set of commands I had to use:

[image]

yohann87, 2020-10-15:
C:\Users\maath>python --version

C:\Users\maath>pip --version
'pip' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\maath>pip install --upgrade pip
'pip' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

je ne comprends pas pourquoi ca ne fonctionne pas pour moi.

msjnaessens, 2020-10-15:
maybe this will help with your issue, but I think this should have been done automatically when installing python. 

https://www.youtube.com/watch?v=pjCWtppLN3k

yohann87, 2020-10-15:
Microsoft Windows [version 10.0.18362.1139]
(c) 2019 Microsoft Corporation. Tous droits réservés.

C:\Users\maath>pip --version
pip 20.2.3 from c:\users\maath\appdata\local\programs\python\python38\lib\site-packages\pip (python 3.8)

C:\Users\maath>python --version
Python 3.8.6

C:\Users\maath>pip install --upgrade pip
Requirement already up-to-date: pip in c:\users\maath\appdata\local\programs\python\python38\lib\site-packages (20.2.3)

C:\Users\maath>pip install -r requirements.txt
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

ça avance un peu merci.

msjnaessens, 2020-10-15:
select the correct folder in cmd using cd (change directory) command, have to mention to cmd to locate the file in the correct folder. For me, the FortiusANT-master folder is located on my desktop and the requirements.txt file is located in the pythoncode folder within the FortiusANT-master folder. file location is thus: C:\Users\sebas\Desktop\FortiusANT-master\pythoncode

cmd commands:

cd C:\Users\sebas\Desktop\FortiusANT-master\pythoncode

pip install -r requirements.txt

yohann87, 2020-10-15:
j'ai lancé le fichier "install python module".
et tout s'est installé.
Merci beaucoup pour votre aide.

yohann87, 2020-10-15:
merci beaucoup.
j'espère que tout va fonctionner et que je vais pouvoir utiliser zwift quand j'aurais reçu les clés ANT+
[image]

msjnaessens, 2020-10-15:
You're welcome, hope you'll be able to enjoy zwift soon ^^

WouterJD, 2020-10-16:
Hi guys, good work together. Sorry, did not have the chance to react earlier :-) 
@msjnaessens thanks for the advises given, @yohann87 keep going!

WouterJD, 2020-10-16:
> You're welcome, hope you'll be able to enjoy zwift soon ^^

Yes indeed, ping me [on strava](https://www.strava.com/athletes/2885978) and show your training!

yohann87, 2020-10-16:
bonjour,
j'ai essayé de lancer le trainer. Encore une fois il démarre et ne s'arrête pas.
Impossible d'écrire sur la ligne de commande dans la console.
j'ai suivi le manuel mais je n'y arrive pas.
Même avec les fichiers StartUp je ne vais pas plus loin.
Je n'ai toujours pas reçu les clés ANT+.
Désoler pour toutes ces questions mais je n'y connais pas grand chose.

WouterJD, 2020-10-16:
Cést aucune probleme; tu est deja arrivé a un bon point: FortiusANT demarre et parle avec le trainer.

Comment vous lancer FortiusANT?
Quel Startup-fichier vous utilisez?

Si vous aidez une imprime-ecran, ca pourrait aider.
Cette apre midi je suis au MTB, allors la reponse durera

yohann87, 2020-10-16:
je lance FortiusANT avec windows executable.
j'ai essayé en mode admin et c'est pareil.
[image]
le trainer tourne mais ni la cadence de pédalage, ni la puissance, ni la vitesse, ne s'affichent.

j'ai lancé le fichier StartUp FortiusAnt (exe).

[image]

Si j'appuie sur les boutons de l'interface t1942 le message apparaît :  Tacx returns insufficient data.
Après la console me dit d'appuyer sur une touche. Quand je le fais ça ferme tout.

WouterJD, 2020-10-16:
Je repondraibce soir en detail
Point positiv: ca marche!

yohann87, 2020-10-16:
merci pas de soucis.
sportivement.

msjnaessens, 2020-10-16:
To run manual mode, I need at least 1 ant dongle. When there is no ANT+ dongle in my system, speed, cadence, and power stay at 0. Also the wheel starts spinning just like when calibrating, but it never stops (and everything stays at 0). Sounds similar to your problem I think.

Not related, but the other windows executables (gui, gui+manual, etc) give numpy error. When running fortiusant.py from command, I get the same numpy error when using "/fortiusant.py" instead of "python ./fortiusant.py"

yohann87, 2020-10-16:
> Pour exécuter le mode manuel, j'ai besoin d'au moins 1 dongle fourmi. Lorsqu'il n'y a pas de dongle ANT + dans mon système, la vitesse, la cadence et la puissance restent à 0. De plus, la roue commence à tourner comme lors du calibrage, mais elle ne s'arrête jamais (et tout reste à 0). Cela ressemble à votre problème, je pense.

c'est exactement ça.
Donc il faut que j'attende de recevoir mes clés ANT+.

> Non lié, mais les autres exécutables Windows (gui, gui + manuel, etc.) donnent une erreur numpy. Lorsque j'exécute fortiusant.py à partir de la commande, j'obtiens la même erreur numpy lorsque j'utilise "/fortiusant.py" au lieu de "python ./fortiusant.py"

Je n'ai pas d'erreur pour ces actions. Elles se lancent mais je ne sais pas à quoi elles servent réellement.

WouterJD, 2020-10-16:
Calibration (automatic, Tacx Fortius 1932 only) did not work well, which was resolved in the pythond code only; I have compiled and published a new .exe now as well.

numpy has been mentioned in https://github.com/WouterJD/FortiusANT/issues/111#issuecomment-709435905

for paths, see #116

WouterJD, 2020-10-16:
Also, running without ANT dongle should be possible. It was resolved in the pythoncode and should be resolved in the .exe that is distributed today.
Please check and let me know if that would NOT be the case.

So: if using .exe, download the recent version!

WouterJD, 2020-10-16:
> je lance FortiusANT avec windows executable.
> j'ai essayé en mode admin et c'est pareil.
> [image]
> le trainer tourne mais ni la cadence de pédalage, ni la puissance, ni la vitesse, ne s'affichent. [...]

-----------------------------------

Some remarks already made:
1. Download the most recent .exe since some small problems regarding the calibration and 24bytes are solved.
2. You start FortiusANT with "StartUp FortiusAnt (exe).bat"

A BAT file (perhaps you do not see the extension) is a text-file that contains commands for the system.
You can look in it by right-click, edit

"StartUp FortiusAnt (exe).bat" starts FortiusANT without command-line parameters and therefore "best practice settings" are used.

If you look in "FortiusAnt (gui, autostart, manual power).bat" you will see that the settings are explictly specified AND that -m is added for the manual power mode.

Just have a look and try "FortiusAnt (gui, autostart, manual power).bat"

yohann87, 2020-10-16:
Merci pour toutes ces réponses.
je viens de télécharger la dernière version .exe.
j'ai regardé les fichiers .bat et le mode manuel -m. OK
Il me restera à essayer une fois connecté au tacx fortius. (Désolé les enfants sont couchés)
Donc je vous dirai la réponse demain.
Merci pour votre temps et votre aide. ++++

WouterJD, 2020-10-16:
@yohann87 avec plaisir. Soit content que eux dorment, alors c'est pause pour nous quand-même.
J'èspère de voir vôtre excercises a strava bientôt.

yohann87, 2020-10-17:
Salut WouterJD,
Tu es le meilleur. 
Petit essai en mode manuel.
Pédalage à la main uniquement pour essayer.
[image]

yohann87, 2020-10-20:
Bonjour,
voila le résultat avec les clés ANT+: (pas encore essayer de rouler sur zwift mais ça fonctionne)
J'ai fait le calibrage du home trainer.
Demain essai de roulage complet sur zwift.
Merci encore pour tout.

[image]

WouterJD, 2020-10-20:
@yohann87 je suis content que ca marche.
Alors je ferme cette Issue, poser des question est naturellement toujours possible.
Curieux de te voir a Strava!

closed 2020-10-20 by WouterJD