WouterJD/FortiusANT #112: Supported Tacx trainers

Jobson1980, 2020-10-07:
Hoi Wouter,

Ik ben zeer geïnteresseerd om jou software te gebruiken. Ik heb nog geen tacx trainer. Maar het is mij niet duidelijk waar ik op moet zoeken als ik op marktplaats een geschikte trainer wil vinden. 
In de manual noem je vooral de headunit types die ondersteund worden. Maar dat zoekt niet makkelijk.
Welke tacx types zijn geschikt? iMagic, Fortius, nog meer?

gr,
Job

WouterJD, 2020-10-07:
Hoi Job, leuk je te zien op github.

Ik heb zelf een Tacx Fortius met de witte headunit met blauwe ring; dus dat werkt in ieder geval.
De Tacx Fortius is een prima trainer en te koop voor 100-200 euro; een beetje wat je er voor over hebt.
Ik zou de verkoper vooral vragen of alles heel is en goed werkt.

Vervolgens heb je een PC nodig met twee ANTdongles; één voor FortiusANT en de andere voor Zwift/TrainerRoad/Rouvy oid
Koop geen CYCLUS ANSELF dongels, teveel problemen!

Ik zie er al twee; of die het doen weet ik natuurlijk niet.
https://link.marktplaats.nl/m1581333598?utm_source=ios_social&utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios
https://link.marktplaats.nl/m1607319470?utm_source=ios_social&utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios

Succes!

Jobson1980, 2020-10-07:
Hoi,

Die had ik ook gezien ja.

Ik snap het principe met de dongles. Het ging mij er meer om welke modellen Tacx er precies mee kunnen werken.
Dus Fortius iig. Maar nog andere modellen waar ik op kan zoeken?

 

https://link.marktplaats.nl/m1581333598?utm_source=ios_social <https://link.marktplaats.nl/m1581333598?utm_source=ios_social&utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios> &utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios
https://link.marktplaats.nl/m1607319470?utm_source=ios_social <https://link.marktplaats.nl/m1607319470?utm_source=ios_social&utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios> &utm_medium=social&utm_campaign=socialbuttons&utm_content=app_ios

WouterJD, 2020-10-07:
FortiusANT werkt met alle USB connected trainers en iVortex. Fortius is denk ik wel preferred, er zijn vergelijkbare trainers. Ik heb geen vergelijkingsinfo.
Moderne trainers (direct drive) zijn beter maar ook duurder.

WouterJD, 2020-10-07:
Zie ook: https://github.com/totalreverse/ttyT1941/wiki#tacx-products

Jobson1980, 2020-10-07:
Ik zag deze nog: Tacx Flow fietstrainer/PC-interface/Stuur-unit/Software
€ 150,00
https://link.marktplaats.nl/m1608001903?utm_source=android_social&utm_content=vip&utm_medium=android_social&utm_campaign=socialbuttons

Lijkt mij ook compatible.

In de link die jij mij stuurde met een hardware overzicht staat ook een
tabel "Power limits"
https://github.com/totalreverse/ttyT1941/wiki#power-limits

Wat wordt er bedoeld met de kolom "Incline (%)"?
Bij een aantal staat daar niks ingevuld. Heeft dat met de helling simulatie
te maken?

WouterJD, 2020-10-07:
9 personen hebben gemeld dat ze met een Flow werken; dus dat lijkt inderdaad te werken.

Er zijn trainers (met een headunit) waarop je vermogen en/of hellingshoek kunt instellen; blijkbaar is daarvoor een limiet.

De Fortius wordt door de software aangestuurd door middel van weerstand, die berekend wordt aan de hand van vermogen of helling; bij de Fortius wordt een maximum van 1000Watt aangegeven. Dat betekent uiteraard ook een maximum helling/snelheid; maar  waarschijnlijk is dat een grens die jij niet bereikt.

Jobson1980, 2020-10-07:
Top! Ik laat je weten als ik er verder mee ben gekomen.

Jobson1980, 2020-10-08:
Ik kan een Tacx Flow T1901 met T1932 headunit kopen

En ik kan een Tacx Fortius T1941 met T1932 headunit kopen.

Weet jij of er nog wezenlijk verschil zit tussen die beide Tacx modellen? Zo ja, welke is beter?

WouterJD, 2020-10-08:
Ik denk dat een Fortius met het frame robuuster is en dat de fiets er steviger op staat
Een Flow lijkt handzamer

Jobson1980, 2020-10-08:
De flow heeft ook een frame: https://link.marktplaats.nl/m1608293072

En er wordt zelfs een wiel met cassette en trainingsband bij gedaan. 

 

De flow heeft een electro-magnetische rem, en de Fortius een motor rem zo te lezen. Maakt dat nog verschil?

WouterJD, 2020-10-08:
Dat kan je het beste bij TotalReverse bestuderen; ik heb daar geen detail kennis van.
Gezien het daar gemelde vermogens verschil lijkt de Fortius wat zwaarder, maar pin me er niet op vast; ik heb maar één trainer (de Fortius).

Jobson1980, 2020-10-10:
Ik heb inmiddels de Fortius in huis. De libusb driver is geïnstalleerd en
de Tacx wordt herkend door FortiusAnt. Verder valt er nog niet veel te
testen/doen totdat ik de Ant+ dongles in huis heb toch?

WouterJD, 2020-10-10:
Toch wel; in manual mode kan je trainen.
-m voor manual power
-M voor manual grade

Jobson1980, 2020-10-11:
Ik had gisteren de cadans sensor nog niet aangesloten. Nu wel. Opgestart
met -m -M.
Toen ik 1x met het pedaal rond ging, begon het wiel hard te draaien en
stopte niet meer. De knoppen op de headunit deden niks. Ik heb hem maar
uitgezet. Moet nu weg, maar kijk vanmiddag nog wel een x

Jobson1980, 2020-10-11:
Ik begin een klein beetje te begrijpen hoe het werkt. Ik heb ook logfiles gemaakt om te kijken of mijn cadence sensor werkt. Dat doet hij.
Klopt het dat ik zonder dongle's verder nog niet kan kalibreren? In de gui zie ik verder ook geen meters uitslaan. Alles blijft op 0 als ik trap... 
In de logfile zie ik wel dat er een Candence is, en de Speed is rond de 20.. Maar ik heb het idee dat ik er nu verder weinig uit kan halen

WouterJD, 2020-10-15:
Je kunt zonder dongle werken met de -m of -M flag; daarmee kan je handmatif de trainer sturen met power of grade

Jobson1980, 2020-10-15:
Ik heb inmiddels dongles. Ik had eerst die cycplus dingen. Dacht andere te
hebben besteld maar dat bleken toch cycplus. Ze raakten continu de
verbinding kwijt. Ik heb inmiddels dongles van Xand en die werken
uitstekend. Alles loopt prima dus! Nog bedankt voor al je antwoorden en
uiteraard voor het bouwen van de software!

closed 2020-10-26 by WouterJD