WouterJD/FortiusANT #113: Cadans meter + connectie naar Zwift

LitaliaFC, 2020-10-07:
Goedenavond Wouter,

Dank voor je voorgaande hulp; ik heb de TacX inmiddels gekoppeld gekregen aan Fortius, echter krijg ik geen cadans output, deze blijft op 0. Moet ik hiervoor nog iets bijzonders doen/aansluiten? Fortius laat wel de Power en Speed zien. Ik heb nu twee ANT+ usb's in mijn laptop, echter lukt het mij niet om Fortius te koppelen aan Zwift. Het programma vind geen ANT+ in 'Paired Devices' scherm. Het zou kunnen dat ik een stap gemist heb, weet je misschien waar dit aan kan liggen?

Gr Livio

WouterJD, 2020-10-07:
Er loopt één USB kabel naar FortiusANT en dat interface lever Power Snelheid en Cadans. Geen cadans betekent dat er bij de Tacx iets los zit of kapot is. Goed controleren. Magneet aan pedaal?

WouterJD, 2020-10-07:
Ik ga er vanuit dat FortiusANT gestart is, dat je fietst èn dat je snelheid/power ziet. Dan zou Zwift FortiusANT moeten zien.

Ken je Zwift en weet je hoe het met andere trainers werkt?
Stuur anders wst foto's.

LitaliaFC, 2020-10-23:
Goedemorgen Wouter, excuus voor de late reactie. Ik vermoed dat mijn 'magneet' voor de meting van de cadans niet meer werkt. Volgens de instructie heb ik deze correct gemonteerd (zie foto). [image]

Wat betreft de connectie met Zwift, dit is inmiddels gelukt. Fortius koppelt de trainer met Zwift, echter heb ik nog geen betere ANT+ dan die 2 van Cycplus. In Fortius zie ik dat de ANT+ regelmatig de verbinding verliest. Zodra ik een ANT+ van een ander merk heb, hoop ik dat dit probleem verholpen is. Elke keer dat de verbinding met de ANT+ wordt verbroken, lijkt het alsof de resistance flink veranderd waardoor het nu nog niet mogelijk is de trainer te gebruiken. De bestelling voor een andere ANT+ is gedaan..

Frenske8, 2020-10-23:
Misschien het magneetje op crank nabij de pedaal vergeten ? Ik zie die niet op de foto.

WouterJD, 2020-10-23:
De foto toont de cadans sensor. Die wordt geactiveerd door een magneetje op het uiteinde van de cranck. Goed gezien @Frenske8 !

WouterJD, 2020-10-30:
Laat je even weten of e.e.a. werkt; dan kan dit issue gesloten worden

LitaliaFC, 2020-10-31:
Wat mij betreft kan deze gesloten worden; ik heb connectie met Zwift, echter laat de Ant+ levering wat op zich wachten. Wat betreft de magneet, die zal waarschijnlijk bij de vorige eigenaar zijn verloren gegaan, maar ik zoek nog even door de doos. Bedankt beiden voor de hulp

closed 2020-10-31 by WouterJD

torpedox88, 2020-11-04:
Do I understand correctly from the previous post that CTP's can already couple with FortiusANT without Ant+ dongles?

WouterJD, 2020-11-04:
No that is not correct; to communicate with CTP an ANT-dongle is required.
Manual mode can be done without

torpedox88, 2020-11-04:
Thanks for the quick reply Wouter! 

It is also not possible to only "read" data from FortiusANT? 
Reason for asking is that I'm thinking about running TrainingPeaks for instance to record all data while running FortiusANT in manual/grade mode (or is there maybe another way to record all FortiusANT data?)

WouterJD, 2020-11-04:
Always handy to create a separate issue for a separate request😉
You mean: create tcx export of excersize? If you create it, I'm happy to integrate.

torpedox88, 2020-11-04:
Hmmm you're right, let's do so if anyone is going to work on implementing this.

Indeed something which records and export parameters like speed, cadence and power as a function of time.
I checked the .tcx format which might already be too detailed, it seems that a simple .csv file will do the job as well.
Is it maybe possible to add an option do the -d function which does exactly this? (I do have little programming experience, but not enough to make something like this I'm afraid...)

WouterJD, 2020-11-04:
😉

WouterJD, 2020-11-05:
TCX-export is implemented; please test.
No special flags required; in manual mode an export is created.