WouterJD/FortiusANT #114: Tacx fortius downhill

Thijzelt, 2020-10-08:
First things first, thank you for all the hard work!
I like the program a lot, i use it 3 days a week, i am glad that we can use the old Tacx with zwift and bkool.

The only thing is i use the fortius in the bedroom and it can be very noisy. especially downhill when the motorbrake is driving the rearwheel, without me having any cadence. Can i use a parameter to turn off driving the motor on a download and just use the resistance uphill. If it is a dumb thing to ask please say so, i could not find it in the manual.

WouterJD, 2020-10-08:
Hi, where are you from? Always curious where the software is used😉

So I understand you are happy with how FortiusANT works, but you do not want negative power. What minimum would you suggest? ==> When required power drops below, it's minimized there?

Thijzelt, 2020-10-08:
I live in the Netherlands, close by.

The minimum is hard to tell, when the slope is below 0% the motor is running the rearwheel up to 60 km/h without me giving any pedaling watts. Is it just possible to use resistance only by slope uphill. And downwill 0 resistance, and that it does not drive the motorbrake.

Thank you for the fast reply

WouterJD, 2020-10-08:
So you would like a minimum slope of 0% = uphill OR flat

WouterJD, 2020-10-09:
Download usbTrainer.py and FortiusAntCommand.py; .exe is not created
Try -u command line variable
Let me know whether this is what you want.

Thijzelt, 2020-10-19:
It works! The -u parameter works, it is flat or uphill without the motorbrake speeding to 60 kph
I am so happy that it does!
My kids can sleep again!

And i live in Eibergen nearby Enschede, is that closeby?

Keep up the good work!

WouterJD, 2020-10-19:
Hi @Thijzelt . Eibergen is only 80km from Elst near Arnhem. So yes that's close.

Nice that it works and thanks for feedback.

WouterJD, 2020-10-19:
.exe was published last week. So issue can be closed. If there is anything else just let me know

closed 2020-10-19 by WouterJD