WouterJD/FortiusANT #115: FortiusAnt on iMac

MrMR-G, 2020-10-14:
Hi,

I am from Scotland and looking to set up FortiusAnt to use with zwift. I have the T1932 head unit and fortius multiplayer trainer. 

I've not done any coding before but have am happy to try and learn the basics required to get this up and running. 
I've read the manual but not quite sure where to begin with trying to load it on an iMac. I've downloaded Python3 and have a couple ANT+ dongles coming that should be here in a few days. 

Any help would be appreciated.

Thanks, Matt.

WouterJD, 2020-10-15:
Hi @MrMR-G always curious who I work with and where FortiusANT travelled to, so tell some more on yourselves!

#45 shows that FortiusANT works on MacOS; I did not yet get much info on what steps to take to get it installed.
I think the manual will guide you through the basics; it's all I have today.

Basics:
- install python
- install libusb (if you could register the steps I'd be happy to add to the manual)
- install ANT+ which is quite straight forward

MrMR-G, 2020-10-15:
Yeah I did read through that and see that it managed to work, that's where I found what dongles to order. I've managed to download Python3 and installed libusb through Brew but can't seem to start the FortiusAnt? I think I'm just being stupid but not sure how to execute it on Mac?

WouterJD, 2020-10-16:
Well, I do not believe you are stupid. Done to much to disqualify yourself like that!

Please be more specific what your system says; what error is given.
You're close!

MrMR-G, 2020-10-16:
My dongles arrived today and I've managed to figure out loading up FortiusAnt and the GUI. I ordered the anself dongles from amazon but they are actually CYPLUS dongles when loaded. It seems to be running and getting readings of the motor and cadence but I keep getting the Dongle not available, try to reconnect after 1 sec, dongle reconnecting. Then it drops out and repeats. Is this just a case of getting new dongles that are not CYPLUS ones? I did read through #65 but don't think there is a solution to this issue?

WouterJD, 2020-10-16:
I do not want to be negative on products because I do not have indept knowledge why these dongles don't work well.
Perhaps it's FortiusANT doing something unexpected triggering the issue.

BUT ... Anself CYCPLUS dongles drop the connection and those who have investigated the subject did not find a solution better than what has been implemented.

I do not have a solution.

MrMR-G, 2020-10-17:
Yeah I have ordered some other dongles so hopefully be up and running with them soon! It all seemed to be running correctly apart from the dongles.

MrMR-G, 2020-10-21:
Okay so the new dongles have arrived and they seem to be working correctly. I'm having issues with the running of the program. It starts calibrating and turns the rear wheel to 20km/h and appears to finish the calibration. When I open zwift it reads all the inputs from the dongle and when I start pedalling seems to give correct readings but after every few seconds the motor brake cuts in for a second or so then this just repeats?

WouterJD, 2020-10-21:
Well this needs some investigating.

Create logfile with-d127 option ...  I will look into it when there's some time. Hope soon😉

MrMR-G, 2020-10-21:
Okay, I'm just out just now but I'll run that when I get home.
I don't know if it's related at all but when trying to do the runoff I couldn't get any more than 3.5seconds no matter how tight or loose I had the roller on the tyre?

WouterJD, 2020-10-21:
The logfile will help👍

MrMR-G, 2020-10-21:
[FortiusANT.2020-10-21 18-30-06.log](https://github.com/WouterJD/FortiusANT/files/5418081/FortiusANT.2020-10-21.18-30-06.log)

This is the log file from a start up, calibration then just cycling, the motor brake keeps cutting in every few seconds then goes back to normal and just repeats.

WouterJD, 2020-10-22:
Hi Matt

I have looked into your logfile; extracted the trainer-communication from it. See attachment.
The trainer reports the speed which at a moment drops to zero.
If there is a zero-speed, the resistance would be infinite to achieve a required power.
To avoid that, resistance is set to zero as well.

So it seems that your trainer reports incorrect speed; which may be a hardware failure.
Does your trainer work well with Tacx-software?

I also see your trainer always reports cadence = 0, which is odd. This may be a loose contact.
Now and then, PE (pedalecho) is not zero (4); which is the signal that the pedal passes the magnet. But cadence remains zero.

So please verify hardware and check with Tacx software for proper functioning.
If I oversee something, please let me know.

```
18:30:42,634: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=7.2 TargetRes=4965 CurrentRes=897 CurrentPower=15, pe=0 Connected to Tacx Trainer T1932
18:30:42,888: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=6.9 TargetRes=5919 CurrentRes=960 CurrentPower=15, pe=0 Connected to Tacx Trainer T1932
18:30:43,146: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=5.9 TargetRes=3570 CurrentRes=876 CurrentPower=11, pe=0 Connected to Tacx Trainer T1932
18:30:43,401: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=7.3 TargetRes=3270 CurrentRes=845 CurrentPower=14, pe=0 Connected to Tacx Trainer T1932
18:30:43,647: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=5.7 TargetRes=5868 CurrentRes=855 CurrentPower=11, pe=0 Connected to Tacx Trainer T1932
18:30:43,899: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=7.4 TargetRes=3210 CurrentRes=797 CurrentPower=13, pe=0 Connected to Tacx Trainer T1932
18:30:44,155: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=5.6 TargetRes=5786 CurrentRes=896 CurrentPower=11, pe=0 Connected to Tacx Trainer T1932
18:30:44,404: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=7.3 TargetRes=3180 CurrentRes=734 CurrentPower=12, pe=0 Connected to Tacx Trainer T1932
18:30:44,659: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=6.0 TargetRes=5876 CurrentRes=907 CurrentPower=12, pe=0 Connected to Tacx Trainer T1932
18:30:44,910: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=5.7 TargetRes=3300 CurrentRes=844 CurrentPower=11, pe=4 Connected to Tacx Trainer T1932
18:30:45,163: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=4.3 TargetRes=3210 CurrentRes=844 CurrentPower=8, pe=4 Connected to Tacx Trainer T1932
18:30:45,411: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=2790 CurrentRes=0 CurrentPower=0, pe=0 Connected to Tacx Trainer T1932

18:30:45,060: clsTacxTrainer.Refresh(False, 2)
18:30:45,060: clsTacxNewUsbTrainer._ReceiveFromTrainer()
18:30:45,163: Trainer recv data="array('B', [3, 136, 3, 0, 0, 2, 0, 0, 10, 0, 0, 0, 0, 0, 0, 0, 0, 0, 58, 4, 0, 0, 0, 0, 3, 19, 2, 0, 53, 145, 0, 0, 11, 5, 40, 105, 0, 0, 76, 3, 228, 12, 4, 0, 0, 0, 2, 188, 75, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
18:30:45,163: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=4.3 TargetRes=3210 CurrentRes=844 CurrentPower=8, pe=4 Connected to Tacx Trainer T1932
18:30:45,163: clsTacxNewUsbTrainer.TargetPower2Resistance(100, 1291) = 2790
18:30:45,163: clsTacxUsbTrainer.SendToTrainer(T=2, M=1, P=100, G=0, R=2790, W=85, PE=4, S=4.3, C=0)
18:30:45,163: Trainer send data="01 08 01 00 e6 0a 04 00 02 0a 10 04" (len=12)
18:30:45,163:                   tacx mode=2 target=2790 pe=4 weight=10 cal=1040

The trainer reports Cadence=0 Speed=4.3
The command to the trainer is R=2790

18:30:45,163: clsTacxUsbTrainer.Refresh()
18:30:45,163: mp-MainDataToGUI(19597, (4.3, 0, 8, 1, 100, 0, 2790, 0, 15))
18:30:45,310: Sleep(0.15) to fill 0.25 seconds done.
18:30:45,310: clsTacxTrainer.Refresh(True, 2)
18:30:45,311: clsTacxNewUsbTrainer._ReceiveFromTrainer()
18:30:45,411: Trainer recv data="array('B', [3, 136, 3, 0, 0, 2, 0, 0, 10, 0, 0, 0, 0, 0, 0, 0, 0, 0, 58, 4, 0, 0, 0, 0, 3, 19, 2, 0, 53, 145, 0, 0, 0, 0, 64, 105, 0, 0, 0, 0, 138, 12, 0, 0, 0, 0, 2, 171, 76, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0])" (len=64)
18:30:45,411: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=0.0 TargetRes=2790 CurrentRes=0 CurrentPower=0, pe=0 Connected to Tacx Trainer T1932
18:30:45,411: clsTacxNewUsbTrainer.TargetPower2Resistance(100, 0) = 0
18:30:45,411: clsTacxUsbTrainer.SendToTrainer(T=2, M=1, P=100, G=0, R=0, W=85, PE=0, S=0.0, C=0)
18:30:45,411: Trainer send data="01 08 01 00 10 04 00 00 02 0a 10 04" (len=12)
18:30:45,411:                   tacx mode=2 target=0 pe=0 weight=10 cal=1040

The trainer reports Cadence=0 Speed=0.0
The command to the trainer is R=0
```

[FortiusANT.2020-10-21.18-30-06.txt](https://github.com/WouterJD/FortiusANT/files/5421251/FortiusANT.2020-10-21.18-30-06.txt)

MrMR-G, 2020-10-23:
I had a look and the cadence sensor was adjusted wrong and that was giving the zero cadence and the motor brake is the fortius way of letting you know there is no cadence. 
All up and running properly now, thanks for all your help!

eeraserhead, 2020-10-25:
Hi all!

I'm from Canada and trying to use zwift with a tacx i-vortex (t1960) w/ a t1962 head. I want to run it from my macbook pro mid-2014 (Catalina 10.15.6). I do have an ANT+ usb dongle. I have no background in programming, but I'm an avid learner and happy to help if I can.

I manage to install brew and libusb, but I'm at lost when it comes to run FortiusANT. Any tips on where to go from now?

Thanks!

WouterJD, 2020-10-26:
Hi @eeraserhead . The ANT-dongle can be used by Garmin Connect or Zwift; check that that software is NOT running.
Note that each program on a PC uses it's own dongle and then can communicate to many devices.

So for FortiusANT communicating with Zwift, two dongles are required - unless Zwift is running on another system(dongle there) or smartphone (which is ANT aware)

WouterJD, 2020-10-26:
> All up and running properly now, thanks for all your help!

@MrMR-G good work; see you on Strava!

eeraserhead, 2020-10-26:
Thanks for your reply, @WouterJD 

As I read further on, I realised my ANT+ dongle was not supported by FortiusANT (ID 4096),
I will order new ones and test again.

WouterJD, 2020-10-27:
Hi @MrMR-G; and yes [it works](https://www.strava.com/activities/4247289115) :-)

WouterJD, 2020-10-27:
> As I read further on, I realised my ANT+ dongle was not supported by FortiusANT (ID 4096),
> I will order new ones and test again.

Adding a supported "dongle ID" is a no-brainer; assuming the ANTdongle actually is compatible.
See: `antDongle.py` function `__GetDongle()`  code `dongles = { (4104, "Suunto"), (4105, "Garmin"), (4100, "Older") }`

eeraserhead, 2020-10-28:
So apparently I was wrong, and my Dongle is indeed a ID 4100.  (see log)

In the meantime, I received the other dongle I ordered. I changed my setup a bit, because I was unsure how to run libusb on mac. 

I installed FortiusANT on another PC running windows, and I run Zwift from my Macbook, using a ANT+ dongle there as well (so two dongles, 1 on windows and 1 on mac). What am I doing wrong?

After running ExploreANT, I got this log.
[ExplorANT.2020-10-26 17-43-49.log](https://github.com/WouterJD/FortiusANT/files/5454220/ExplorANT.2020-10-26.17-43-49.log)

WouterJD, 2020-10-28:
May be the case.
Now you have a working environment, switch the dongles and see what happens. If the vortex is not visible, it's clearly the dongle.

Success

eeraserhead, 2020-10-28:
Right, no vortex found when switching. So close, yet so far...!
I will order another dongle, and see what happens.
I keep you updated.

Thank you so much

WouterJD, 2020-10-28:
Discard the bl***dy thing. So it will not popup again causing lots of energy.
Like to hear from you; see you on strava

WouterJD, 2020-11-04:
Hi @eeraserhead how is delivery of Antdongle proceeding?

eeraserhead, 2020-11-04:
Hi!

I received the dongle yesterday and did a quick test just to see if it work. I ran FortiusANT from my PC and Zwift from my macbook pro. Everything seemed fine for about 5 minutes, but then it starded disconnecting every few seconds. I restarted everything, and I couldn't get a connection anymore after that. So I'm a bit confused.

I noticed that the first time I started FortiusANT, it paired rather quickly to both the trainer unit and the trainer head. But then when I tried to reconnect it after starting Zwift, it would only connect to the head or to the unit.

I will have time to investigate a bit more this week-end. If I find a solution in the meantime, I will uptade you.
If you have any clue, I can do a quick test tomorrow morning too.

WouterJD, 2020-11-04:
Question 1: what dongle do you use (ref #61)

Regarding the speed of pairing; I have not understood untill now what determines the speed. Sometimes it takes long, other times it does not. i-Vortex can take minutes! Note that the Vortex trainer must be paired first before the headunit can be paired (the trainer provides the code for the headunit).

eeraserhead, 2020-11-04:
Dongle info:

manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)

From [Amazon](https://www.amazon.ca/-/fr/PerfPRO-CycleOps-Virtual-Trainer-TrainerRoad/dp/B01MXXDBNK/ref=pd_rhf_sc_p_img_1?_encoding=UTF8&psc=1&refRID=QNGK2X9Y20B1DANMCCK0)

WouterJD, 2020-11-04:
pfff...not ANSELF :-)

eeraserhead, 2020-11-04:
But those takes a month to ship in Canada :(

Well I guess if that's the only issue, I guess it's not too bad...

WouterJD, 2020-11-04:
My experience today:
- two laptops next to each other; both have the dongles on the left side: intermittent connection issues
- both laptops, dongles facing: no issues and faster pairing

Perhaps there should not be electrical equipment between the dongles; not that ANT is low-energy protocol expected to be used on body (HRM) or bike (SCS)

Check it out ...

eeraserhead, 2020-11-05:
It's alive!!

So I changed the setup a bit, since it was complicated to put the two computers aside + the trainer. 

In fact, I just plugged in the 2 dongles in the macbook. I tough it wouldn't work because I never manage to run libusb on mac, but FortiusANT and Zwift could talk anyway. The connection was super quick. I tested it for a few minutes, and it was pretty stable.

Now, does Zwift can talk to my trainer and actually adjust the Watt, or do I have to use the trainer head?

I will do a FTP test this week-end to see if everything works fine, but so far so good.

Thanks for your help!

WouterJD, 2020-11-05:
If you use a non-USB trainer, you would not need libusb. But perhaps the software is dependant of it; something I could remove in case of i-Vortex trainers. Does FortiusANT work without libusb installed?

WouterJD, 2020-11-05:
> but FortiusANT and Zwift could talk anyway

and

> Now, does Zwift can talk to my trainer and actually adjust the Watt, or do I have to use the trainer head?

I do not understand your question, please explain.

eeraserhead, 2020-11-05:
> 
> 
> @eeraserhead; you need libusb to interface with ANT and USB. Please explain `I tough it wouldn't work because I never manage to run libusb on mac`; it appears your system is running.

This is where my lack of computer skills start to surface, but I tought that I had to overwrite the usb driver on macOS like I did on my Windows OS with libusb. (From User Manual; 3.5.1 Windows - install option 2). I did install libusb on my mac, but that's it. I did not go through the same steps as ''install option 2'' because I didn't know how on macOS. But it appears as if I don't need to.

> > Now, does Zwift can talk to my trainer and actually adjust the Watt, or do I have to use the trainer head?
> 
> I do not understand your question, please explain.

Will the trainer adjust its resistance automatically according to the input from Zwift (or any CTP)? Or do I change the resistance myself during workout?

WouterJD, 2020-11-05:
> Will the trainer adjust its resistance automatically according to the input from Zwift (or any CTP)? Or do I change the resistance myself during workout?

Yes it will.
The CTP can operate in two modes: PowerMode or TrackMode. 
Zwift usually works in TrackMode and sends the slope to the trainer, who will then adjust the resistance according to the grade: heavy for +20% and ligt for -10%. The faster you cycle, the more power you have to produce.

During ERGmode training, the CTP operates in PowerMode and sends the required power to the trainer, who will then adjust the resistance to the speed you are cycling, resulting in a constant power. (Usually the CTP will execute a program, varying the power in structured blocks).

eeraserhead, 2020-11-06:
So, I did a ramp test yesterday to test the setup further and the resistance on the trainer seems off. 

I see that the power value (w) change on FortiusANT GUI following the given power on Zwift (ERGmode), but I only feel quick bursts of added resistance, then it drops. 

Also, I notices that the trainer doesn't calibrate when I hit Start in FortiusANT (after I turned the pedal).

I tested the trainer alone, and the resistance works fine.

WouterJD, 2020-11-06:
Please provide debug file (-d 127) for further investigation

eeraserhead, 2020-11-06:
[FortiusANT.2020-11-06 21-21-32.log](https://github.com/WouterJD/FortiusANT/files/5503626/FortiusANT.2020-11-06.21-21-32.log)

Same setting, with ramp test.

WouterJD, 2020-11-10:
> Also, I notices that the trainer doesn't calibrate when I hit Start in FortiusANT (after I turned the pedal).

Calibration is only supported on Tacx Fortius with T1932 headunit

WouterJD, 2020-11-11:
I have the impression that the power received from FE-C (Zwift?) is correctly transmitted to the i-Vortex
See attached excel sheet analysing the logfile
[#115 FortiusANT.2020-11-06.21-21-32.xlsx](https://github.com/WouterJD/FortiusANT/files/5523488/115.FortiusANT.2020-11-06.21-21-32.xlsx)

eeraserhead, 2020-11-11:
Thanks for going through the logfile.

In the meantime, I did some new tests, all using Zwift w/ structured training in ERGmode:
First, I added a cadence sensor to be sure that there was no drop, but it didn't change anything.
Second, I tried to run FortiusANT on manual power (-m), but still, there were drops in resistance.
Third, I ran FortiusANT adding the -P command line (-a -g -H0 -t i-Vortex -n -m -P)

That last attempt seems to work, I was able to do a full training in Zwift without any drops at all! 
I guess we could say that the issue is closed, until further bug!

I can list the steps I went through to run the i-vortex w/ FortiusANT on mac if I can do anything to help the project.

WouterJD, 2020-11-13:
Hey, nice idea.

Do you have TWO CTP's running simultaneously? Perhaps even two Zwift installations?

If one ZWIFT sends Power-commands and the other sends Grade-commands, the ttrainer will alternately doing what is being requested. The -P flags prevails power.

Please check

eeraserhead, 2020-11-13:
Did another training yesterday with no problem.

After double checking, there's only one Zwift installation, and no other CPT.

It seems like FortiusANT try to read both Power and Grade input at the same time when -P is not specified, even though ERGmode is active within Zwift.

WouterJD, 2020-11-15:
Hi all. Version 3.6 has a new logging feature -d32 to create a JSON file for this kind of analysis.
Please let me know whether that helps to investigate.

WouterJD, 2020-11-16:
> It seems like FortiusANT try to read both Power and Grade input at the same time when -P is not specified, even though ERGmode is active within Zwift

A logfile -d127 (.log + .json) would be helpful; of course without the -P to reproduce the issue.

WouterJD, 2020-11-18:
Issue closed, appears not to be a power curve issue

closed 2020-11-18 by WouterJD