WouterJD/FortiusANT #117: Support for Busido trainers (pre-smart)

mac2444, 2020-10-17:
Hi, 

Any chance this will work with the pre-smart Bushido trainers?  I've been using antifier with my old i-flow but have a chance to get a Bushido from a friend.  Model number is T1981, it is headless but works fine on TTS3/4  Thanks a lot!

WouterJD, 2020-10-17:
How does bushido connect to TTS3/4
USB or wireless

mac2444, 2020-10-17:
The Bushidos are all wireless.   The pre-smart versions (T1980, T1981) use the proprietary ANT+ to connect to TTS.   There is also a wireless head unit but this is optional as the Bushido could pair directly to the computer.

WouterJD, 2020-10-17:
i-Vortex has a head unit which communicates wirelessly with the computer.
The head unit has buttons so you can use it manually.

If it is i-Vortex compatible it may work.

mac2444, 2020-10-18:
I think the Vortex and the Bushido are quite similar.  Is there already a way to run a Vortex and a power curve for it?

WouterJD, 2020-10-18:
-t i-Vortex see manual
If more info needed, lrt me know. Description is not very extensive

mac2444, 2020-10-18:
I read that. I'll try it out when the trainer arrives and let you know how it works.  Thanks!

WouterJD, 2020-10-19:
HI @mac2444; just keep me posted what happens.
In the meantime I'm always curious where FortiusANT landed; please introduce yourself and let me know where you are located!

mac2444, 2020-10-19:
Wouter, I'm in Calgary, Canada.  The weather here has already taking a towards the winter ahead so obviously bike trainers enter the mind.  With all the people forced to work from home or out of work all together, bikes and trainers have been sold out completely here for months. That's what got me digging into options and helped me find your site.

WouterJD, 2020-10-19:
Hi "Mac"; great that FortiusANT has found its way to sporty Calgary. I'm from the Netherlands and winter has started here as well, combined with a semi-lockdown. So: tacx-time; although we can (mtb)bike all year since there is never much snow.

Greetz Wouter

Muleguy-Security, 2020-10-27:
Hello Wouter! I too may be able to assist with Bushido testing. I have the same unit from the sounds of it. I will download and give it a try. 

P.S. Writing from Ottawa, Ontario in Canada.

Muleguy-Security, 2020-10-27:
@mac2444 - I think we are going to have to work through some of the same steps described in Issue 101 - [https://github.com/WouterJD/FortiusANT/issues/101](url) to try to determine whether either the brake or the headunit can be seen and whether it takes the same commands as i-vortex headunit or brake does.

mac2444, 2020-10-27:
Sadly, the trainer I was going to get was sold.  I have a friend with the same one, I'm going to borrow it off him when I can and see if it will show up on the program. I'll let you know what happens.

WouterJD, 2020-10-27:
Plenty 2nd hand @ marktplaats.nl. may be similar in other countries.

mac2444, 2020-10-27:
Not here, I'm afraid.  With the early cold weather, trainers are sold out in all the stores until next year and used ones are gone the same day they are listed.

WouterJD, 2020-10-28:
All; indeed please follow #101 and check https://github.com/fluxoid-org/CyclismoProject/wiki/Tacx-Bushido for interface.

WouterJD, 2020-10-28:
> Not here, I'm afraid. With the early cold weather, trainers are sold out in all the stores until next year and used ones are gone the same day they are listed.

If thinking global:
https://link.marktplaats.nl/m1617341693 Tacx Fortius EUR 100,= Shipping worldwide EUR 100,=

mac2444, 2020-10-28:
I already have an i-flow and an i-magic trainer, so that will keep me going till the new ones are available again.  I was just hoping to upgrade to the bushido as the price was very good on it. Thanks for looking though

Marmi86, 2020-11-08:
Hello,
Can I  use the Bushido with the ivortes setting -t I-Vortex?

WouterJD, 2020-11-09:
No, since Bushido uses different codes. Please follow this thread for developments.

WouterJD, 2020-11-09:
@Marmi86 welcome in our community; where are  you from?

m472, 2020-11-09:
I implemented support for the Genius brake (see aforementioned issue #101). According to the information sources I used, the Bushido should be compatible with the Genius protocol.

Therefore you could try the code in #131 and see if it works when starting FortiusANT with the option ```-t i-Genius```.

Please keep me updated if you try it out, I would be curious if it works with Bushido.

Marmi86, 2020-11-09:
> @Marmi86 welcome in our community; where are you from?

Hello Wouter,
I came from Germany .

@m472 
Thank you ,
i will testing your setting with i-genius

My Tacx Hardware:
- Tacx Flow with Headunit Works nice with FortiusANT
-Tacx Bushido

m472, 2020-11-09:
Hi @Marmi86 

Took me a moment to realize where the problem lies...
I updated and tested only the python source code as I am running FortiusANT on Linux.
As a result the Windows executable you are trying to run is not built from the latest version of the source code.

I will try to build the Windows executable and upload it.

Marmi86, 2020-11-10:
> Hi @Marmi86
> 
> Thank you for trying to test my modifications!
> 
> Took me a moment to realize where the problem lies...
> I updated and tested only the python source code as I am running FortiusANT on Linux.
> As a result the Windows executable you are trying to run is not built from the latest version of the source code.
> 
> I will try to build the Windows executable and upload it.
> 
> Edit:
> I was not able to build the .exe yet, but I will try again tomorrow evening

Can you send me a instruction how to install FortiusAnt on Linux?
I will use FortiusAnt with a Raspberry

m472, 2020-11-10:
For installation instructions please have a look at the FortiusANT user manual: https://github.com/WouterJD/FortiusANT/blob/master/supportfiles/FortiusANTUserManual.pdf

As far as I understand actually both devices connect to FortiusANT via separate channels.

TinusVee, 2020-11-11:
Hi Guys, 

I recently bought a Bushido on Marktplaats (and found out it was a T1981 which doesn't work with Zwift....) and after a struggle with TTS4 I stumbled into FortiusANT!
So I'm very curious on the developments in this thread. If I can help in any way with testing, I will be glad to do!

Marmi86, 2020-11-11:
@m472 

i testing on Ubuntu 20.04  ,i have the same issues:
pc@ubuntu:~$ sudo python3 ./FortiusAnt/pythoncode/FortiusAnt.py -t i-Genius

19:20:23,737: Command line error; -t incorrect value=i-Genius

Version 3.5

WouterJD, 2020-11-13:
As soon as the Genius software is stable and confirmed, I can integrate.

Muleguy-Security, 2020-11-14:
Wouter, I did managed to try the 3.5 release with my Bushido T1980 and T1981 head unit. I tried using the -t i-Vortex command and was not able to find the trainer. Without the -t I was able to find a Wahoo SCS on a previous run so at least I can verify that my ANT+ sticks are working. From the post from Marmi86, it looks like there will soon be new i-Genius code to try with it so I will wait for that until further testing unless you have other suggestions. 

Output from test below. 

C:\Users\hlindley\Downloads\FortiusANT-master\FortiusANT-master\WindowsExecutable>Fortiusant.exe -t i-Vortex
23:36:07,196: Dongle - Using Dynastream Innovations dongle
23:36:07,196: Tacx   - Pair with Tacx i-Vortex and Headunit
23:36:07,198: AntHRM - Heartrate expected from Tacx Trainer
23:36:08,340: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
23:36:08,522: FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592
23:36:08,707: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
23:36:08,939: FortiusANT receives data from an ANT+ Tacx i-Vortex Headunit (VHU Controller), any device
23:36:09,148: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
23:36:09,340: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
23:36:09,525: Ctrl-C to exit
23:36:09,525: Tacx   - Pair with Tacx i-Vortex and Headunit
23:36:09,525: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:10,549: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:11,576: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:12,616: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:13,649: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:14,682: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:15,706: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:16,731: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:17,752: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:18,791: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:19,821: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:20,862: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:21,901: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:22,952: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:23,984: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:25,017: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:26,062: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:27,070: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:28,114: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:29,134: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:30,179: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:31,223: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:32,245: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:33,283: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:34,328: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:35,352: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:36,376: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:37,429: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:38,455: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:39,495: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:40,542: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:41,573: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:42,606: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:43,663: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:44,683: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:45,711: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:46,753: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:47,790: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:48,822: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:49,875: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:50,909: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:51,937: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:52,982: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:54,014: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:55,037: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:56,079: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:57,113: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:58,149: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:36:59,189: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:00,211: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
[... cut ...] 0 r= 100 T= 15
23:37:02,264: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:03,287: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:04,324: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:05,357: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:06,402: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15
23:37:07,439: Target=100W Speed= 0.0kmh hr=  0 Current=  0W Cad=  0 r= 100 T= 15

WouterJD, 2020-11-15:
Bushido appears to have other ANT+ id's (network numbers), so -t i-Vortex does not work.
Please follow #101 !

WouterJD, 2020-11-17:
The title is changed, because the head-unit is the significant device; the head unit hides details from the software
Related issues: #46 = Vortex, #101 = Genius, #117=Bushido

david-evanochko, 2020-12-14:
Apologies if this is answered in another thread, but has there been any updates with getting the Bushido (T1981 or T1982) to work with FortiusANT?

switchabl, 2020-12-14:
@david-evanochko Yes and no. There has been a lot of progress on the Genius, which is very similar in many ways (#101). I expect to complete Genius support in the next few days. We more or less know what needs to be changed for the Bushido, so it should not be too hard. I don't have one myself, so I will need some help with testing. But there will probably be something for you to try quite soon.

david-evanochko, 2020-12-14:
I appreciate the response @switchabl. That sounds great about future plans. Feel free to mention me in this thread for testing out the Bushido.

WouterJD, 2020-12-15:
@david-evanochko Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
Thanks for joining to get the Bushido working as well!

switchabl, 2020-12-18:
@mac2444 @Muleguy-Security @Marmi86 @TinusVee @david-evanochko

I have created an experimental branch with support for the Bushido, based on the work that @m472 and I have done on the Genius: https://github.com/switchabl/FortiusANT/tree/bushido
For Windows users there is also a binary available: https://github.com/switchabl/FortiusANT/suites/1695046665/artifacts/31854302

Before you get too excited, please note that I do not have a Bushido myself, so this is basically untested and there is a significant chance it doesn't work (yet)! But if you are feeling adventurous, give it a try and report back. I'm sure with your help we will get there.

The Bushido is controlled through the T1982 head unit, so the head unit has to be active and connected to the brake. I have (temporarily) changed the default command-line arguments to `-a -g -t Bushido -H0 -d 127`, which should be fine for testing.

This will also create a number of log files, please share those so I can see what is going on.

TinusVee, 2020-12-19:
@switchabl Thank you for the effort!!
I'll give it a try this weekend. As soon as I hve some resultss I'll share them here.

philhaupt, 2020-12-19:
Hi switchabl,

thanks for providing your code.

I did some rudimentary testing (I was able to pair the head unit to FortiusANT) but I am not sure if my logfiles are helpful...

[logfiles.zip](https://github.com/WouterJD/FortiusANT/files/5719343/logfiles.zip)

Yesterday, I received a second ANT dongle (unfortunately a CYCPLUS)
Both dongles are displayed under ["libusb-win32 devices"](url
[image]
)

Please let me know if you like me to test a more specific procedure.

switchabl, 2020-12-19:
@philhaupt Thanks, those log files are very helpful! I can see that pairing works and we are receiving som data (good), but setting it to PC control mode doesn't seem to work. So it looks like I need to do some more investigation and make a new version.

philhaupt, 2020-12-19:
just one additional comment: the head unit showed a "PC connection".

WouterJD, 2020-12-19:
@philhaupt thats what a vortex would display as well. It seems it means it's not controlling the brake but is communicating with PC

switchabl, 2020-12-19:
Thanks. Yes, it did switch the mode, I was just listening to the wrong response page. Fixing it now.

switchabl, 2020-12-19:
Ok, next try, hopefully this one gets us further. Things to try:
- does the head unit show "Start cycling"?
- if so, when you start pedaling, do you see sensible speed/cadence/power values?
- run in manual power/slope mode ("-a -g -t Bushido -H0 -d 127 -m" or "-a -g -t Bushido -H0 -d 127 -m"); up/down buttons on the head unit should change power/slope
- finally if things look good (and you have a working second ANT dongle), try with Zwift/Rouvy/etc

Again, whether it works or not, please share the log files so I can see what is happening.

WouterJD, 2020-12-19:
Congrats; development & tester👍👍🚴‍♀️🚴 Ride on!

WouterJD, 2020-12-19:
@switchabl question.
We now use -t to tell FortiusANT what trainer to use.
It's something bothering me and caused by the structure/location of the active ANTloop

We detect the ANTdongle (USB device) we could also do a pairloop to find the trainers. This requires quite some redesign.

My suggestion is to pick this up after Bushido/Genius integration. 

While typing - over a cup of coffee here- I think I will make a separate issue for it.

In the meantime I will pick it up, thinking how to rework FortiusANT. (Now active on BLE)

Let's integrate/release Bushido/Genius first "as-is" so that community can use it. After that time for improvement.

switchabl, 2020-12-19:
@WouterJD Yes, there should be no reason why we couldn't detect everything automatically. I think normally what is done is to open a separate "background scanning" channel for that and only open the device channel once you know the device is there. I think there is an ANT application note about that. That could maybe also speed up pairing. But in our case it may be a little more complicated because we have devices using three different frequencies. I would have to look into that. 

I have tried not to make unnecessary changes to the unrelated parts of the code when working on the Genius/Bushido. E.g. the calibration logic for the Genius is completely disconnected from the calibration code in the main loop. So it might make sense to rework calibration/"runoff" code as well at some point in the future to make things cleaner and maybe enable on-demand (re-)calibration instead just on start-up.

WouterJD, 2020-12-19:
So work-in-the-shop as we say in Dutch.

What is your experience in pairing-speed? I experience minutes for Vortex and sometimes HRM id not detected; as if the pairing is not exactly right...

WouterJD, 2020-12-19:
How does calibration for Bushido/Genius work: Fortius-motor-brake-style (calibrate) or magnetic-brake-style (runoff)?

switchabl, 2020-12-19:
@WouterJD Pairing the Genius takes something like 30 seconds for me I think. So not annoying enough for me to investigate properly so far. But should really be almost instantaneous if things worked properly.

Calibration on the Genius is Fortius-style, but the brake does everything internally. We just request calibration and wait for it to finish.
On the Bushido, calibration is spin-down (it cannot do downhill drive because it has no mains connection). But it should be initiated through the head unit. From our perspective, we do not deal with calibration at all.

rjon65, 2020-12-19:
Let me start with a big thank you for the amasing work that has been put in this project trying to avoid electronic waste.

Struggling with a running injury I dug up my old Bushido 1980 (1981+1982) only to find out that TTS4 support had stopped and that on Zwift could not connect up as a controllable.
A month ago I stumbled over this project and while awaiting the outcome of this thread I started learning python, git, and tried to get up-to-speed with your code base.

One of the things I have done is built an ANT message forwarder module to understand what was going on beween the different components (HU, TTS4, Brake), because I could not get things to work based on the info @switchabl gathered + some other source.

Now, unless I got it completely wrong my conclusion is that controlling the brake through the HU does _not_ work and one needs to control the Brake directly.
Having elaborated on this avenue, I have all messages to control and read the brake implemented. The only thing I'm still struggling with is the resistance setting: from the traffic I analysed from the HU or TTS4 - in power or slope mode - towards the brake I further concluded that there is only some sort of "resistance" setting.

I successfully used this on Zwift, but as said need to get my head around concerning the resistance setting.

Not being a proficient python programmer or git user, not sure what is the best way forward to make my code base available (which is still very much debug oriented, asserting all the assumptions it is based on)

Again, I may have missed somewhere a key point :-(, but maybe my findings could useful.

WouterJD, 2020-12-19:
Now, unless I got it completely wrong my conclusion is that controlling the brake through the HU does _not_ work and one needs to control the Brake directly.
> 

I cannot judge that, perhaps @switchabl can. If true that would be like how i-vortex works; brake is controlled directly. Headunit only serves to press buttons to control FortiusANT.

switchabl, 2020-12-19:
@rjon65 Thanks! This is definitely interesting (and a bit concerning). The functionality to control the Bushido through the head-unit definitely exists in TTS4 (even though it seems to default to controlling the brake directly, at least in newer versions). The same is true for the Vortex by the way! That is no guarantee that it actually works of course. Since I don't have a Bushido, I can only play with TTS, see what it would do and try to replicate that.

But https://github.com/fluxoid-org/CyclismoProject went to the trouble of actually implementing the protocol in an Android app, so I am assuming it did work for them and hope it will eventually work for us. It might also be something stranger, like it doesn't work with all firmware versions.

Still, it would not be the end of the world if controlling it through the head unit didn't work. It would just be a bit more work (we couldn't reuse Genius code and would have to do calibration ourselves). I haven't looked at the brake protocol in detail, but with your findings, the CyclismoProject wiki and a bit of digging on my side we could probably figure it out. I think the resistance value is actually force (similar to the Fortius), question is just what units are used.

TinusVee, 2020-12-19:
[FortiusANTbushidolog.zip](https://github.com/WouterJD/FortiusANT/files/5719764/FortiusANTbushidolog.zip)
Head unit shows "pc connection" but stays in "paused" (no "start cycling").
I have a second ANT stick which connects with Zwift, in Zwift power source cadence and controllable are connected to Tacx FE-C 57591. So it looks that the Zwift connection is working, but the head unit is paused.

rjon65, 2020-12-19:
With my Bridge (message forwarder) configured for the Bushido HU type ID, so sitting between HU and TTS4,  I tested two scenarios:

1) Brake paired with TTS
=> HU goes into PC connection mode and can be used to navigate in TTS screen (actually had never used this before, so learned a new feature :-))
Training can be started

2) Brake paired with HU
Same thing here, be it that training cannot be started (button in TTS4 remains inactive)

With your code base I've tried sending codes 1-5 (mode) to the HU, but did nothing made it show up speed, power, ...
I could try to cycle automatically through all codes to see if there some hidden magoic code

switchabl, 2020-12-19:
@TinusVee Looking at the log file, this is at least one step further. I see regular speed/cadence/power/etc. updates coming in from the head unit now, so something is working. I also see an error code from the head unit 0x08000 which I haven't seen before, but 0x_8000 codes are usually "communication error" I think. Is the brake connected to the head unit (brake is only powered while you are pedaling)? Does it work if you use it with the head unit in standalone mode?

@rjon65 Hm, I am currently going to mode 4 (PC connection) and then mode 3 (paused). Looking more closely at the Cyclismo code, it seems maybe we need to switch _again_ to mode 2 (active). I kind of thought it would do that automatically when you started pedaling, but maybe not.

Yes, the latest version of TTS only uses mode 4 and seems to use the head unit only for the buttons. The code for controlling the brake through the head unit is still there, but apparently never used. I think I need to dig out an older version and play with that. If I remember correctly, controlling the brake directly from TTS (without the head unit) was a feature added later.

rjon65, 2020-12-19:
@switchabl I'm on TTS 4.22.0 for what it is worth. So another variable in the game then (SW version), which would then explain why I did not see any significant traffic between TTS an HU

switchabl, 2020-12-19:
@rjon65 Apparently the change was made in TTS 4.12. I got 4.3 to run and got it to pair with my fake Bushido head unit. It does not pair with the brake at all, it instead asks the head unit what brake is connected and only talks to the head unit. So it should work in principle, it is just about getting the annoying details right.

@everyone
In any case, I've made a new version that is now closer to the Cyclismo implementation (request mode 2 (training) if mode 3 (paused) is detected, which old TTS does as well). I apologize for the trial-and-error, this may not be the last iteration. But hopefully with your help we will get there sooner rather than later.

> Ok, next try, hopefully this one gets us further. Things to try:
>
> -  does the head unit show "Start cycling"?
> -  if so, when you start pedaling, do you see sensible speed/cadence/power values?
> - [...]
 >
> Again, whether it works or not, please share the log files so I can see what is happening.

rjon65, 2020-12-19:
Great! I got to the second point now - buttons do not have an effect as yet though.
[FortiusANT.2020-12-19 23-21-31.log](https://github.com/WouterJD/FortiusANT/files/5720218/FortiusANT.2020-12-19.23-21-31.log)

switchabl, 2020-12-19:
@rjon65 Thanks, that looks really promising now! The button presses (I see UP several times in the log) are actually received, I just completely overlooked that they are sent as acknowledged data (actually the same on the Vortex), not broadcast, so they are discarded... Sorry for that.

Anyway, should be fixed now, now let's see if the power/slope is actually set correctly as well.
https://github.com/switchabl/FortiusANT/suites/1698424559/artifacts/31947872

rjon65, 2020-12-19:
Buttons work (in 50W steps, right?) - I'll call it a day now

[171 more replies]