WouterJD/FortiusANT #118: Adjust virtual flywheel according to virtual gearbox

switchabl, 2020-10-28:
When using the virtual gearbox, the virtual speed does not match the speed seen by the trainer. The acceleration differs by the same factor, so the flywheel force calculation will be off. I propose to adjust the flywheel mass accordingly to compensate:

```
Weight = max(0x0a, int(self.PowercurveFactor * self.UserAndBikeWeight))
```

This seems to work well for me and should give physically correct inertia depending on the virtual gear. The dynamic behaviour directly after a gear shift is probably not simulated correctly (depending on how the virtual flywheel is implemented internally, probably just F = m * a), i.e. in reality you would need to put in some additional power to accelerate as you shift up. But this is a more complicated topic.

WouterJD, 2020-10-28:
Seems quite logical and I feel OK to implement.
It's amazing how detailed you studied the code to get here; great work!

@mattipee, I appreciate your view.

mattipee, 2020-10-28:
Without looking into the details, if the Fortius unit is calculating inertia for a given rider+bike mass, then it does so based upon the wheel speed, not the virtual speed. So unless I'm missing something else, this suggestion seems sound.

As for the maths, linear is probably correct but I haven't looked into it just now.

Good spot, @switchabl !

WouterJD, 2020-11-03:
Implemented

closed 2020-11-03 by WouterJD