WouterJD/FortiusANT #119: FE Page 48 - Basic Resistance

switchabl, 2020-10-28:
Currently, basic resistance requests (page 48) are ignored. The Garmin Edge 520 (and probably other models too) uses these when you choose resistance mode (Menu/Training/Indoor Trainer/Set Resistance). Setting a power target or replaying a previous activity/workout is also supported, but setting a fixed grade manually (page 51) is not. So it would be nice to have some support for this.

According to the FE specification, we should apply a given percentage of the maximum resistance available at the current speed. This is quite straightforward for a magnetic brake, but I am not sure it makes sense for something like a Fortius. For my own use, I ended up setting a grade between 0% and 15% (https://github.com/switchabl/FortiusANT/blob/7530c06bee48581fd01777d6bb2123dcc70620c0/pythoncode/FortiusAntBody.py#L908), but I am not sure this is a great solution. Something like a Flow can't do have that.

WouterJD, 2020-10-28:
Page 48 seems as if it implements the sportsschool trainer with the turning-knob. 

I have an Edge that does not support these functions so cannot try.

I get the impression that the edge implements this turning-knob, from 0...100% and then leaves it to the trainer what to do.
In stead of using the Fortius headunit, now the Edge is used, which could work accordingly.

Since the Fortius specs are that a [maximum of 1000W](https://github.com/totalreverse/ttyT1941/wiki#power-limits) can be reached; % x 1000 would be good. But note that Tacx Neo specs state [2200W at 40km/hr](https://tacx.com/nl/product/neo-2t-smart/). I think for us humans, 1000W would be a nice maximum. Show me your training on strava where the 100% is a real limit and in that case we move up :-)
FortiusANT receives the Power or Grade from the commander and translates it to the parameters suitable for the trainer connected, so that Flow does not know about grade is not an issue (here). 

We already have the -m and -M flags, indicating we are doing a manual Power or Grade.
I agree on your approach which could be extended as follows (informally written):

```
if clv.manual:
    TacxTrainer.SetPower(ant.msgUnpage48_BasicResistance(info) * 1000)
elif clv.manualGrade:
    TacxTrainer.SetGrade(ant.msgUnpage48_BasicResistance(info) * 20)
else:
    # unexpected data page, since we're not in manual mode
    pass

```

switchabl, 2020-10-28:
Yeah, exactly, this is basically supposed to be an old-school resistance knob. For the reasons you mention, I don't think it makes sense interpret it as a percentage of maximum _power_ (and my Edge at least can set power in Watts directly with page 49 anyway). Adding a separate "resistance" mode in FortiusANT just for this is probably not worth it, unless it has some other use as well. That is why I went for setting a grade. The question is just what 100% should mean in this context. Ideally it might be something different on a Fortius than on a Flow, but I am not sure if this is easily done (or really necessary).

To be honest, I have used it mostly for testing, so I could set a grade manually without having to start SimulANT. I know that there is manual mode, but then you have to restart (and potentially recalibrate) for that.
Mixing the two might be confusing since it states that manual mode "ignore[s] target from ANT+ Dongle"?

WouterJD, 2020-10-28:
- why would purpose for FortiusANT be different from Flow?
- different meaning for -m an -M would mean a consistent implementation
- "does not listen to ANT" very good point; to be considered how to separate indeed! You really studied the code🤛

Main question remains why you would want to use the Edge where you already have the head-unit

WouterJD, 2020-10-28:
In modern words; what is the use-case?
As a tacx-er
I want to use a remote control
Because: ...

WouterJD, 2020-11-04:
Implemented in version 3.5 as requested

closed 2020-11-04 by WouterJD