WouterJD/FortiusANT #121: ANT remote control support

switchabl, 2020-10-28:
I wanted something more usable for virtual shifting (https://github.com/WouterJD/FortiusANT/issues/120) than the Tacx head unit, so I decided to implement the remote control profile.

For reference, see (free account required)
https://www.thisisant.com/developer/resources/downloads/#documents
D00001307_-_ANT+_Device_Profile_-_Controls_-_2.0.pdf

Most of the specification is irrelevant (audio/video stuff; I am not sure this has ever been implemented in the real world), the protocol is pretty simple. FortiusANT ("controllable device") is the master device and broadcasts Page 2, indicating it supports "generic control", interleaved with the usual Page 80/81 manufacturer/device info. This allows the remote control to pair and keeps it alive (when you stop broadcasting it goes into standby to preserve battery). When you press a button, the remote control sends Page 73 (Generic Command) with a key code. That's it.

I am not sure if it makes sense to include this, because there aren't a lot of devices available (and they are somewhat over-priced usually). I use this one https://www.thisisant.com/directory/antremote/. A German eBay seller sill has a few listed for 20€ and will accept 15€ offers (I have no relationship with them other than having bought two). I have sucesfully tried the Magura eLECT remote too, it is basically the same (except it supports suspension/seatpost profile in addition to generic control). The only other option seems to be the Garmin Edge remotes. Shimano unfortunately uses some proprietary profile, so DI2 buttons are not supported.

The profile implementation itself should be more or less fine (https://github.com/switchabl/FortiusANT). I have also implemented Page 71 (Command Status), which is theoretically required, but probably never requested in practice. The button handling needs some work. It should probably be consolidated with the Tacx buttons and it might make sense to have some configurability of the key codes. The o-synce remote sends MenuUp/MenuSelect/MenuDown (and 3 others that I don't remember when you press for 2 seconds). I don't what which codes the Garmin remotes send. It might be cool if some buttons could also be configured for turning in Zwift or something like that.

WouterJD, 2020-10-28:
Well for zwift, you are expected to use the companian app.
When not, the remote control goes outside FortiusANT, FortiusANT does not control Zwift (Zwift controls FortiusANT).

Having the Fortius head-unit, why would you want the remote. Apart from the fun of making it work?

WouterJD, 2020-10-28:
In modern words; what is the use-case?
As a zwifter
I want to use a remote control
Because: ...

mattipee, 2020-10-28:
I have an o_synce on the way. Personally, I would like fingertip gear changes for a single speed bike. I might even end up with two, to implement front and rear shifting separately.

switchabl, 2020-10-28:
Yes, same for me. I have attached the remote control so that I can shift virtually without moving my hands of the hoods. I also _don't_ have the Fortius head unit on the handlebars. It fits my handlebars poorly and I'm also not too confident it is sweat-proof.

@mattipee: I have also bought two with the idea of having one on each side for front/rear shifting. For 2x11 it works well enough with three buttons though, so I haven't implemented it yet. But it should be easy enough. The remote control sends its serial number with each command.

The Zwift-thing was just an idle thought. I do use the companion app for turning, but I find it somewhat annoying. Maybe it would be possible to emulate an arrow-key press when one presses a button on the head unit (or ANT remote). It is not really related to the control profile specifically.

switchabl, 2020-10-28:
Also, as mentioned before, ANT remote controls are not exactly popular. It is just that I got one cheap (well, two) and it works really well with my setup. But it is quite possible that nobody except me and maybe @mattipee will ever use this. So I would totally understand if you didn't want to include this. The code should be pretty much done though and I would of course be happy if I didn't have to maintain my own fork.

WouterJD, 2020-10-28:
> Maybe it would be possible to emulate an arrow-key press when one presses a button on the head unit (or ANT remote). It is not really related to the control profile specifically.

If I understand right, Zwift is controlled through
- `ANT+ FE-C` and/or
- the keyboard of the PC (or virtual keyboard on smartphone) where Zwift is running.

In fact I have a similar use-case, using Trainer Road where some buttons can be pressed to increase/decrease ERG, add time to cool-down, etc. So I have a small wireless keyboard with touchpad in front of the bike. The thought to be able to control through FortiusANT crossed my mind, but emulating mouse-movement and click with button-keys is not exactly simple.

Suggestions how to neatly implement this and not becoming Zwift-dependant are welcome.

switchabl, 2020-10-28:
How about allowing to run a configurable external command on button press? Then you could run an AutoHotkey script to do whatever you want. AutoHotkey is really powerful, no need to reinvent the wheel. For Zwift it should be as easy as "Send {Left}" and "Send {Right}". I haven't used Trainer Road.

mattipee, 2020-11-02:
O_synce remote arrived today. I'll try your branch later @switchabl , probably with remote mounted on tribar for rear shifting, and fortius headunit for chainring selection. I'll play with your flywheel correction and see if I can feel the difference.

mattipee, 2020-11-02:
Actually, great stuff @switchabl! I see now you put toggle on the menu button.

While riding, I was thinking about long-press behaviour and came up with something like:

- long-press up, 3 gear jump
- long-press menu, toggle chainring, but maintain closest ratio
- long-press down, 3 gear jump

The long-press up/down could also auto-change chainring and select appropriate ratio rear gear, if reaching the limit.

An additional thought - short-press up/down could also limit available rear gears for given chainring to avoid virtual cross-chaining.

In any case, thanks for your implementation of remote, and establishing the front/rear chainrings - meant I simply had to set reference ratio and ride. It felt real-enough for me on 8% in 1st gear.

mattipee, 2020-11-03:
Actually, scratch the long-press ideas. It would be better to have multiple presses result in multiple gear-shifts. This is easy to do as you can shift directly in the code where ANT page is received. There is no need (other than keeping alike code together) to have ControlCommand variable set and acted upon in the loop. For the buttons on the head unit, these are reported every cycle (0.25s) so they need to go there. However, queuing up a few ANT pages and having each of them shift gives a better response to multiple shifts.

switchabl, 2020-11-03:
Thanks, @mattipee, glad you found it useful.

I had some similar thoughts that led to the current configuration. I've used it for a number of rides now and it has proven to be surprisingly ergonomic (at least on a road bike with the remote just below the hoods). I really didn't like the long-press for shifting, the delay is just too long. I might end up using them changing routes in Zwift.

I think multiple presses kind of work already. In theory, the remote control profile runs at 4Hz and so does the loop, so there shouldn't be multiple button presses queued up (I don't think it is possible to send multiple commands in one time-slot?). But I agree, the shifting command should be moved into the ANT loop. For one, I am not convinced the ANT loop actually manages to run at 4Hz in all circumstances. At least, I have seen quite a few RX misses on the other ANT interface. It is probably a good idea to move the button handling code to a separate method. Then we can keep it together, but call it immediately when a button event occurs, from whatever source.

If I get around to it, I'd probably want support for a flexible key mapping through a configuration file, for both the Tacx remote and potentially several ANT remotes. Then for each key code you could choose from a number of predefined actions (resistance up/down, shift up/down, toggle, etc) or run an external command.
Once that is there, it would be easy to add more configuration options for shifting as well, like defining chainrings/cassette flexibly, or disallowing cross-chaining if you want that. The reference gear too, although it might be nice to have a way to change that at run-time. I am not sure about the best way to do that yet.

WouterJD, 2020-11-04:
Hi guys!
@BikeBeppe64 joins in the interest for remote control and actively participates.
Please let me know when code is ready to integrate into FortiusAnt.

mattipee, 2020-11-04:
Perhaps for me, long-press menu to override Zwift grade mode and put trainer into ERG at current power level, up/down to change set point and long-press menu again to revert to grade simulation. Just thinking out loud.

WouterJD, 2020-11-05:
@mattipee could you explain me in a short highlight what the purpose of the remote control exactly is. If integrated into FortiusANT it must be explainable to other users as well; even if it remains high-end functionality. Thanks.

switchabl, 2020-11-05:
Well, once the custom keymap is implemented: You can add buttons to your bike in any convenient location, which can be used to control FortiusAnt (resistance, virtual gearbox, etc) or your training app (or Netflix app if you prefer). Does this make sense?

WouterJD, 2020-11-05:
Yes the latter sure does, e.g. controlling music volume

BikeBeppe64, 2020-11-06:
yesterday while training, I needed to check data, I checked the lap settings, and go back, I had to do everything by pressing on the display.  it might be interesting to do it with the controller

[image]

[image]

switchabl, 2020-11-06:
When running Rouvy on the same PC, controlling it through FortiusANT is probably in the cards (either with the Tacx remote or with an additional remote control). With Rouvy on a separate phone/tablet, probably not so much (at least not until/unless we actually have Bluetooth support).

But it appears that the o_synce remote can be connected to Rouvy directly: https://support.rouvy.com/s/article/ROUVY-WORKOUTS-Remote-control?language=en_US
Note that on iOS/Android you need the new version of the remote control ("multiremote") with Bluetooth support. The older one I have is ANT only.

BikeBeppe64, 2020-11-06:
> When running Rouvy on the same PC, controlling it through FortiusANT is probably in the cards (either with the Tacx remote or with an additional remote control). With Rouvy on a separate [...]

Thanks for info me about it, I use Rouvy on Tab S2 which Ant+ is natively integrated, so that will be usable O-SYNCE, as yours as well, bought from same e**y seller

switchabl, 2020-11-06:
I am curious if that actually works, the link suggests it may not. But maybe they are just cautious because most Android tablets don't have ANT built-in.

If it does, we could in principle use that to control Rouvy from FortiusANT by translating button presses on the head-unit to ANT remote control commands. This is _not_ something I'm planning on implementing right now, but should not be hard if someone wants that.

I'll see if I can finish the remote control code over the weekend. The ANT part seems to be working well (been using it for a while now). I just need to figure out exactly how the configuration file should work.

BikeBeppe64, 2020-11-06:
Our O-SYNCE is on the way. I'll keep you informed later on.
All quite new Sa***g tablet/mobile phone has Any+ natively.

BikeBeppe64, 2020-11-14:
It doesn't work with Rouvy

switchabl, 2020-11-17:
Thanks for the update. I feared as much. With Bluetooth LE support in the works, I guess we will eventually be able to send commands to Rouvy/Zwift/etc using that (HID profile).

I haven't had too much time, so I am still working on the ANT+ remote for the moment.

WouterJD, 2020-12-14:
I have adopted the ANT+ Remote Control code from @switchabl (thanks).
Working branch: https://github.com/WouterJD/FortiusANT/tree/#121-ANT-Remote-Control

If I understand right, the ANT+ Contral automatically pairs with FortiusANT and responds to the broadcasted messages with the button pressed. (FortiusANT is master and CTRL is slave; intuitively I expected the opposite, allowing FortiusANT to pair with multiple ANT+ controllers, each having it's own function).

I now think it is only possible to use one ANT+ Control, not two at each side of the bicycle handlebar (for example).
Also; the device I have has UP OK DOWN only.
Do I understand that correctly?

switchabl, 2020-12-14:
@WouterJD Yes, maybe a bit counter-intuitive at first, but if you think about it, it makes some sense. You can actually connect as many as you want on the same channel. Like with a HRM (master), where you can connect many displays (slave). The remote does not respond to every message (would only drain the tiny battery). Instead, the broadcast messages from FortiusANT are more keep-alive messages so the remote does not turn off completely. The remote only sends a message when a button is pressed. The message also contains manufacturer/serial number so you can identify the source if you have more than one remote.

If you also have the o-synce, then it should do three additional key codes if you long press the buttons (don't remember which). There are also Garmin EDGE remotes that should work, that probably use different codes. Someone made an Android app that acts as an ANT remote, then people could use their phone. The app is broken at the moment but could be fixed if people are interested.

In general, I would say that the ANT remote code itself is probably fine (it is not very complicated), so feel free to experiment with it. The question is what to do when a button is pressed. Until we have a way for users to map buttons to actions, it is not very useful.

So I think we need to discuss about a configuration file before merging this. I have done some work on that but not pushed anything yet. Let me finish Genius support, then discuss? It is almost complete.

mattipee, 2020-12-14:
I'd be happy for the code to be integrated simply with empty button handling code and some simple comments along the lines of "#do something here". Configurable mapping (and JibberJim's mqtt stuff) perhaps suggests a much larger work piece to define an internal interface.

switchabl, 2020-12-14:
I wasn't thinking of anything fancy:
- a simple json config file that gets loaded into `clv` along with the command-line options (I think we really need that anyway)
- a section for each remote (Tacx USB remote, Tacx ANT remote, one for each ANT+ remote you are using) with {keycode1: [cmd1, cmd2], keycode2: [cmd3], ...}
- default map for Tacx USB that emulates the current behaviour
- a small number of built-in commands (navigation, shiting, manual control) that already exist in some form
- a command to run an external program: e.g. something like AutoHotkey/xdotool to control Zwift or whatever

That should cover most use-cases (I think) and could be done fairly quickly. I had already started on this in fact, I just haven't pushed anything yet.

It maybe also adresses #164 , because you can just un-map Cancel if you like.

WouterJD, 2020-12-14:
Well, I have extended the code to allow a left and right serial number; so that it's playable.
It will remain an option for the high-end users...

Note that the ANT+loop in FortiusANT is only active when Start has been pressed; an early pair-loop would be nicer.

WouterJD, 2020-12-14:
> * a simple json config file

For users who cannot create a .bat file with a command line, a JSON-file is not simple either.

> * default map for Tacx USB that emulates the current behaviour

Done

> * a section for each remote

Hooks are there now (two serials Left/Right); I'm curious what effective implementation suggestions will follow

> * a command to run an external program

Hmmm.... quite carefull here to get alll kinds of complexities.

But: if you could suggest me how to up/down my sound-card from python; I agree I already found a usefull option.

switchabl, 2020-12-14:
> 
> For users who cannot create a .bat file with a command line, a JSON-file is not simple either.
> 
Unfortunately true, I was thinking more simple to implement, not simple to use. But better than editing
the code, because you don't have to redo it for every update. Command-line options for button map
would be possible, but really messy. Eventually there could be a GUI for most common settings,
JSON has the advantage it is also easy to write.

> 
> > * a command to run an external program
> 
> Hmmm.... quite carefull here to get alll kinds of complexities.
> 
Well, true, maybe need some kind of a timeout so we don't accidentally fork 10000 processes if they don't terminate.

> But: if you could suggest me how to up/down my sound-card from python; I agree I already found a usefull option.
On Windows I assume? https://github.com/AndreMiras/pycaw I guess.

This is an example of what I think people could also do using the custom external commands: just run https://rlatour.com/setvol/ (or an AutoHotkey script, https://www.autohotkey.com/docs/commands/SoundSet.htm).

WouterJD, 2020-12-15:
OK. Basic implementation is made:
- If the headunit is unavailable or not mounted on the bike; the ANT+ remote can be used.
- UP / HOME / DOWN implemented

Multiple ANT+ controllers basic implementation created; when -C left/right is defined then the default behaviour is not active and an alternative behaviour can be implemented

WouterJD, 2020-12-15:
[image]

WouterJD, 2020-12-16:
Implemented in Version 4.1

closed 2020-12-16 by WouterJD