WouterJD/FortiusANT #122: Zwift do not detect "controllable"

orrmany, 2020-11-02:
Hi,

I have recently bought an i-Vortex. I have two ANT sticks: one no-name bought from Ebay, and one Tacx T2028. I try to use Fortius ANT on Windows 10 laptop from my employer, using the Windows binaries, as I've run into problems installing Python dependencies. 

Both sticks and the trainer  has been detected as follows. Zwfit Android client running on a Samsung Note9 (which has built in ANT+ radio)  does not detect "Controllable" from FortiusANT, so that is my problem. Any help would be appreciated.

orrmany, 2020-11-02:
Zwift sees "Power Source" (also the Speed Sensor as an alternative) and sees the cadence, but not the FE-C (controllable)

orrmany, 2020-11-02:
Output from ExploreANT:
```
c:\Users\egbozie\Downloads\FortiusANT-master\WindowsExecutable>ExplorANT.exe -D 0 -d 255
15:48:38,899: ExplorANT started
15:48:38,899:          ExplorANT = 2020-05-07
15:48:38,902:          antDongle = 2020-06-16
15:48:38,909: -d 255 (0b11111111)
15:48:38,910: -D 0 (0x0)
15:48:38,916: -H -1 (-0x1)
15:48:38,917: -F -1 (-0x1)
15:48:38,918: -S -1 (-0x1)
15:48:38,920: -V -1 (-0x1)
15:48:38,926: --------------------
15:48:38,927: Dongles in the system:
15:48:38,965: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
15:48:38,968: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
15:48:38,970: --------------------
15:48:38,989: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
15:48:39,514: Using Dynastream Innovations dongle
15:48:39,514: ExplorANT: We're in slave mode, listening to master ANT+ devices
Open channels: 0 15:48:40,105: FortiusANT tries to pair with an ANT+ device, any device
1 15:48:40,237: FortiusANT tries to pair with an ANT+ device, any device
2 15:48:40,368: FortiusANT tries to pair with an ANT+ device, any device
3 15:48:40,498: FortiusANT tries to pair with an ANT+ device, any device
4 15:48:40,629: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
5 15:48:40,781: FortiusANT receives data from an ANT+ Tacx i-Vortex Headunit (VHU Controller), any device
6 15:48:40,936: FortiusANT tries to pair with an ANT+ device, any device
```

orrmany, 2020-11-02:
Here is the outcome from FortiusANT
```
c:\Users\egbozie\Downloads\FortiusANT-master\WindowsExecutable>FortiusANT.exe -g -d 255 -H 0 -t i-Vortex -d 255
18:04:38,278: FortiusANT started
18:04:38,279: -d 255 (0b11111111)
18:04:38,280: -g
18:04:38,281: -H 0
18:04:38,287: -p 1.0
18:04:38,289: -t i-Vortex
18:04:38,291: ------------------
18:04:41,402: FortiusAnt GUI started in child-process
18:04:54,040: GetDongle - Try dongle: manufacturer=Dynastream Innovations, product= ANT USBStick2, vendor= 0xfcf, product=0x1008(4104)
18:05:18,251: FortiusANT broadcasts data as an ANT+ Controlled Fitness Equipent device (FE-C), id=57591
18:05:18,377: FortiusANT receives data from an ANT+ Heart Rate Monitor (HRM display), any device
18:05:18,530: FortiusANT receives data from an ANT+ Tacx i-Vortex (VTX Controller), any device
18:05:18,697: FortiusANT receives data from an ANT+ Tacx i-Vortex Headunit (VHU Controller), any device
18:05:18,850: FortiusANT broadcasts data as an ANT+ Bicycle Power Sensor (PWR), id=57596
18:05:18,993: FortiusANT broadcasts data as an ANT+ Speed and Cadence Sensor (SCS), id=57595
```
The devices 57596, 57595 are detected, but 57591 is not, which is my issue here

orrmany, 2020-11-02:
[FortiusANT.2020-11-02 18-41-57.log](https://github.com/WouterJD/FortiusANT/files/5477284/FortiusANT.2020-11-02.18-41-57.log)
[FortiusAntGUI.2020-11-02 18-42-00.log](https://github.com/WouterJD/FortiusANT/files/5477285/FortiusAntGUI.2020-11-02.18-42-00.log)

Frenske8, 2020-11-02:
I have the same issue when using my Android Samsung S9 smartphone (built-in Ant+) with Zwift. 
When using Bkool (work out app) on S9 even power, speed&cadence are not detected.

msjnaessens, 2020-11-02:
Zwift on android does not support controllable trainers via ANT+ yet, unfortunately...

orrmany, 2020-11-02:
@msjnaessens , ah then it is not a FortiusANT issue :) 
Downloading and checking with the PC app then...

Frenske8, 2020-11-02:
Thanks @msjnaessens

orrmany, 2020-11-02:
...on the other hand, I already have an nRF52840 SoC "thingy" which can be programmed to do the conversion. Eventual quest to make that happen, btw. :) 
Also, this problem to be mentioned at issue #103

msjnaessens, 2020-11-02:
More easy to use this thing I think :p
https://www.google.com/amp/s/www.dcrainmaker.com/2013/07/viiiiva-bluetooth-bridge.html/amp

orrmany, 2020-11-02:
@msjnaessens , are you sure it implements the FE-C profile, as well?  Ray talks about non-controllable sensors only.

orrmany, 2020-11-02:
Confirmed, the PC-version detects the FE-C, issue closed. Thank @msjnaessens  for the tip.

closed 2020-11-02 by orrmany

Frenske8, 2020-11-02:
I don't think it will do the job: the viiiiva has a one-way bridge, for a controllable trainer you probably need communication in both directions ( ant to ble and vice versa)

orrmany, 2020-11-02:
@Frenske8, it might, or might not. Despite the fancy name of an "FE-C" device (Fitness Equipment -- Control) it is still an ANS+ slave and the controlled "FE" device opens am ANT+ master channel for communication. I assume the peer situation exists on the BLE side, too, so implementing FE-C might not be such a big issue. NordicSemi has a boilerplate C++ sample code for the ANT+BLE translation for the HRM profile, so implementing FE-C might not be such a big deal, I assume

WouterJD, 2020-11-03:
Hi all; work is started on #103