WouterJD/FortiusANT #123: Errors when running Fortius without Ant+ dongles

torpedox88, 2020-11-02:
Hello everyone,

My name is Erik, 32 years old and living in the middle of the Netherlands, just a bit south of Utrecht.

After discovering and becoming very enthousiastic about turning legacy Tacx trainers into smart ones last week, I bought a used Tacx Fortius last weekend and ordered 2 Ant+ dongles, which are on their way from an obvious place to the Netherlands. 

Off course I cannot wait until the dongles arrive, so I was pretty happy about the possibility of running FortiusANT in manual mode, such that I can already do some easy riding as TTS3 does not work anymore after a fresh install. However, after directly running fortiusant.exe and trying slightly different commands as fortiusant.exe -g -m I get the following error:

Microsoft Windows [Version 10.0.18363.1139]
(c) 2019 Microsoft Corporation. Alle rechten voorbehouden.

C:\FortiusANT\WindowsExecutable>fortiusant.exe -g -m
19:26:04,656: Connected to Tacx Trainer T1932
Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "FortiusAnt.py", line 370, in Tacx2DongleThread
  File "FortiusAnt.py", line 111, in Tacx2Dongle
  File "FortiusAntBody.py", line 467, in Tacx2Dongle
  File "FortiusAntBody.py", line 479, in Tacx2DongleSub
AssertionError

I searched already through all other issues and this one seems to match partly with #107, but it's not clear to me if it was solved and if the manual mode without Ant+ dongles with the Fortius was working in the end.

Hopefully someone can help me out with this issue since I eagerly want to ride my indoor bike :)

WouterJD, 2020-11-03:
Hi Erik, welcome in thee FortiusANT community.
Sorry for the incomplete implementation; please give the current version (3.4) a try; it shoould be possible to operate without dongle now; untill now evidently nobody tried :-)

torpedox88, 2020-11-03:
Hi Wouter,

Thank you very much for the warm welcome and the quick support!

No problem, I directly tried version 3.4 and it seems to work: The errors are not present anymore and runoff + calibration are working as they should.
However, when I start cycling in manual mode after calibration there is a sudden increase of resistance after around 4 seconds until the wheel is almost standing still, after which the resistance goes back to normal level and the same happens again.
Do you have any idea if this is software related or maybe related to the hardware as in slipping of the tyre, too much force of the tyre to the brake or a broken brake/powerback?

Many thanks in advance again for your help!

WouterJD, 2020-11-03:
Please create testrun with -d127 flag and send created logfile

mattipee, 2020-11-03:
> a sudden increase of resistance after around 4 seconds 
> 
> Edit: I'm not sure if makes any difference, but I don't have the cadence sensor connected as well.

@torpedox88 I believe the Fortius does this when it doesn't detect cadence, it's a safety mechanism I think.

torpedox88, 2020-11-03:
> > a sudden increase of resistance after around 4 seconds
> > Edit: I'm not sure if makes any difference, but I don't have the cadence sensor connected as well.
> 
> @torpedox88 I believe the Fortius does this when it doesn't detect cadence, it's a safety mechanism I think.

After some internet search I indeed found that the Fortius does not work without cadence sensor (most probably because of the reason you mentioned), so I will try to make the cadence sensor work first before creating a logfile

WouterJD, 2020-11-03:
I've added it to the manual; to be uploaded next round.

This issue (running without dongles) can be closed; even though real test is pending.

closed 2020-11-03 by WouterJD

torpedox88, 2020-11-03:
Connecting the candence sensor solved the problem! :D

Many thanks @WouterJD and @mattipee for solving my issues with FortiusANT so quickly!

torpedox88, 2020-11-06:
> > a sudden increase of resistance after around 4 seconds
> > Edit: I'm not sure if makes any difference, but I don't have the cadence sensor connected as well.
> 
> @torpedox88 I believe the Fortius does this when it doesn't detect cadence, it's a safety mechanism I think.

Do you by any chance know how this works for the Tacx iFlow? In other words, does the iFlow work properly without cadence sensor?

WouterJD, 2020-11-06:
A Tacx Fortius does not work without cadence sensor

torpedox88, 2020-11-06:
> A Tacx Fortius does not work without cadence sensor
That's what I know by now since it made my Fortius work earlier this week, but some people around me became enthousiastic about FortiusANT as well so I'm looking for another legacy Tacx.
I found an iFlow for a good price but it does not have a cadence sensor, hence the question :)

WouterJD, 2020-11-06:
Post the question as a separate issue, who knows who will be able to respond
Now the title of the issues does not cover the topic

BikeBeppe64, 2020-11-06:
> That's what I know by now since it made my Fortius work earlier this week, but some people around me became enthousiastic about FortiusANT as well so I'm looking for another legacy Tacx.
> I found an [...]

I have an IFlow and works with its internal Cadence (box) and with an external Ble + Ant+ Speed & Cadence

torpedox88, 2020-11-06:
> > > A Tacx Fortius does not work without cadence sensor
> > 
> > 
> > That's what I know by now since it made my Fortius work earlier this week, but some people around me became enthousiastic about FortiusANT as well so I'm looking for another legacy Tacx.
> > I found an iFlow for a good price but it does not have a cadence sensor, hence the question :)
> 
> I have an IFlow and works with its internal Cadence (box) and with an external Ble or Ant+ Speed & Cadence.

With the internal cadence you mean the original cadence sensor supplier with the Tacx (as show in the picture below)? If yes, the Tacx also works (with FortiusANT in manual mode for instance) without it?
[image]

WouterJD, 2020-11-06:
Tacx Fortius does not know "who" is at the other end of the USB cable. To my knowledge Fortius does not work without cadense sensor.

I do not know the other trainere

BikeBeppe64, 2020-11-07:
> With the internal cadence you mean the original cadence sensor supplier with the Tacx (as show in the picture below)? If yes, the Tacx also works (with FortiusANT in manual mode for instance) [...]

Tacx IFlow works by Wouter FortiusAnt with cadence as showed in my picture. I get external cadence & speed (BLE or ANT+) using Zwift/Rouvy just in case of problem with Tacx cadence & FortiusAnt

[image]