WouterJD/FortiusANT #124: Track new releases / updates?

WouterJD, 2020-11-03:
Question: How can I keep track of new releases / updates?
I've installed version 3.3, and now I read 3.4 is out.
Is there a way to know this, and are there releasenotes somewhere?
As an IT-er I'd like to know what has changed and when to update.

_Originally posted by @kwouters1 in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-721189217_

WouterJD, 2020-11-03:
Nice suggestion for next version

orrmany, 2020-11-03:
Well, the usual ways is as follows:
1. use branches in your repo(sitory). Use the "master" branch for released versions only. Create (a) "development" (dev) branch(es) from the latest released-version (i.e., from the master/HEAD version). Do the development on the dev branch(es). 
2.. Whenever you do make a new release, then merge (all) the dev branch(es) o be released back  to the master in a single step. Label the new master/HEAD with the release version number. Use of [Semantic Versioning](https://semver.org/) is highly recommended. 
3. include a (version controlled) Release Notes document into the repository. Whenever you make a new release (i.e., new version in master branch) update the Release Notes accordingly. With the above scheme and with proper git commit-comments the Release notes comments can be generated semi-automatically from git history listing.

This way the "stable/released code" is on the master branch, your users are supposed to use only that one. During your development you should checkout the dev branch and commit there frequently, even the half-baked code. Comment the commits extensively for the purpose of easing the content creation for the Release Notes.

WouterJD, 2020-11-03:
What I seek -and will implement- is that a user is informed by FortiusANT that a new version is available (without checking github himself).
In next version😉

switchabl, 2020-11-03:
Ah I see, I would only ask that you don't make it into a pop-up, I usually find those annoying. And when I set up the trainer, I probably don't want to download an update _right now_, I will likely do it after training.

The easiest way would probably be to add a VERSION file to the Github repo and download that on launch and compare it to the current version?

Some of @orrmany 's suggestions, while good practice in general, may be unnecessary for a relatively small project. Still, at least tagging the releases in git (and creating Github releases for them) would be quite nice.

WouterJD, 2020-11-04:
Implemented in version 3.5, to be published shortly.

closed 2020-11-04 by WouterJD