WouterJD/FortiusANT #125: Please implement TCX-export in manual mode

WouterJD, 2020-11-04:
Thanks for the quick reply Wouter! 

It is also not possible to only "read" data from FortiusANT? 
Reason for asking is that I'm thinking about running TrainingPeaks for instance to record all data while running FortiusANT in manual/grade mode (or is there maybe another way to record all FortiusANT data?)

_Originally posted by @torpedox88 in https://github.com/WouterJD/FortiusANT/issues/113#issuecomment-721666917_

WouterJD, 2020-11-05:
Implemented in version 3.5

torpedox88, 2020-11-05:
Very nice, thanks a lot for the very quick implementation Wouter!

I noticed that power values are saved using the -d2 setting as well, but it is a bit cumbersome to only get the relevant data out of the log-file.
Next to that TXC files can be loaded into Strava for instance as well so this is perfect, I will test it later today and let you know about the outcome of it.

WouterJD, 2020-11-05:
-d2 is for debugging and not intended for your purpose.
Curious to hear your results!

torpedox88, 2020-11-06:
At first sorry for my late reply, I already did a ride yesterday evening but didn't had time until now to provide the outcome.

The TCX file is succesfully generated and can be uploaded to Strava or any tool to process TCX files, which is showing cadence and power as a function of time.
However, after observing that distance and speed are not shown in Strava it seems that those parameters are not stored in the TXC file.
Was this done on purpose (I used the trainer in "-m" manual power mode) or was it intended to save these parameters and is maybe not functioning well?

WouterJD, 2020-11-06:
Yes that is by design; ERGmode training does not have a track. Speed is quite virtual.
If you have suggestions, please provide

mattipee, 2020-11-06:
One could assume a gradient of 0% (or other) and then "do maths" to calculate speed/distance.

torpedox88, 2020-11-06:
@WouterJD You are right that the speed signal does not really make sense in ERG mode, but it seems that Strava for instance (at least in the browser, the mobile app was working fine) is struggling with having no speed and distance, so it makes sense to implement a solution as proposed by @mattipee (and maybe for the speed gauge in the GUI in -m mode as well?)

WouterJD, 2020-11-06:
> One could assume a gradient of 0% (or other) and then "do maths" to calculate speed/distance.

Well trainer road does not and others don't.
On a stationary trainer you don't move, so no distance is quite obvious.

Ref: https://www.strava.com/activities/4286798312

WouterJD, 2020-11-06:
I will check

torpedox88, 2020-11-07:
> I will check
> 
> And compare with https://www.strava.com/activities/4286798312

This is probably the result of a training in TrainingPeaks which is synchronized with Strava afterwards?

When loading the TCX file from my workout 2 days ago I get the overview in Strava as below, and in the "Analysis" tab nothing is shown (but in the mobile app it does, so this could be a Strava bug as well).

[image]

WouterJD, 2020-11-12:
First tests being done, not yet ready for distribution.

WouterJD, 2020-11-12:
> https://physics.stackexchange.com/questions/226854/how-can-i-model-the-acceleration-velocity-of-a-bicycle-knowing-only-the-power-ou/226856

I use the reverse of Speed2Power(); see https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L810

mattipee, 2020-11-12:
I know... interested to see the result.

I like how testing software burns calories these days. :)

torpedox88, 2020-11-13:
> > https://physics.stackexchange.com/questions/226854/how-can-i-model-the-acceleration-velocity-of-a-bicycle-knowing-only-the-power-ou/226856
> 
> I use the reverse of Speed2Power(); see https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L810

Although the solution proposed by @mattipee which takes acceleration into account is well looks very nice, I was also already thinking about reversing the Speed2Power equation (at 0 grade) which might be much simpler to integrate.

Good to hear that testing is on-going, hopefully the function can be implemented soon!

WouterJD, 2020-11-13:
Pfff Strava is not happy yet :-(

torpedox88, 2020-11-14:
> Pfff Strava is not happy yet :-(

Saving the inversed Speed2Power data in the TCX file does not do the job?

WouterJD, 2020-11-14:
@mattipee @torpedox88 Please check attached TCX. I would think it is correct, but the speed is displayed incorrectly.
Suggestions welcome!

PS. .TCX renamed to .TXT since github does not accept TCX.

[FortiusANT.2020-11-14 21-17-34.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5541489/FortiusANT.2020-11-14.21-17-34.tcx.txt)

torpedox88, 2020-11-15:
> @mattipee @torpedox88 Please check attached TCX. I would think it is correct, but the speed is displayed incorrectly.
> Suggestions welcome!
> 
> PS. .TCX renamed to .TXT since github does not accept TCX.
> 
> [FortiusANT.2020-11-14 21-17-34.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5541489/FortiusANT.2020-11-14.21-17-34.tcx.txt)
> 
> [FortiusANT.2020-11-14 21-41-26.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5541523/FortiusANT.2020-11-14.21-41-26.tcx.txt)

I've uploaded the files to both Strava and [CyclingAnalytics ](https://www.cyclinganalytics.com/try) and they are working fine, but speed and distance are indeed not corresponding with the delivered power.

@WouterJD What are the units of speed and distance in the TCX file? In the first seconds of the TCX file for instance the speed is around 20 km/h (I assume), but the distance is 5.22 in the first instance and 5.65 in the second instance, around 1 second later. Maybe there is a discrepancy between the units of speed and distance (or an error calculating from one to the other) which is recognized by Strava for instance after which it only uses distance to calculate speed?

WouterJD, 2020-11-15:
> What are the units of speed and distance in the TCX file?
If I am not mistaken; speed=km/hr and distance=meters.

Compare with next file as exported from Trainer Road
[2020-11-04 20-04 Trainer_Road_Junction_1.tcx.txt](https://github.com/WouterJD/FortiusANT/files/5542406/2020-11-04.20-04.Trainer_Road_Junction_1.tcx.txt)

WouterJD, 2020-11-15:
Zoals zo vaak: als je het niet weet zie je het niet.

<DistanceMeters> is de afstand vanaf het begin, niet sinds vorige trackpoint.
Kortom: de oplossing is nabij :-)

WouterJD, 2020-11-15:
You can do your tests with 3.6; please let me know results.

closed 2020-11-15 by WouterJD

mattipee, 2020-12-21:
@WouterJD  @torpedox88 ... a quick, related after-note because I've just discovered this... thanks to @ericchristoffersen

GoldenCheetah has a sim model which correctly uses power over time to calculate a virtual bike speed. A recent change was made specifically to use this model in ERG mode, _assuming a slope of 0%_, so that ERG mode workouts have a consistent speed and distance.

Such a model shouldn't be difficult to implement (I've played with one) and could well be a little enhancement for FortiusANT, to replace the inverse calculation.

https://github.com/GoldenCheetah/GoldenCheetah/issues/3712
https://groups.google.com/g/golden-cheetah-users/c/o3_0vHeGc1c/m/n0490j0aCAAJ