WouterJD/FortiusANT #126: Help needed - Adjusting the resistance control

torqueNoFriction, 2020-11-05:
Thank you for setting all this up! 

I am using a T1932 and have it all setup, able to connect to zwift and ride. However the resistance seems very off and I am not sure how to change it.

For example, when going on a flat surface it seems like there is barely any resistance and Im spinning a lot (i.e. feels much easier then it should be). When going uphill it applies too much resistance.

WouterJD, 2020-11-06:
Hi @torqueNoFriction, just to respond to my curiosity; please introduce yourself. I always like to know who I talk with, where you are located and what Tacx you use. 
Regarding power curve, please use the section on power curve validation (added today) which may be of any use

torqueNoFriction, 2020-11-06:
Hey, okay I'll check it out!

I'm using the tacx T1932 head unit with the t1941 fortius base unit. I'm from Montreal, Quebec, Canada.

WouterJD, 2020-11-06:
So you have the same setup as I have (T1932 + Fortius); repeating the power curve test could reveal the issue.
When doing the test, use the -d127 flag to generate the logfiles.

torqueNoFriction, 2020-11-06:
Unfortunately I do not have a power meter to write down the results. Any other way to run this ?

Just to make sure I am doing this right. I normally run the WindowsExecutable ForitusAnt. Wait for the bike wheel calibration to finish then connect to Zwift. Let me know if I am doing something wrong.

BikeBeppe64, 2020-11-06:
To get logfile, add -d127 on your usual command "WindowsExecutable"

torqueNoFriction, 2020-11-06:
I just click the executable to run it, do I need to run if via command line and pass this parameter ?

WouterJD, 2020-11-06:
Refer to https://github.com/WouterJD/FortiusANT/blob/master/StartUp/FortiusAnt%20(exe).bat
as an example to start with a command-file

Add -d127 behind "..\WindowsExecutable\FortiusAnt.exe"

WouterJD, 2020-11-06:
> Also I normally do not play with the headunit at all as I am unsure of what the settings are on it. Do you recommend to play with it at all during the ride ?

It's described in the manual; "4.1.5 Buttons on the Tacx head unit" if unclear please let me know

torqueNoFriction, 2020-11-06:
Thanks for the support! A few small questions

1) To run the test with the log I would need to have power-meter correct ? As I do not have one I cannot really validate this.

2) Regarding the buttons, yes I read that section. So if I am connected to Zwift for example, clicking OK it will reset the system to default. By clicking Up/Down it modifies resistances by +-10%. However should I be playing with this while I am connected to zwift or do I just simply click OK and make sure its at neutral ?

JBaptisme33, 2020-11-06:
Hello Guys, @WouterJD , thanks for your work.
I seem to have same issue than @torqueNoFriction . I have i-Vortex. (I live in France).

After calibrate i-vortex (with the head unit function), i run Zwift. (FortiusANT.exe -a -A -g -H0 -t i-Vortex)

On zwift, I have a resistance issue. It does not change regarding grade and sometime (I can't say when), resistance change dramaticly for only few seconds, and come back to basic one afterthat.

I can help to troubleshoot this issues with tests if you want ?

WouterJD, 2020-11-06:
@torqueNoFriction :
1- If you submit the logfile, I want to investigate whether things appear normal. Please indicate at what point you think that the resistance is NOT correct.

2- Consider the head-unit to operate a virtual gear: OK = 0, Down = lighter, Up = heavier. Just use it while zwifting, it does not impact FortiusANT or Zwift. Note: left = exit!

3- The "role" = the "resistance role" against the tire. The rubber tire should not slip (then you feel no power) or be too tight (you would feel too much resistance)

WouterJD, 2020-11-06:
> On zwift, I have a resistance issue. It does not change regarding grade and sometime (I can't say when), resistance change dramaticly for only few seconds, and come back to basic one afterthat.

@JBaptisme33 welcome in the FortiusANT community. Please introduce yourself (name, country); we now already now you're riding a bike on the i-Vortex :-)

Same for you: if you submit a logfile, including WHEN things do not feel right, I'm willing to take a look.

Note that, you can upload logfiles in the "Attach file section" while typing this note; if using  webbrowser (not email)

JBaptisme33, 2020-11-06:
I'm Jean, , I live in France 👍 
I've just tried a workout.
I've notice some things:

1- during all session , cadence can have issue during some seconds ( cadence can be false , can change (15%) during 3 to 5 seconds, for reason I can't explain
2- same on 0% grade, resistance change (not a lot, but sometimes it change)

3- at 17h56 (16h56) for logs, I had a big resistance changes whereas on Zwift, it was not so difficult

4- at the end (17h02-03 on log) Zwift change "dramatacly" of grade (from 0 to 5% then 4%) => I had no resistance change at this moment. (and I've quit 1 min later)

Big log is attached

[FortiusANT.2020-11-06 16-35-31.log](https://github.com/WouterJD/FortiusANT/files/5502548/FortiusANT.2020-11-06.16-35-31.log)

torqueNoFriction, 2020-11-07:
@WouterJD , okay I am going to play with my role first and get back to you with results thanks!

1) Do you by change have any coding for the turning/handlebar control ? I was going to maybe see if I can map this to keys on my computer instead to work with the GTA5 mod.
2) Do I need to adjust any default config like my wheel size, weight, etc ?

WouterJD, 2020-11-08:
Wauw @torqueNoFriction you ask a lot of short but complex questions (1,3) in one post. Please be more specific.

You do not need to change wheel size/weight. The speed is measured on the role, your weight is either default or provided by CTP(Zwift )

WouterJD, 2020-11-09:
@JBaptisme33 
> cadence can have issue during some seconds
this is usually caused by loose contacts on cadence sensor. Please check

sgtjlanc, 2020-11-10:
Hi.

I'm having the same problem as @torqueNoFriction. I'm using my dad's old tacx i-magic and when I ride on flats and downhills on zwift the resistance is very light and when I hit rises the resistance kicks hard and higher than it should be. I've seen that a candence sensor is needed to make the trainer work properly- I don't think I have one as the monitor for RPM's doesnt change on the fortiusant GUI but I use a separate one for zwift. Could this be the cause or could it be something else? I am very unfamiliar with the way the software works, I have only looked at the bits that were needed to get the program up and running. Thanks for all your hard work @WouterJD

WouterJD, 2020-11-11:
hi @sgtjlanc welcome on the FortiusANT community. I'm always interested to known where the software is used, so please tell some more about yourself.

WouterJD, 2020-11-11:
I changed the title to cover the subject.
Note that there are Fortius/Magic/Vortex users who work with FortiusAnt with succes; hardware issues MAY be the case. It's difficult for me to judge.

sgtjlanc, 2020-11-12:
Hi @WouterJD, I'm from the UK!

WouterJD, 2020-11-15:
Hi all. Version 3.6 has a new logging feature -d32 to create a JSON file for this kind of analysis.
Please let me know whether that helps to investigate.

WouterJD, 2020-11-17:
I'm digging into this "Power curve" subject and get the following impression: headunit/brake combinations T1932/T1941 and T1902/T1901 work, but the others don't.

Please respond to this post: what bundle did you buy, and what brake and what head unit do you use?

Bundle | Brake | Head unit | Remarks
-- | -- | -- | --
T1930 Tacx Fortius | T1941 Motor | T1932 | USB interface tested, see manual 2.6.2
T1900 Tacx i-Magic | T1901 Magnetic | T1902 | Legacy USB interface tested, see manual 2.6.3
T???? Tacx Flow | T1901 Magnetic | T1932 | Issues: #102 #143
T???? Tacx i-Magic | ? | T1904 | Issues: #128

switchabl, 2020-11-17:
The combination T1901/T1932 has been sold as T2270 i-Flow and T2220 Flow Multiplayer.
https://activesport.co/WebRoot/Store5/Shops/80c85f8f-7a95-4b1c-9c30-e64b314f3f2e/MediaGallery/tacx/T2270_i-Flow_Tech_Info_1310.pdf
https://www.qbp.com/diagrams/tech_content/pdf/tacx/T2220_Flow_Multiplayer_Tech_Info_1108.pdf

The basic T2200 Flow came with a T2200 head-unit without USB:
https://www.qbp.com/diagrams/tech_content/pdf/tacx/T2200_Flow_Tech_Info_1108.pdf

There were also 110V variants that just come with a different brake unit (T1911).

sgtjlanc, 2020-11-17:
> I'm digging into this "Power curve" subject and get the following impression: headunit/brake combinations T1932/T1941 and T1902/T1901 work, but the others don't.
>
> Please respond to this post: what [...]

tacx i-magic T1900 bundle, T1901 magnetic brake and T1932 head unit

WouterJD, 2020-11-18:
This seems the same as #143 and I will proceed from there
The [response from @cyclingflow ](https://github.com/WouterJD/FortiusANT/issues/143#issuecomment-729382076) is promising.

closed 2020-11-18 by WouterJD