WouterJD/FortiusANT #127: Unexpected resistance code

ceautje, 2020-11-07:
Hi Wouter,

Thanks a lot for doing this!
My name is Cor and I am from the Netherlands.

Just took my old tacx, bought a ant+ dongle and went through the manual. After starting the executable I got the message ‘start pedalling’. But when I start pedalling I get the message ‘unexpected resistance code’ and it stops calibrating. If I continue pedalling it constantly change resistance. 

P.s I did not connect Zwift yet, because it’s 64bit.

Thanks in advance!

If you need a log file, please let me know

WouterJD, 2020-11-08:
Hi Cor

What Tacx do you have? You can disable calibration with -n (no calibration). Calibration only works for Fortius/1932.

P.s I did not connect Zwift yet, because it’s 64bit.
That makes no difference; the interface (ANT messages) is independant: Zwift can run on Mac/64 where FortiusAnt runs on Windows32/python32 :-) As it should be

ceautje, 2020-11-08:
Hi Wouter,

Thanks for your reply.
I do have the tacx fortius 1932.

Regards, Cor

WouterJD, 2020-11-09:
Calibration _should_ work on Fortius 1932.
When looking at your logfile, I see one negative value 
`15:17:34,893: ReceiveFromTrainer() = hr=0 Buttons=0 Cadence=0 Speed=20.0 TargetRes=2117 CurrentRes=-96 `

then the resistance becomes positive, which is unexpected.

Question: do you simply leave the Fortius do it's work, or are you pedalling? You should **NOT** and leave the Fortius do it's own work.

Please give it a try

mattipee, 2020-11-09:
After calibration is complete, I can also see that no cadence values are reported - so separate to Wouter's question as to whether you stop pedaling during calibration as you should, do you have the cadence sensor fitted? If so, I wonder if it's working? Consequence of no functioning cadence sensor is that when pedaling, it will run for a few seconds but then stop (massive increase in resistance).

ceautje, 2020-11-09:
Thanks Wouter and mattitipee.

Indeed I started pedalling myself, so I did perfrom the test once again and it started ok, by running 20Km/h but after a few seconds I got the same message.

WouterJD, 2020-11-11:
@ceautje an idea to test: disable calibration and do a manual test.
If that works, let's proceed on calibration

ceautje, 2020-11-14:
Hi Wouter, I’ve tried that with unfortunately the same result. Probably it’s my cadence sensor I just can’t get that to work. 

The original one I don’t have, so I’ve tried another ant+ sensor. It connects but I don’t get sufficient data.

Thanks for your advice, I think I just throw the tacx out.

WouterJD, 2020-11-15:
An ANT+ sensor will not help; it seems that the tacx+headunit need the cabled sensor itself.
Note that the Tacx Fortius 1932 have no ANT+ interface.

All you need is a cabled sensor; would be a pity to trash the trainer for that only.

At the other hand: a brandnew Tacx Neo to cheer-up your mancave is a nice Sinterklaas surprise :-)

If you need more support let me know, will close issue for now.

closed 2020-11-15 by WouterJD