WouterJD/FortiusANT #128: Power on i-Magic

Mk2mark, 2020-11-07:
Hi and thanks for this great bit of code. Installed easily and worked first time! I am running the windows executable version.
I have an i-Magic with green/white controller (1901?)and have used it with your GUI and with Zwift. I just have one small issue.

When i compare power numbers on your GUi with the power on my bike power meter they are very different and the difference is not a linear %age so i can't fix it with the -p flag.
I can still run the TACX TTS software and that matches my bike power meter closely so the trainer is obviously sending good data. 
The number on your GUI matches the number on the scrolling display of data in the terminal window and also gets passed through to zwift correctly, so everything from that point on is fine.
I assume this is a problem with the mapping of i-magic data to power numbers - is it driven by a table of some sort?
Is there some way to correct this?
Thanks, Mark.

WouterJD, 2020-11-07:
Hi @Mk2mark welcome! Just for my curiosity: where are you from?

For the powercurve, please check the manual where is explained how the imagic formulas were created. With your pwr mtr you can reproduce

Suc6

Mk2mark, 2020-11-07:
Hi, Thanks for the very prompt response.
I am recently retired and from Northern England, a rural part of Northumberland, but still a keen cyclist despite my age! I have read the manual but i can't find the part you refer to. Where is it? I'm happy to do any work or logging required and to share results of thats useful.
I am IT literate but have not coded anything for 30 years now (I was quite good when 'C' was the state of the art!)

Mk2mark, 2020-11-08:
I have just tried running the system with your GUI. Whether or not i set -m flag the trainer does not force a constant power. is this a clue?
When I pedalled at a  constant 100w on your display my own power meter said 221w. at a constant 54w it said 98w. So a very big discrepancy.
I attach the log files if that's of any help
[50w (98w)FortiusAntGUI.2020-11-08 10-34-39.log](https://github.com/WouterJD/FortiusANT/files/5506176/50w.98w.FortiusAntGUI.2020-11-08.10-34-39.log)
[100w (221w) FortiusAntGUI.2020-11-08 10-30-44.log](https://github.com/WouterJD/FortiusANT/files/5506177/100w.221w.FortiusAntGUI.2020-11-08.10-30-44.log)
[FortiusANT.2020-11-08 10-30-43.log](https://github.com/WouterJD/FortiusANT/files/5506178/FortiusANT.2020-11-08.10-30-43.log)
[FortiusANT.2020-11-08 10-34-38.log](https://github.com/WouterJD/FortiusANT/files/5506179/FortiusANT.2020-11-08.10-34-38.log)

WouterJD, 2020-11-09:
> I have read the manual but i can't find the part you refer to. Where is it?

2.6	Power curve validation	12
2.6.1	General	12
2.6.2	Test for Tacx Fortius (T1932)	13
2.6.3	Test for i-Magic (T1902)	14

Please provide a result-table as in the manual, which saves me debugging the logfiles. Thanks

Mk2mark, 2020-11-09:
Thanks - and sorry for wasting time - I had an older version of the manual with no section 2.6!
I will see what I can do now I have it.
I am also getting a frequent error message that the Ant dongle is dropping out. Is this likely to be the problem I have read about with the Anself dongle, which is the one i have? If so any recommendation for a better one?

WouterJD, 2020-11-09:
See related issue.
Take care to have dongle close to each other without interference

Mk2mark, 2020-11-09:
Attached xls file with power data.
If it makes sense and there's anything else I can do let me know.
[T1904 power.xlsx](https://github.com/WouterJD/FortiusANT/files/5510709/T1904.power.xlsx)

Mk2mark, 2020-11-09:
Updated file - there was one data entry error (highlighted).
Also some analysis showing that the resistance calculation as present is approx 420*Pwr/Speed and is working.
Analysis of power reading discrepancies between GUI and intended; Garmin and intended and GUI and Garmin.
I cant draw any clear conclusion form these numbers but hopefully you can.
[Uploading T1904 power2.xlsx…]()

Mk2mark, 2020-11-09:
[T1904 power2.xlsx](https://github.com/WouterJD/FortiusANT/files/5512218/T1904.power2.xlsx)

Mk2mark, 2020-11-09:
My trainer does not have a cadence sensor - could that explain anything?

WouterJD, 2020-11-11:
A Tacx Fortius does not work (well) without cadence sensor.
I do not have the specs from Tacx so I do not know

Will look into the file shortly

Lorangaw, 2020-11-14:
I do have a simular problem. My vector 3 pedals says something completly different from FortiusAnt suggest. Yesterday, when I started from the command with the -d127 flag, Zwift worked alright with Fortius but today after using the latest versions of the scripts I was not even able to start from command line.
[image]

WouterJD, 2020-11-15:
Hi all. Version 3.6 has a new logging feature -d32 to create a JSON file for this kind of analysis.
Please let me know whether that helps to investigate.

Lorangaw, 2020-11-15:
> I do have a simular problem. My vector 3 pedals says something completly different from FortiusAnt suggest. Yesterday, when I started from the command with the -d127 flag, Zwift worked alright with [...]

I suppose it is likely that my problems with the power is that I don't get the cadence sensors to work. However, now I think it works fine with zwift when I run with the -d127 flag. But today something strange happend. I think that my Garmin  watch picked up the powersignal from Fortius and not from the vector. Zwift gave the corret power but my watch gave a value significatly lower until I reconnected the vector pedals...
Regarding the cadence, would it be possible for FortiusANT to use an external cadence sensor?

WouterJD, 2020-11-15:
FortiusANT does not **use** the cadence, only **displays** it.
Perhaps Pedal Stroke Analysis is depending on it, but that would be all (so do not use -A).

The point is that it **seems** (I do not have the specs) that the Tacx Fortius (iMagic) does not work well without cadence sensor. Reading the cadence from an ANT+ sensor would not change that.

`I think it works fine with zwift when I run with the -d127 flag` would be strange since logging _should_ not change the program's behavior.

Lorangaw, 2020-11-15:
Ok. That I understand. That is however the "state" I have tested that works fine. I tried using the GUI yesterday and got the impression zwift did not work ok. I had a restday and did only test in short set's. I will do some more tests. Good to know cadens is not involved. I'm struggling with the code since I am a novice.

WouterJD, 2020-11-15:
No problem, if you invest the time to understand the code - I will support if I can; not having the i-Magic for doing tests myself.

Lorangaw, 2020-11-16:
Great! Let me know if you want me to conduct any specific testing.
BR Morgan

WouterJD, 2020-11-16:
> [T1904 power2.xlsx](https://github.com/WouterJD/FortiusANT/files/5512218/T1904.power2.xlsx)

Sorry for not responding for so long.
FortiusANT receives the speed from the headunit (SpeedKmh) .
The so-called VirtualSpeedKm is displayed. This is only different in GradeMode when you change the virtual gear.
I would think, the speed on the GUI is displayed as received from the headunit.

If the wheelspeeds differs from the rotation of the brake-spindle, this would indicate that the tire is slipping.
Slipping would never cause the brake-spindle to rotate faster though.

As you may have read in the manual, the power calculation for the "Legacy USB trainer" is 100% empirically, based on one measurement (i-Magic, 1902 headunit):
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L1460
ALL OTHER headunits use the following calculation:
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L1636

Note however that ONLY the 1902 is considered a legacy headunit.
This is based upon [TotalReverse info](https://github.com/totalreverse/ttyT1941/wiki#t1902-usb-legacy-protocol-6-bytes-out-21-bytes-in) Reading TotalReverse info is very informative. Check the exact configuration.

Perhaps also the 1904 should be considered "Legacy" (although the description is quite different); the different calculations are found here:
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L345
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L405

So what I understand the only headunit that should use the legacy calculation is the 1902.
All others use regular calculation.

Please consider this

WouterJD, 2020-11-16:
@M2mark I have been thinking about this and returned to the excel sheet supplied.

What you do is:
- Set the resistance (row: Resistance?)
- Ride at a certain speed (set: 10, 20, ... 50)
- Measure the power (row: Garmin)

I have put it in a graph to detect some logic:
[image]

If I look at the figures for 10km/hr, with an incrementing resistance (2100-14605) the power is almost stable (even decreasing)
10 | GUI | 35W | 107W | 110W | 110W | 110W | 110W | 110W
-- | -- | -- | -- | -- | -- | -- | -- | --
  | Garmin | 29W | 88W | 85W | 87W | 72W | 73W | 68W
  | Resistance? | 2100 | 4300 | 6259 | 8346 | 10045 | 12500 | 14605

at 20 and 30km/hr, the graph has some logic:
Garmin | 80W | 90W | 200W | 200W | 235W | 250W | 247W
-- | -- | -- | -- | -- | -- | -- | --
Resistance? | 1060 | 2130 | 3120 | 4253 | 5423 | 6380 | 7400

Garmin | 140W | 145W | 156W | 260W | 420W | 430W | 420W
-- | -- | -- | -- | -- | -- | -- | --
Resistance? | 713 | 1427 | 2140 | 2817 | 3664 | 4390 | 5200

What I would do, is repeat the test as described in the manual and see what happens then

WouterJD, 2020-11-16:
To do the test:
- Use the python version, so you can modify the code
- Modify the function "TargetPower2Resistance" at
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/usbTrainer.py#L1636
- Replace `rtn = self.TargetPower * self.PowerResistanceFactor / self.WheelSpeed`
 by `rtn = self.TargetPower * 10`

In this way you can set the Resistance manually (up/down by 500) and see what happens.
It will NOT provide a solution, but gives you control to generate a usefull table.

Perhaps it turns out that the i-Magic allows a maximum value for Resistance of 8000

WouterJD, 2020-11-16:
Could you tell me how your configuration is setup:
- iMagic = Txxxx
- headunit = Txxxx
- brake = Txxxx
?

Lorangaw, 2020-11-16:
Hi, I will try. But maybe not before the weekend.
Br Morgan

WouterJD, 2020-11-17:
I'm digging into this "Power curve" subject and get the following impression: headunit/brake combinations T1932/T1941 and T1902/T1901 work, but the others don't.

Please respond to this post: what bundle did you buy, and what brake and what head unit do you use?

Bundle | Brake | Head unit | Remarks
-- | -- | -- | --
T1930 Tacx Fortius | T1941 Motor | T1932 | USB interface tested, see manual 2.6.2
T1900 Tacx i-Magic | T1901 Magnetic | T1902 | Legacy USB interface tested, see manual 2.6.3
T???? Tacx Flow | T1901 Magnetic | T1932 | Issues: #102 #143
T???? Tacx i-Magic | ? | T1904 | Issues: #128

Mk2mark, 2020-11-17:
Hi, sorry for being slow to respond and thanks for your continued work on this. Even as it is this is a great step forward for me and Zwift works fine in normal use - just ERG mode is difficult with the screwy power numbers.

I am taking in stages. I have resolved my Ant issues - moving both dongles to the end of a  1m extension to get away from WiFi interference helped, but still got dropouts. Switched from Anself to using two garmin dongles and everything is perfect.

Next I have procured a cadence sensor (from Slovakia!) and hope to have it by the weekend when I will fit it and redo the power tests as you suggest. Logically the cadence should be irrelevant but it's possible the head unit tries to do something clever with the information and that interferes with the power numbers. 

I have a T1904 head unit and a T1901 brake unit.

Thanks for all the info and the links - very interesting and helpful.

I will get back to you when I have some more numbers.

WouterJD, 2020-11-18:
This seems the same as #143 and I will proceed from there
The [response from @cyclingflow ](https://github.com/WouterJD/FortiusANT/issues/143#issuecomment-729382076) is promising.

closed 2020-11-18 by WouterJD