WouterJD/FortiusANT #129: insufficient data, len=24

joacableman, 2020-11-08:
Hi,
my name is Joa from Germany. (for the map info)

I tried to use Fortius Ant under Ubuntu and on Windows. In both cases i get the insufficient data, len=24 error with my Fortius trainer (blue interface, T1932). What could be the cause for that message?
I found the same problem in issue #104 but with the newest version from github the problem is not gone...
my ANT+dongle is detected as Dynastream innovations dongle (garmin)

many thanks,
joa

omedirk, 2020-11-09:
I have the same issue. However it occurs only after I press a button on the headunit.
before it works fine.
I have a 1942 headunit running on debian (RPi). 
(but I had the same issue when running it on windows before)

WouterJD, 2020-11-09:
len=24 errors:
- Too fast polling of the headunit: disable pedal stroke analysis
- Bad connections; check cabling

omedirk, 2020-11-09:
Thanks, found the analysis option in the other thread. Will have a look tonight. Will also try to if I can fix it, maybe do smart polling or something.

mattipee, 2020-11-09:
@joacableman @omedirk Is it just a single error and it keeps working normally, or repeated errors requiring a restart of FortiusANT?

omedirk, 2020-11-09:
repeated after the button press

joacableman, 2020-11-09:
@WouterJD do you mean using the -A option for pedalstroke analysis? this does not change anything.
the error repeats endless and pressing any button on the headunit seems not to give any result...
changed the usb cable - no change...

WouterJD, 2020-11-09:
-A activates PDA and is on when no command-line flags specified

Succes!

joacableman, 2020-11-09:
@WouterJD just to be clear: so you suggest **not** to use the -A option? yes this is what i did...

WouterJD, 2020-11-09:
In a case like this, at least specify the complete command-line;

omedirk, 2020-11-09:
@WouterJD I've just read up in the code (usbtrainer.py). I think it's in the extra polling after a button press (which i would advice against anyway.  With 0.1 sleep you shut down the whole thing for 100ms. A no-go in embedded control). 
This function was also mentioned in the other issue as a source of problems if i rememnber well. 

What I usually do with buttons is using only the rising edge. That way you never get multiple button presses.
The function requesting the button press should keep the last value though, but a very simple button class can be made that only outputs true when a new rising edge is detected.

mattipee, 2020-11-09:
@omedirk - I was just about to sit down to make changes to precisely this, to then create a PR. Have you played with any changes yourself yet?

mattipee, 2020-11-09:
One way to fix this, with a minimal footprint... @WouterJD @omedirk 

The approach uses two new variables (**self.ButtonState** and **self.ButtonAcknowledged**), and only sets **self.Buttons** on a rising edge.

In the two _ReceiveFromTrainer() functions (clsTacxLegacyUsbTrainer and clsTacxNewUsbTrainer), saved the tuple[nButtons] value off into a new variable **self.ButtonState**, rather than assigning directly to **self.Buttons**.

Then in clsTacxusbTrainer function Refresh(), propose the following code as substitute for the while():

    ```
    if (self.ButtonState == 0):
            self.Buttons = 0
            self.ButtonAcknowledged = False
        else:
            if (self.ButtonAcknowledged == False):
                self.Buttons = self.ButtonState
                self.ButtonAcknowledged = True
            else:
                self.Buttons = 0
```

If head unit buttons are zero, clear down.
If head unit button is not zero, then only assign value to **self.Buttons** variable once, otherwise setting back to 0, using **self.ButtonAcknowledged** as a control flag.

WouterJD, 2020-11-09:
Yeah, I will look into it asap.
Implementation is not "ideal"; I will implement concept of rising edge: the buttonpress is only valid when the previous message did not have a button pressed.

Thanks to open source and your participation implementation improves...

One BUT though: the implementation works on my T1932 so it may be that the issue remains.
But that's for later

WouterJD, 2020-11-09:
> ps. Thanks for the extensive documentation/commenting! I seldomly see such code in a new project

thanks @omedirk for the compliment. It pays off now☺️

mattipee, 2020-11-10:
PR here...
https://github.com/WouterJD/FortiusANT/pull/134

WouterJD, 2020-11-10:
See issue 135: Implemented in current upload of usbTrainer.py; please validate
Perhaps I should be more advanced on github etc; but I think this is the quick fix - even cleaning up the refresh() function that was specially for this.

omedirk, 2020-11-10:
Works like a charm for me. Thanks

WouterJD, 2020-11-11:
@Joa from Germany, how is it for you now? PS: if no button was pressed, the new software does not change behaviour.

@omedirk Could you tell some more on your configuration, perhaps in a separate issue. It's quite a new setup compared to others. If possible, also add some text how you installed, etc.

joa, 2020-11-11:
While I am also from Germany, I am not @joacableman :)

WouterJD, 2020-11-11:
Welcome @joa ; what tacx do you use?
Trying to keep track of (in the meantime) 120 FortiusAnt contacts!

WouterJD, 2020-11-11:
@joacableman What's the status for you?

joacableman, 2020-11-11:
@WouterJD i use a Fortius Multiplayer with T1932 Head unit blue one with the circle. (and a  steering frame)
At the moment i am not able to run the py file under windows because i get an AttributeError: function 'usb_find_busses' not found if i try to start the python version by 
..\pythoncode\FortiusAnt.py -g -a -H0 -d127 -Mpause
i tried to solve this problem last night but did find a solution so far
so could you also provide a windows executable with the recent changes so that i could try it out?

many thanks,
joa

WouterJD, 2020-11-11:
Ok will do. What python did you install?
The new version will not solve your problem though

joacableman, 2020-11-11:
@WouterJD  on windows 7 i use python 3.8.6

joa

joacableman, 2020-11-11:
i tried the new button code on my raspberry pi3, and now i can use the headunits buttons for the first time! I will try to use it with zwift as a next step.
i will keep you informed!
many thanks for this great project! 
joa

joacableman, 2020-11-12:
I could do my first zwift session today.
The buttons on the headunit are working now with the new button code but
i still get the insufficient data, len=24
message (repeated endless)
but communication with zwift works in both directions
Setup : 
Raspberry Pi 3 startoptions: without GUI ..\pythoncode\FortiusAnt.py 
Garmin Ant dongle
Fortius Multiplayer T1932 Headunit
Zwift under Mac osx, another Garmin Ant+ Dongle

WouterJD, 2020-11-13:
That is good news. It means the setup works, some finetuning to be done.
It cabling errors are excluded, check for timing - perhaps going to lower frequency of polling.

Hasoth, 2020-11-14:
Hi, I have the same issue with T1942, full blue head unit runnin on windows 10. " insufficient data, len=24" is running forever making the GUI part to not respond. Just quick question, was FortiusANT.exe file compiled to use new usbTrainer.py file from the issue #135? Do I need to make some modifications on my own? Sorry if that's a silly question...

WouterJD, 2020-11-15:
Hi @Hasoth are you sure that .exe is related to Ubuntu?

Anyways; today the .exe (3.6) is updated so must be in synch with the python code.
By the way; I like to know where FortiusANT is used and on what trainer; please tell some about yourself!

WouterJD, 2020-11-15:
> I could do my first zwift session today.

@joacableman; Issue closed, if more support needed - please let me know.

closed 2020-11-15 by WouterJD