WouterJD/FortiusANT #135: Pressing button "long" halts FortiusANT

WouterJD, 2020-11-10:
@WouterJD I've just read up in the code (usbtrainer.py). I think it's in the extra polling after a button press (which i would advice against anyway.  With 0.1 sleep you shut down the whole thing for 100ms -> and keeping the button pressed, you hang there for even longer! A no-go in embedded control). 
This function was also mentioned in the other issue as a source of problems if I remember well. 

What I usually do with buttons is using only the rising edge to detect a press. That way you never get multiple button presses.
The function requesting the button press should keep the last value though, but a very simple button class can be made that only outputs true when a new rising edge is detected.

ps. Thanks for the extensive documentation/commenting! I seldomly see such code in a new project

_Originally posted by @omedirk in https://github.com/WouterJD/FortiusANT/issues/129#issuecomment-724232789_

WouterJD, 2020-11-10:
Implemented in current upload of usbTrainer.py; please validate

mattipee, 2020-11-10:
@Wouter My only question is whether this code should, as you have it now, live in the top-level clsTacxTrainer Refresh() and not clsTacxUsbTrainer Refresh() as it was (and is in my PR). How does this code play nicely with the other trainers which are not 1/4 second polled over USB? I don't know the answer...

WouterJD, 2020-11-10:
The whole clsTacxUsbTrainer Refresh() only contains the "solution" to wait untill the button is unpressed.

What happens now in the clsTacxTrainer Refresh() is that a button-press is accepted when previously the button is unpressed.
If the underlying code "reports" a button press in multiple cycles after each other, these are ignored.

When inspecting the code where "self.Buttons" is set, I think this is the correct solution.
i-Vortex users should positively acknowledge....

mattipee, 2020-11-10:
From a quick glance I think the code you've added will play nicely with Vortex... but I think strictly speaking it's a peculiarity of the USB trainers, so even your implementation as is could be pushed down to clsTacxUsbTrainer, as a substitute for the old wait-until-unpressed. But functionally, I think it's fine.

mattipee, 2020-11-10:
Also, worked great for me. Tonight's session was on your master branch - new calibration plus buttons fixed. I was quite happy with behaviour of the head unit buttons. Still prefer the ANT+ remote and specific gearing. :)

WouterJD, 2020-11-15:
Implemented in 3.6

closed 2020-11-15 by WouterJD