WouterJD/FortiusANT #136: Latest windows exe deleted by Norton antivirus

SpeedColumbus, 2020-11-10:
When unzipping the latest windows file FortiusANT.exe gets deleted by Norton Antivirus with the warning message: Heur.AdvML.B heuristic virus detected. Is this a false positive?

Marmi86, 2020-11-10:
I habe the same problem with BullGuard

WouterJD, 2020-11-10:
That's bad news. On my system, the uploaded FortiusANT.exe does not give errors (Windows 10, Microsoft Defender).

WouterJD, 2020-11-10:
Hi @Marmi86 and @SpeedColumbus ; welcome in the FortiusANT community. I would appreciate that you introduce yourselves, just for my curiosity what Tacx is used and where you are from

SpeedColumbus, 2020-11-10:
Hi Wouter, Sure.. My name is Jordy and I live in The Hague, The Netherlands. I have an old Taxc Flow T1901 brake with a T1932 upgraded VR interface (USB). I have tested the FortiusANT file in a sandbox environment without Norton active, 2  Ant+ USB dongles and THEN the GUI works fine and also seems to work great with ZWIFT (I couldn't get Zwift working with the original antifier).

WouterJD, 2020-11-12:
Hi all; I have verified .exe and no issues at my side. Please verify at your side what's wrong.

closed 2020-11-12 by WouterJD