WouterJD/FortiusANT #137: Improvements for Raspberry PI

WouterJD, 2020-11-11:
@martin-vi; I create this issue for you.

Hello, i like to introduce myself. I am from Germany and live in the city of Dresden.

I became aware of the project, as I had been toying with the idea to see if someone had managed to bring the good old Taxc Trainer into the current century.

My hardware:

* Tacx T1901 Trainer
* Tacx T1902 Head Unit (the boxing says T1915)
* ANT with Dynastream Innovations ANTUSB2 Stick (0fcf:1008)
* Raspberry Pi, Version 1 Mod. A (yes the very old one from 2013!)
* Wahoo Bolt with HR, Speed and Cadence Sensor
* some network stuff ..

[image]

Currently I just plugged everything together and equipped the Rasberry Pi with Raspbian and FortiusANT. Without having really used it, functional is everything. 
The head unit is recognized and the Ant USB adapter transmits data. The Wahoo Bolt recognizes the trainer via Ant FE-C and reads plus sets the values correctly. The switching of the trainer modes (ergo/resistance) and the power values works. Fantastic work, thank you.
I would also like to try Zwift. But because of a lack of a computer with M$ Windows this will not work for the time being.

Currently I am still curing a small cold from (no covid-19 🤞), so no training for me at the moment. Probably until the end of the week.

Nevertheless I have encountered a few things where I would like to support or contribute merge requests.

* The usb adapter occasionally refuses to work. Disconnecting and reconnecting (physically) helps, but the solution is unsatisfactory. During debugging I found the place in the code where an usb reset is done. A new (additional) reset helps.

* My (very) old rasberry pi lacks X11 / gui. Nevertheless the wx import is globally anchored in the main code although this is not necessary. The installation of wx via pip aborts because of too little memory. I have helped myself by installing wx over the Debian / Raspbian repository. Not very satisfying because it unnecessarily eats up lot of memory.

* The Wahoo sends the user configuration after startup when connecting to FortiusAnt (page 55 User Configuration). For unknown reasons the rider weight is given as 0. Maybe to make Wahoo drivers faster :D. In any case, FortiusANT then exits with an assert test. I think we should move the plausibility check to where the incorrect value is read and prevent it from being taken over.

* The project does not follow the python packaging best practice. Is that intended? The use of setup.py with package dependencies etc. should save a lot of errors and work in the long run.

* The Linux installation guide is in need of revision, for example: The use of the Ant bridge project, is not necessary. Using sudo everywhere is not recommended. Are there plans to put the manual editable in git e.g. in Markdown format?

Sorry for the complaining. I just want to help FortiusANT to become even better. Therefore also the question, how should I arrange the support / how is that desired. I suggest I will create a few pull requests one by one.

_Originally posted by @martin-vi in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-725375714_

WouterJD, 2020-11-11:
Hi @martin-vi, thanks for feedback.
I will honestly admit: I learned python and github with the purpose to get my Tacx Fortius work. All kinds of learning-errors may be present and improvements are welcome. Windows/Tacx Fortius users will be helped a bit more since that is what I have but I'm trying to assist other trainers as good as possible.

- python packaging; a suggestion is welcome
- the manual is written in Microsoft Word, because that allows me all formatting possibilities; the github wiki is far too simple for a decent document (in my view). Suggestions welcome. If you submit a section, I'm happy to integrate.

The number of requests is currently quite high, but I am available to support if I can.
Note also #70, #57 #9, developments not being picked up yet.

omedirk, 2020-11-11:
@WouterJD in my opinion 57 is not really an issue as martin and me are using it without trouble.
It just needs some documentation maybe?
Also the LCD issue might be solved in many different ways. -> a web page might be simpler and more generic for all users.

WouterJD, 2020-11-13:
Well; I do give support on the FortiusANT setup as it is.
If changes are suggested fitting in the design, I am happy to integrate.

omedirk, 2020-11-16:
@WouterJD I was looking at the main code yesterday, do I understand correct you use multiprocessing / trheading from wxpython?
I ask this because an Idea I had was running the rpi without a screen  but with a remote control web-page.

WouterJD, 2020-11-16:
yes I use MP

MeanHat, 2020-11-26:
Hi @martin-vi; I am trying unsuccessfully to use a raspberry pi (Version 1 Mod A like yours) and pi zero w. Did you record your installation that you would be willing to share please? (@WouterJD; if you would prefer that I raise this separately please say).

WouterJD, 2020-11-27:
> you use multiprocessing / trheading from wxpython

Yes, but look [here](https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAnt.py#L533) it is not active when you do NOT use GUI; which is the case for you.

WouterJD, 2020-11-27:
> raise this separately

No it's fine here. Too many issues does not help.

trygvelu, 2020-11-27:
@MeanHat - I pulled the Genius-Support branch from m472, and then moved/removed some code to get rid of the wxPython/FortiusAntGui references. IMO it would be a good idea to separate the GUI calls so that one can run FortiusAnt without installing wxPython, which is the biggest obstacle for constrained systems. I can try to upload my modified version later today. I am also interested in further modifications, to support FE-C over Bluetooth LE on the raspberry pi (using Linux/Bluez).

martin-vi, 2020-11-27:
@MeanHat  Very gladly I support. I made a few notes with my setup at that time, which I like to share. If necessary these can also be included in the manual.

Unfortunately the manual describes the installation of too many things that are not really needed. The following parts can be left out:

* AntBridge - the usbserial kernel module can be used
* openant - FortiusANT implements ANT completely by itself and does not use openant
* antfs-cli - same as openant

This simplifies setup considerably.

The following steps are generally suitable for Debian based Linux distributions. This includes Ubuntu and also the Raspberry Pi OS (formerly Raspbian).

The instructions can be found here : https://gist.github.com/martin-vi/5206b6b26faf6f2d6121f6c8f300e44d

Gladly feedback. I can also support you if there are any problems. The best way is to contact me directly or use the comment function of the gist to not bloat the ticket here.

martin-vi, 2020-11-27:
> IMO it would be a good idea to separate the GUI calls so that one can run FortiusAnt without installing wxPython, which is the biggest obstacle for constrained systems.

@trygvelu I intend to make a merge request for that, but I haven't found time yet.

WouterJD, 2020-11-27:
> separate the GUI

The gui is separated, FortiusANT CAN run without (is active only when no command-line parameters are specified or -g). Small modifications required to be independent og wxPython, I think

WouterJD, 2020-11-27:
> It would be best to have FortiusAnt installed via setuptools

Suggestions very welcome! I am happy to adopt.

WouterJD, 2020-11-27:
> without installing wxPython

I think the easiest way would be to adjust FortiusAnt.py as follows:
```
try:
    import wx
    wxAvailable = True
except:
    wxAvailable = False
else:
    import FortiusAntGui
    import RadarGraph
```
and make some code conditional (of wxAvailable), so that python will not complain; -g flag is not allowed and must be set to false. Operationally, FortiusANT can work without GUI already.

wx in FortiusAntBody.py seems unnecessary
wx in FortiusAntGUI.py and RadarGraph.py are no issue, since those modules will not be used without GUI.

Comment on this approach and/or other suggestions are welcome.

trygvelu, 2020-11-27:
> wx in FortiusAntBody.py seems unnecessary
> wx in FortiusAntGUI.py and RadarGraph.py are no issue, since those modules will not be used without GUI.
> 
> Comment on this approach and/or other suggestions are welcome.

I concur. I also moved these constants from FortiusAntGUI to FortiusAnt:
```
mode_Basic          = 0     # Basic Resistance
mode_Power          = 1     # Target Power
mode_Grade          = 2     # Target Resistance
```
I'm guessing they don't really belong in the GUI code?

WouterJD, 2020-11-30:
But also, FortiusAnt does not belong in FortiusAntGui.
I will create a separate FortiusAntConstants; something I should have done long ago.

WouterJD, 2020-12-24:
Implemented in 4.1.1

closed 2020-12-24 by WouterJD

wtal710174, 2021-01-04:
@wouterstrand @MeanHat: I have written a short instruction for the setup of Bluetooth on the RaspberryPi 3B in my personal GIT. Please look at the following link:
https://gist.github.com/wtal710174/23081b986ab05c9a17facfec2d738202. Any comments and recommendations are certainly appreciated. This is the first quick and dirty version. @WouterJD Please feel free to download and add the md file to your repository. Thanks to @martin-vi for putting me on the right track with his Bluetooth.md and the Linux Raspberry ANT instruction!