WouterJD/FortiusANT #139: Install Fortiusant on Ubuntu 20.04

Marmi86, 2020-11-11:
Hello,
I installed FortusAnt on a fresh Ubuntu 20.04 VM
I followed the installation instructions from the manual,
but at this point I get an error message:

.....
LOC=/lib/modules/uname -r/kernel/drivers/usb/serial/
sudo mv $LOC/usb-serial-simple.ko ~/Documents

[image]

Open terminal in AntBridge folder
make; make;
sudo make install 

[image]

Can someone help me?

Marco

mattipee, 2020-11-11:
Just looking at face value, in the first LOC= bit, do you need backticks around the uname -r?

Marmi86, 2020-11-11:
I follow the FortiusAnt manuel steps

closed 2020-11-11 by Marmi86