WouterJD/FortiusANT #141: No Tacxtrainer found

Spider39, 2020-11-13:
Heb Fortiusant software op de computer staan (Windows 10) als ik de usb van de tacxtrainer aansluit . Wordt de tacxtrainer niet gevonden.
In de GUI onder nummer 6 staat  No Tacx trainer found.

WouterJD, 2020-11-13:
Heb je libusb geinstalleerd en TTS verwijderd?

PS. Stel je even voor aub, altijd leuk te weten wie FortiusANT gebruikt, met welke trainer en waar je vandaan komt!

Spider39, 2020-11-14:
Oh Sorry zal me even voorstellen ik ben Hendrik kom uit Friesland . Ik heb een tacx fortius die met een usbkabel is aangesloten. Ik heb een laptop die op windows 10 draait. Ik heb fortiusant er op gezet en ook libusb geïnstalleerd (ik denk dat het daar fout gaat).Op de laptop staat geen tacx software (TTS). #

WouterJD, 2020-11-14:
Nogmaals welkom, Hendrik!
Heb je libusb geinstalleerd vogens handleiding 3.5.1?

Spider39, 2020-11-15:
Hallo Wouter.
Heb het nu voorelkaar dat fortiusant software de tacxtrainer heeft gevonden en ook de ant+ dongle gevonden heeft. Nu eens zien of de kalibratie juist verloopt.
Wat een super project .
Tot zover bedankt ik hou jullie op de hoogte.

WouterJD, 2020-11-15:
@Spider39 top! Succes en ik zie je op strava !

closed 2020-11-15 by WouterJD