WouterJD/FortiusANT #143: I think that Python could do better algorithm

WouterJD, 2020-11-16:
Hi @BikeBeppe64  I have created this issue on your request, so it's not forgotten:

Ciao Wouter tutto bene? Scusa il disturbo...

Just my 2 cents for FortiusAnt
- Negative grade as downhill : while Rouving into my Strava "Up&Down" I got 63km/h during a pedalling downhill (-7/8%) but using with it  FortiusAnt max speed was more more less, around 15 km/h
- with no pedalling no speed, but in real biking is not so
- Power at < positive 1.2/1.4% FortiusAnt was 40w. My opinion could be at least 80w as intermediate biker.

I think that Python could do better algorithm... ;-) and I don't know how to help you.

WouterJD, 2020-11-16:
Well giuseppe.
Don;t know exactly what to answer; some people are happy with the current implementation and others are not.

The calculations are described in the manual and -if accessible for you- openly available in the python code; suggestions are always welcome.

BikeBeppe64, 2020-11-16:
I try to explain better my think 

1a) if there is a slow speed in climbing an uphill there should be an higher speed in downhill.  I noticed that in a downhill of 12%, fortiusant shows max 20km/h while ROUVY was 63km/h. It's also true that in an uppung hill same grade 10 % ROUVY was half of Fortiusant

1b) in the same downhill if I pedal or not I got 40w , in real world no pedalling no watts or changing the gear I got more watts. 

2) downhill or uphill with a slope of + 1.5% the power was fixed at 40w.  in my opinion we intermediate biker , have a force, at least, of 80w

A part of all, as you know, I do appreciate all your work that you are doing for free, I print "IFlow FortiusAnt by Wouter JD" on Strava account to thank you

WouterJD, 2020-11-17:
Hi @BikeBeppe64 

> I print "IFlow FortiusAnt by Wouter JD" on Strava account to thank you
**Thank you - highly appreciated!**

Zwift commands to ride uphill with -12%
You decide to ride with the wheelspeed of 20km/hr
FortiusANT calculates a required power: (grade/speed --> power --> resistance) see manual page 12
FortiusANT sends power to Zwift
Then Zwift calculates a speed (power/grade --> speed)

The two formula's are not necessarily the same.

At the other hand, this seems related to #102 
See also #115 #126 #128
Owners of these types should perform a test, see especially #128 where some progress is being made

WouterJD, 2020-11-17:
I'm digging into this "Power curve" subject and get the following impression: headunit/brake combinations T1932/T1941 and T1902/T1901 work, but the others don't.

Please respond to this post: what bundle did you buy, and what brake and what head unit do you use?

Bundle | Brake | Head unit | Remarks
-- | -- | -- | --
T1930 Tacx Fortius | T1941 Motor | T1932 | USB interface tested, see manual 2.6.2
T1900 Tacx i-Magic | T1901 Magnetic | T1902 | Legacy USB interface tested, see manual 2.6.3
T???? Tacx Flow | T1901 Magnetic | T1932 | Issues: #102 #143
T???? Tacx i-Magic | ? | T1904 | Issues: #128

BikeBeppe64, 2020-11-17:
My bundle T2250 Tacx I-Flow - T1901- T1932

cyclingflow, 2020-11-18:
I might have some pointers to help out.

I have a Flow (1901 Brake) and 1932 combination. Could not get FortiusAnt to work. Calibration attempts with error messages,
(Yes i know, i switched it off) flawed power readings, power mode not working well, grade mode totally off.

However, i have downloaded and adapted both antifier(power_curve) and FortiusAnt, now they both do work. I have been calibrating the Flow + software with a powermeter on the bike.

I probably have missed the way it was intended to work,   but I basically don't see see how the current implementation in usbTrainer would work correctly with my combination in grade mode.
I have implemented an additional clsTacxUsbTrainer (clsTacx1901UsbTrainer)
that incorperates some of the original logic of antifier, including reading a custum power curve file.

Yesterday  I finished an 1 hour time trial in Rouvy AR and everything worked well. Have also used the adapted antifier with Rouvy Workouts and Zwift.

Anyway, thx for FortiusAnt, Wouter, I like it.

WouterJD, 2020-11-18:
Welcome @cyclingflow to the FortiusANT community
- - - - 
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - - -
I am very interested in your clsTacx1901UsbTrainer because that seems to solve a set of issues.
I did not manage to get the original antifier work succesfully with Fortius (power curve incorrect) and that is how FortiusANT forked from it and developed to what it is now.

Great to hear that you understand the subject, we can make numerous Flow'ers happy, I think!

cyclingflow, 2020-11-18:
I have forked FortiusAnt and Antifier. (Antifier will be needed primarily because of the power_curve script. You could include it in FortiusAnt, but is currently certainly not up to your standards. And needs conversion to python3)

I will be uploading modified files.

I think some clarifications of the changes I made my help you in understanding, and may be bold enough to suggest a few minor 'refactor' suggestions.
If and how would you like to receive them?

WouterJD, 2020-11-19:
@cyclingflow I could download from your forked/branched implementation - this no issue.
I will certainly study how your clsTacx1901UsbTrainer class works.
As said, I did not manage to understand antifier's approach for the powercurve; the code is quite complex.

But it would be very helpful, if you would want to execute a test as done by @yegorvin, see manual "2.6.3 Test for i-Magic (T1902)".  In your case, you have a working model, so no need to tweak.

Purpose is to redefine TargetPower2Resistance(self), converting TargetPower to TargetResistance (using WheelSpeed).

If you don't mind, please merge logfile.py (so you can create the json-file) and add 
https://github.com/WouterJD/FortiusANT/blob/master/pythoncode/FortiusAntBody.py#L796

Then perform a test in manual mode (-m flag): (16 minutes test time after good warming up)

Target Power | 10km/hr | 20km/hr | 30km/hr | 40km/hr
-- | -- | -- | -- | --
100 | 1  minute |   |   |  
150 |   |   |   |  
200 |   |   |   |  
250 |   |   |   |  

and [if there is energy left - we could start with the results of the first test] in manual grade mode (-M flag):

Target Power | 10km/hr | 20km/hr | 30km/hr | 40km/hr
-- | -- | -- | -- | --
0% | 1  minute |   |   |  
5% | 1  minute |   |   |  
10% | 1  minute |   |   |  
15% |   |   |   |  
20% |   |   |   |  

the JSON file should provide the resulting data.

Using the JSON file contents, I will see whether I can execute the mathematics like done for @yegorvin.
It will also reveal the range of TargetResistance

WouterJD, 2020-11-20:
@cyclingflow would you mind doing the test as described? Would help us forward in understanding how i-Magic should be addressed

cyclingflow, 2020-11-20:
Well, i have been doing lots of calibration tests to provide curve_factor_power factors. 

Have just been doing such a grid test, results not yet satisfactory.

But let me explain where i stand.

cyclingflow, 2020-11-20:
DIFFERENCES AND IMPLEMENTATION

What are the difficulties in the current coding for using 1 Flow type 1901 brake?

- The 1901 only reports back a limited set of resitance values. (14 values, or 14 resistance levels)
- There is no guarantee that these values really linearly related to a single multiplier,
moreover, as i will explain in next section, one lienar equation is not enough anyway
- each level may need different tuning
- The current mechanism always uses powermode, and updates resistance based upon power,
which made it difficult for me to implement antifiers approach, if not false.
A manual grade to resistance is easier anyway.

So, I took the easy way, and started with antifier approach,  with estamation per level, a clear grade vs power mode,
and manual grades.

It worked, although not completely (next section).
Later on i gota better understanding of the code, and have been reversing the approach to be more in line with your implementation, such that for example the real life grade->power-> resistance can be used.
Although this makes no sense for this brake, as the power grid test will clearly show.

cyclingflow, 2020-11-20:
THE ANTIFIER APPROACH

Antifier's approach can be easily understood if you take a look at a plot of speed vs reference power (power meter), based upon a run of a adapted power_curve.py script. I have done many the last days, because it introduced a new problem: inconsistent results over reruns.
First the approach. Take the next plot, at resistance level 5, (starting with 0, the sixth)
[image]
You can clearly see it is highly linear -as aspected-, but does not go through (0,0) as a single multiplier approach assumes.

Using this approach, with a linear equation (including intercept or constant b in most tutorials) for each level, and a calibration run of power_curve.py, you get reasonable results, **for higher wattages!.**

My initial run had an average power of 203 as estimated by the new FortiusAnt usbTacx1901Trainer, and an average reference power of 196W. Which is fine, especially given the fact that i use a one crank power meter on a leg with busted knee: a clear tendency to underperform the other leg.

However, consideration of the reason for the unexpected huge intercepts for high resistance levels, made me think of to problems in this approach.

cyclingflow, 2020-11-20:
ANTIFIER APPROACH PROBLEMS

First, my suspected reason of the plot as shown above. For lack of  data (inobtainable) at low speeds, you cant really see it, but i expect the brake to curve back to (0,0) at low speeds. What i think the brake does, is slowly increase the brake to the required resistance level at a certain  speed, say 10 kph, and being fairly linear afterwards. Makes sense to me given the electro-magnetic nature, but specialist may comment/explain. (For example, if possible, it would probably require high currents at low speeds to induce the required resistance.)

_Problem1_
This approach gives overestimated powers at low speeds. May be not so much a problem for many users, but it also makes power control at low powers a mess.
I originally expected to be using an integrated spline approach to the estimation if needed. After looking at the plot i decided to use a piecewise lienar approach with an knot. Say the knot is at 10 kph, you will have the original linear approach above 10 kph,
and a lienar approach from (0,0) to the point at 10 kph on the line.
Straightforward, and it solves the estimation of the lower powers.

_Problem2_
However, if you do this with a separate approach for each resistance level, you violate monotonicity constraints.
In practical terms, imagine that you cycle at a given speed, and given resistance.
Now, if you either increase resistance, or increase speed, in both cases the estimated power should always increase.
Antifiers approach with separate estimation certainly does not guarantee that over the complete range of speeds,
although it does work at high speeds and high wattage.
It also does not help that getting consistent emperical data turns out to be difficult.

Now, as an estimation problem all equations could be solved in the least squares framework, using an combined approach to all 14 estimation problem with inequality constraints on the parameters. For me fun to do, and not to difficult, but overkill.

I have manually regularized the parameters to satisfy the constraints (increasing slopes, increasing intercepts). And i suspect that in the end if we have good estimations of the parameters, we can resolve everything back to a singular approach, because these parameters will be related to each other and the resistance levels.

cyclingflow, 2020-11-20:
THE GOOD NEWS
is that i have implemented everything i wanted, and will upload an exe. It's working.

The bad news is that my current power_curve_factors file is not good enough:
I have difficulty getting consistent results when doing power_curve runs (even one shortly after the other, on a warmed up system). So that i need many tests to get accurate results.
I have been testing and testing and manually adjusting factors. And overdone it. An extensive test as you requested 
showed that powers at low speed are OK, but now high powers are underestimated. 

[image]

cyclingflow, 2020-11-20:
UPLOAD.
Wanted to.
Uhm, exe is to big. I have used the spec file included in the project.

cyclingflow, 2020-11-21:
There is a bug in first linear piece, i will bring out a new release.

I have now more knowledge of my calibration/reference power issues. I realized that i do not need power_curve.py, but are far better off with using my bike computer and the resulting fit file, to analyze speed vs power, provided i set the wheel circumference exactly right.

Results are much better, but also indicate i have a problem with my stages crank power meter: power readings are not consistent over cadence differences. A couple of months ago i noticed a large difference in the calibration factor of the device, having been consistent in the months before, and being consistently differently after.
So i will have to send in the powermeter to recalibrate it.

WouterJD, 2020-11-21:
@Cyclingflow great work you perform; thank you very much.
Please proceed to get the data correct for the Flow

Last year, around this time, I started with the antifier software and was impressed by the tables and approach. BUT it works!
I also tried (like many others) to get a power curve defined for the Fortius but did not manage to.
I found TotalReverse's work and combined it and so FortiusANT became antifier 2.0

The Fortius and the Flow both work based upon formula's and hence I believe that also the 1932/1901 combi must be possible to be programmed analogously

What I have done now, to understand the subject better, I have ported the formula's to excel to visualize the calculations.
It's not finished, but it's time to call it a day.
I wanted to share the intermediate result with you anyway.

Conclusion of the exercise: the FortiusANT/USB formula provides results in the same range as antifier formula's; but not close enough to be satisfying.
Todo: create a formula that provides the power curve that is compliant with the antifier-approach.

Summary:
- I am absolutely impressed by the antifier approach with tables and external files; it looks like too complex for the subject though. But - as said before - it works and therefore provides the data we need.
- Tacx' TTS works for all Flows without parameterization, so one approach should be good for all
- Using your excellent work, I will find out the formula's and work towards a testable version shortly

cyclingflow, 2020-11-22:
Interesting! And they clearly show the monotonicity 
>>Tacx' TTS works for all Flows without parameterization, so one approach should be good for all
True, but perhaps at the sacrifice of accuracy. Will see.

[image]
An example of my latest results. I order all values on the time scale and select platforms of speed and power, avoiding the speedup jumps, calculating average speed and power for a platform.

I call it a day too ;-)

WouterJD, 2020-11-22:
Do you use the external file or the internal values?

cyclingflow, 2020-11-22:
They are actually, the same, there are two correction factors in there i have reset to zero.

But adjusting, to see the much better results. Only using FortiusAnt to select the grades. It's satisfying to see the much better results. :-)

I'm calibrating now through my bike computers fits files, going the speeds steps, at one particular resistance level. works fine.
[image]

I can hand pick a flat level and get the info at the selected range. Much more accurate than power_curve.py, no additional cleaning of data needed.

cyclingflow, 2020-11-22:
I have been thinking about the returned resistance values: do they help, or do we simply need equally spaced values.

Anyways, i have been expecting my approach to be able to come down to this (because slopes and intercepts are highly linearly related):
Step 1: (slope = factor 'a' in linear power equation, intercept = factor 'b',  power=a*speed+b)
Get the slope for resistance level
Slope = (resistance or integer)*a_1 + b_1
Get the intercept for resistance level
Intercept = (resistance or integer)*a_2 + b_2
Step 2:
Power = speedKph * Slope + Intercept.

This will bring the whole table down to at most 4 constants a1, a2, b1, b2.
Looks like it's going to be like that. Will certainly ease calibration for other users.
Which brings up the calibration values returned by the trainer: are they of any use? 

I would like NOT to bring it down exactly to the fortius formula's, because it would imply dumping the intercept.
That will always harm the accuracy.

And i'am dreaming of bringing in the acceleration factor in the future. Bear with me, i'am probably as passionate about good estimates as you are about programming :-)

WouterJD, 2020-11-22:
You are absolutely right

Note that:
1- Legacy USB works well (magnetic = exponential curve)
2- Newer USB has resistance in right range, but the curve is not right

What I think:
==> Formula 1) must be corrected to the range of 2) and then we're done

Doing other things today, just answering some questions :-)

cyclingflow, 2020-11-22:
Have fun!
Am going outside soon too, nice wheather

cyclingflow, 2020-11-22:
(Oh, and BTW, when you have time: could you provide me with power supply details if you have them? 
I got a bargain on a Fortius yesterday, 50 euros, but without power unit. Fine with me, if i don't get it working, at least i have a second 1932 unit, and get my cycling mate training with my second flow.
No rush)

WouterJD, 2020-11-22:
I have tried to port the antifier's formula's to Excel to visualize what it does.
See attached excel (zipped) 
[AntifierCyclingFlow-V2.zip](https://github.com/WouterJD/FortiusANT/files/5580436/AntifierCyclingFlow-V2.zip)

Following support would be appreciated:
- who checks whether the formula's are ported to Excel correctly?
- who explains the logic to me?
- who proves that the two formula;s CurrentResistance2Power_antifier() and TargetPower2Resistance_antifier() are each other's opposites?

Untill now I did not deciphre the logic behind it.

cyclingflow, 2020-11-23:
i will check

cyclingflow, 2020-11-23:
I have finished calibration runs.
Using the Per_level_separate_equation method i get good results now. 
Will produce an power * speed matrix

cyclingflow, 2020-11-23:
Some more formula's for the simultane - all at once - estimation of the 4 factors directly upon the calibration points
Given the earlier equations
Step 1:  
Slope = resistance*a_1 + b_1
Intercept = (resistance or integer)*a_2 + b_2
Step 2:
Power = speedKph * Slope + Intercept.

Lets substitute step1 into step 2.
Let
m = number of levels
r_i = resistance at level i
and
per level we have a set of observations
O_i = (S_i_j, P_i_j) of size N_i
where,
P_i_j = power at level i for observation j in O_i
S_i_j = speed at level i for observation j in O_i
Then we have the equation
P_i_j = r_i*S * a_1  + S_i_j * b_1 + r_i*a_2 + b_2

Consequently, if you setup a matrix of independents X_i for level i with columns
r_i*S_i_j, S_i_j, r_i, 1
and stack all levels below each other in
X=[X_1',X_2', X_3', ..., X_m']'
and
Y=[P_1', P2', P_3',...,P_m']'
we have the typical linear regression equation
Y=X'[a_1, b_1, a_2, b_2]
that we can simply solve with LS regresion (or some other method we like).
Due to the equation structure, monotonicity is guaranteed.

cyclingflow, 2020-11-23:
The problem so far is the lack of linearity of the slopes
[image]
Vertical = Slopes, Horizontal = resistance values.
(NB, the 'send' resistance values have a slightly higher correlation with slope and intercept, than just integers, or the received resistance values)

WouterJD, 2020-11-23:
> Regarding your post
> [image]
This shows that realized power = requested power (to an acceptable level)

I have the following proposal:
- Since the power curve logic in your fork is satisfactory now (see above)
- Please merge the current Fortius code, so you can export the JSON file
- Execute the test as described https://github.com/WouterJD/FortiusANT/issues/143#issuecomment-730326732

I will then take the figures of the JSON file and create the required formula's.

--
**I have also made a decision**:

@cyclingflow; great that you have merged the antifier code, so we are able to drive the flow succesfully now.

I also want to be honest: to save a lot of time at my side; this is the way I want to go.
I do not want to invest time in the antifier formula's anymore; I have done that last year and now again and I do not see the logic nor support the complexity. 
A simple formula works as confirmed by all Fortius and legacy/flow users, see also Golden Cheetah and TotalReverse.

So that is the way I want to put time in.

cyclingflow, 2020-11-23:
**I incidentally made the same kind of decision last night.**

If i understand you correctly, in a different direction.

I don't want to invest more time in bringing down the14x2 factor approach to 4 or even less factors/parameters.
I have spend significant time (the final formula is more simple, the estimation procedure certainly is not).
It's not worth it.
If i use the two methods to estimate the power values of all my calibration runs:
- 14*2 parameters: average residual 3.6W, max residual 11W
(the approach has .999 correlation between observed and calculated power values, for each resistance level)
- 4 factors:              average residual 11W, max residual 35W-45W (depending on estimation procedure)
That's way worse. I can mathematically guarantee you that using your formula - only one factor, if we discard the difference in roll diameter- it will be even significantly more worse. Won't bother you now with the details.

What's even worse, the original fortius approach will force the users to using the realistic grade to power approach, that's way beyond the capabilities of this trainer, as is now clear to me.

**Suggestion:**
I can understand you not willing to spend more time on it, but why not accept the 1901 class as it is? May be accept it partly as a black box.
I don't see a principal problem in using 28 parameters rather than 1 or 4, since the code is already there,
an straightforward. 
Is it that terrible? i'm willing to do some more clean-up.   
Anyway, this is what i do for a living for a large part, providing accurate or fast algorithms /estimates/intelligence for other peoples much larger software solutions.

So i wont be using it. If you want to proceed with your formula's, i can provide you with optimal values.
No need to do another calibration run. Fed up with it, i have done many :-)
The next couple of days of would like to do some normal workouts and rouvy rides for a change.

**Big Compliment**
I think you have done a great job, and a lot of work.  The current program is way more thorough in its approach to communication protocols,
has many more features, and is very nicely setup and programmed, with exceptional documentation.
The fact that i was able to quickly adapt it to my needs, and provide a 1901Trainer class, adding command line arguments,
changing the background picture, it is all a testament to that.
Thanks!

cyclingflow, 2020-11-23:
**The Future**

Ride on the Virtual Trainer! Send in my powercrank for recalibration, and _then_ may be revise my power factors.

**I anybody is willing to do a calibration run with good power meter and bike computer providing fit files, great.
What is required is simply a flow with  a bike computer with speed exactly matching the trainer speed.
I could give you instructions. THIS WILL HELP OTHER FLOW/IMAGIC USERS**

May be work on an acceleration factor. Just for the fun it
I  now have the mathematical framework. Although i believe differential equations would be the way to go, i am thinking of something simpler. The main problem i anticipate is inaccuracy in the USB protocol for speed.

Investigate a little the resistance inconsistencies i experience. Slip? power meter? brake?

Make a second much smaller GUI window, as an overlay to Rouvy or Zwift. Or decide that i do not need it anyway, the program is doing its job :-)

**Hints for understanding antifier 14 equation approach**
Wouter, if you dont want to spend any more time on it i understand. But if you ever want to understand the approach, two tips.

1. your comment
`who proves that the two formula's CurrentResistance2Power_antifier() and TargetPower2Resistance_antifier() are each other's opposites (as I think they should be)?`
In a strict sense: false. Incorrect assumption. The 1901 brake needs you the convert the required calculated resistance to 
the best out of limited set of 14 resistance values. If you don't, i believe it does it for you. Also note that the scale of resistance values _send_ to the unit, is different from the scale _received_. (Probably because the other brakes have a wider range.)
When converting that resistance value into power, it will be different, most of the time, because the actual resistance value is way different

2. You have to completely forget about resistance values for the approach to make sense. John/I assume that having set one of the 14 resistance values, the brake will be maintaining that resistance (might be false, something _is_ going on on my set). Because resistance is constant, its no longer needed for the formula estimating Power, it will be implied in the factors. I suspect, it is also the reason you had trouble converting it to the Fortius. I could gave you an approach to do that, but you are satisfied as is.

cyclingflow, 2020-11-23:
Will be providing last release tonight or tomorrow.

WouterJD, 2020-11-23:
@cyclingflow thanks for a lot of explanation
It's not that i do not want to spend time anymore; I will not spend it in antifiers formula's. 
I do not see any  justification for the 14-values approach, which is not documented anywhere why that would be a way to go.
Enough said on this subject.

For the test to be done, a powermeter is required,. So we are at a temporary stop here.
But next steps can be done when repaired.

One last note for now: mind that the T1901 brake on the T1902 headunit works fine, it does not on the T1932 headunit!

cyclingflow, 2020-11-23:
OK good to know. I obviously have no idea what would be the difference between the 1902 and 1932.
So what i describe as behavior of the 1901, might be the 1932.

BTW, Acceleration, if I want to know more about the formula's involved i will simply look in the literature.

WouterJD, 2020-11-23:
> (Oh, and BTW, when you have time: could you provide me with power supply details if you have them?
> I got a bargain on a Fortius yesterday, 50 euros, but without power unit. Fine with me, if i don't [...]

Power unit = T1941.50 Cosmos / Fortius 210-240V 50Hz 300W
Succes; many sets@ marktplaats

cyclingflow, 2020-11-23:
thx.
I'am hunting for an unit, but bidding wars for complete sets.

Wondered if you new something about the pin configuration.
May be 2 for power motor unit, 2 more for powerchoke?

WouterJD, 2020-11-24:
Refer to totalreverse; fortunately i can limit to sw untill now😉

switchabl, 2020-11-24:
The connector should be Hirschmann CA series (https://catalog.belden.com/index.cfm?event=pd&p=PF_CA3GS). The pin to the right of the notch is PE (protective earth). It is actually marked with an earth symbol if you look really closely. The other pins (1-3) go to the motor windings. I'm pretty confident the motor is a permanent-magnet synchronous motor. Not sure about the configuration though. I would have assumed it would be three-phase, but the inductance between the pins is not the same for all pairs, so maybe not. I never seriously considered making my own control unit, so this is not something I ever explored in detail.

WouterJD, 2020-11-24:
@cyclingflow check this out:
https://link.marktplaats.nl/m1629932165

cyclingflow, 2020-11-25:
>  I never seriously considered making my own control unit, so this is not something I ever explored in detail.
May be not wise to do so, but i always wonder. Thx for the info.
This tells me the other route is much faster:

Right now i have sealed the deal. WIll be getting 2 different head units, 2 motors, 1 power supply.

cyclingflow, 2020-11-25:
@WouterJD 
regarding your remark about head units being interchangeable

I have also calculated the PowerResistanceFactor as you use it, for each resistance level of the brake,
given the speed and power of calibration runs. Both the set of send target resistance values, as well as the set of received current resistance values give large variation for this factor.
I.E. 
[image]
What is supoosed to be a single factor, is highly variant over resistance factors.

But i have been thinking of revising those into an other set of resistance integers, such that we could unify it to a single appoach, with a function using this resistance value.
Not a question of much less parameters, but it would help recalibrating, and may be is more appealing to you.

cyclingflow, 2020-11-25:
**I have submitted an new release, with latest factors, and both the 14*2 and 4 factor model.
It also has a hidden feature an new button on top of the locate button, that will change the window into a much smaller one, with the STAY_AT_TOP flag. (It can also display 6 additional parameters if i need to.)**

cyclingflow, 2020-11-25:
The small screen was very useful to monitor while riding Rouvy.

What i noticed was significant slipping of the wheel: lower speeds for the app as from my bike computer, where i started with good calibration. When driving at moderate constant speeds, at low resistance, it is fine. At higher resistance levels, speeds are too low.

Nothing new of course, one of the reasons they went to direct drives, also allowing for higher flywheel weights.

But i will be increasing the tire resistance. I have always used the flow at higher calibration factors (say 6-8).
Was trying around 1 (manual head unit), to provide others with normal factors, but does not work for me.

WouterJD, 2020-11-26:
Hi @cyclingflow thanks for persistance on the subject.
I'm on it as well and continue to study your achievements and will respond to some questions you ask.
@BikeBeppe64 has conducted a test, using https://github.com/WouterJD/FortiusANT/tree/#143-Powercurve-iMagic

**Question here: who could conduct a similar test?**
- - -
The test is performed as follows:
-	FortiusANT is operated in manual mode (-m flag), modifying the target power from 10...4000 in 50 Watt steps (read next step!).
-	The power as selected is transferred directly to the i-Flow (Resistance = TargetPower) without intermediate formula. (The -r flag is created for this purpose). The assumption is that the i-Flow allowed resistance ranges from 1000…4000 and that is not Watts!
-	The data from FortiusANT is saved as json-file (-d127 flag, creating a full logfile).
-	A bicycle is used with power meter and Garmin Edge, registering speed and power (and cadence, heartrate). This data is available as FIT file and converted to TCX.

-	FortiusANT and Garmin Edge are started.

-	Initial TargetPower in FortiusANT is 100Watt ( Resistance = 150)
-	Start cycling at 12km/hr.
-	Increase TargetPower on the headunit every 10 seconds until 4000 (or that maximum power is reached). This will take 79 steps = 13 minutes.

-	Reduce TargetPower to 1000 (I think that is the minimum value, which we will see from the first test). Note that the head unit is sampled every 250ms, so you can press two times per second only! (It’s not as fast as a keyboard).
-	Redo the test at 24 and 36 km/hr. 

-	Stop FortiusANT and Garmin Edge and save .log, .json and .fit files.
- - - 
Unfortunately, it **seems** that his Flow does not respond well to (any) resistance setting.
I have asked @totalreverse for his view on it and liked to share this

WouterJD, 2020-11-26:
> What i noticed was significant slipping of the wheel

Note that Fortius does not work at low speeds. I always ride in the highest gear to obtain maximum wheel speed even at low cadence. See manual 5.1. There is a clear reason for direct drive trainers.

WouterJD, 2020-11-26:
> OK good to know. I obviously have no idea what would be the difference between the 1902 and 1932.

Difference is the USB-protocol (see @totalreverse).
Untill recently I assumed that the head unit makes the brake transparent; that is untrue.

See  https://github.com/WouterJD/FortiusANT/tree/#143-Powercurve-iMagic where I detect what brake is behind the headunit.
So as soon as we know the algorithm, we do not need a command line parameter to know whether there is a Motor brake or Magnetic brake.

cyclingflow, 2020-11-26:
> See https://github.com/WouterJD/FortiusANT/tree/#143-Powercurve-iMagic where I detect what brake is behind the headunit.
So as soon as we know the algorithm, we do not need a command line parameter to know whether there is a Motor brake or Magnetic brake.

Great

> The test is performed as follows:
I'am sorry. That protocol might not be efficient  and will probably go through the same resistance values multiple times,
because there are only 13 values accepted, rest will be converted.
[1900, 2030, 2150, 2300, 2400, 2550, 2700, 2900, 3070, 3200, 3350, 3460, 3600, 3750]
So it would be better to go through these directly.
If ones uses my version, and chooses manual grade mode (-M) the button will allow you to go through
these resistance values, because the grade % will select different resistance values, displayed right
next to the grade %

Furthermore, one speed is not enough. you need at least two speeds, otherwise no intercept, 
(which is not only included in the antifier approach, but also in your legacy code.)
i go through 10kph 15 20 25 30 35 and 40  IF doable.

A minimal scheme would be
[2030,  2700,  3600] * 15kph 30 kph
this would give enough for a regularized  4-factor approach (or 5 if we take a quadratic approach
to estimating the intercepts, as you have done in your legacy code)

>-d127 flag
Good to know. That was what i was missing. Thought there was an specific 'test log' debug flag.

cyclingflow, 2020-11-26:
1. I have just explicitly tested again the target vs current resistance value. It only converts to the closest accepted value,
if you have wheel speed> 0

2. I'am thinking of calibrating against the original manual unit power values. It is a bit of a hassle,
because i have to manually read them and write them down.
May be this weekend.

switchabl, 2020-11-26:
> 
> 
> > What i noticed was significant slipping of the wheel
> 
> Note that Fortius does not work at low speeds. I always ride in the highest gear to obtain maximum wheel speed even at low cadence. See manual 5.1. There is a clear reason for direct drive trainers.

But wasn't that on a Flow? There should be no "motor slips" on the Flow, so it sounds more like a tyre pressure/knob setting problem?

Btw, the direct drive advantage is not as clear cut as you think. The problem is mainly that the Fortius motor (or motor controller) can't produce enough torque at low speeds. The Genius seems to do much better at this. The first two NEOs had similar problems, though likely not as bad. The new NEO 2T is supposed to have fixed this. 

Why did the NEO have those problems despite its enormous motor? Well, if you drive the motor directly, instead of through a ~670mm wheel and a 40mm roller, your motor will turn more than 16x slower and needs to deliver 16x more torque. So unless you add a gearbox or belt drive (like Wahoo), getting this right on a direct drive is _a lot_ harder. Even more so since the lack of a mechnical flywheel increases the peak torque even further.

Tl;dr Main technical advantages of direct drive are power accuracy (no changing tyre pressure with temperature) and noise (lack thereof). The latest NEO does high gradients despite being direct-drive, by having an enormous motor and a powerful motor controller.

cyclingflow, 2020-11-26:
> But wasn't that on a Flow? There should be no "motor slips" on the Flow, so it sounds more like a tyre pressure/knob setting problem?

True, i have always run on a little lower pressure, higher knob setting, to maximize grip, while reducing vibrations. With good results. 

For the sake of interchangeability of my calibration factor with other users, i tried lower knob settings. Does not work
in my setup.

>Btw, the direct drive advantage is not as clear...

Thx for the info. Interesting. Have never really got into the higher end models. Obviously dont have experience with the 
pro's and cons.

Testing of the flow made me realize it's relative small band of operation. Not a real problem, because we can shift.
Have to shift more, or even mount a climbing cassette for VR training is a bit too much reality for me :-)

cyclingflow, 2020-11-28:
I will explain a bit about my motivation in pursuing accuracy.
**Why accuracy?**
Over the last 8 months i have using the flow along side my bike power meter. Because it is one-sided, i tried to use the flow's power readings to help me analyze differences between left and right leg. I definitely have a problem in one leg.

It became clear to me that there were some systematic differences. Generally i have best results if manually put the brake on the 7th resistance level. The differences are fairly consistent, as long as you keep checking your tire presure, and use the calibration procedures. 

a couple of months ago, the power meter changed, and differences between flow and power meter are a bit larger. Still, the power meter accurately measures _force_, at least within 3% accuracy, as i have tested using weights.

**Measurement Errors of the setup**
We have been talking and commenting about measurement erros and accuracy of the trainers in measureing power, as others have done in reviews. I would like make some clarification on the nature of these errors, after having done a considerable amount of testing my setup.

1. Your typical **Random measurement** errors are extremely small. When doing one speed run at a particular level, i have 
got extremely consistent results. Such that power vs speed is linear with a correlation of .999, granted with just 5 to 7 observation points. All the time. This is really excellent given the fact that i am measure power at the bike, with different gears and varying cadence. With a couple of minutes of warming I can't say i have seen clear indications of addtional effects within the small time of one speed run.

2. **Systematic errors** are much larger. I have had small inconsistencies when changing between resistance levels, and large inconsistencies between sessions. @WouterJD @switchabl , myself, we have made useful comments on the nature of this. 

Given these observation my personal preference is to not introduce additional approximation errors, keeping 1 a small as they are now, while working at an approach that tries to deal with 2 by using a calibration procedure. Note that the 1901 brake does not have a calibration test as a motor brake does.

cyclingflow, 2020-11-28:
@WouterJD  **Comments on current implementation and manual.**

I have to admit i'm a bit pissed off right now. After all the work i have done, you seem to ignore the advice and info of an expert on these matters. It's notthat  you did not do a great job on the program, the formula's the communication protocols. You did do a great job! It's just that you have showed a little less knowledge in this particular area.

cyclingflow, 2020-11-28:
My comments

1. You started off wrong footed. The table you mention in the manual is not from antifier. It is exactly a table i provided in an early stage, and you missed out on the accompanying info.
**Any statement you made on the natures of these tables based on this one is unfounded. _This_ table does not provide real observed estimates using the antifier approach. I manually adjusted the parameters to reguralize the problem. Everything looks smooth, because i explicitly made them look smooth.**
I did so because i was having trouble getting accurate results through antifiers power_curve.py. These were solved when using fit files.
You also ignored results i published here  showing this estimation to be at least 10% off at higher speeds and power.

2. Please, don't use imprecise statements on the 'likely' accuracy of power estimation while you are basicl just messing around with parameter estimates rather than power observation. I know they are false, but moreover, they would have been torpedoed out in a scientific review. Because, in contrast many people really appreciate some results on the power estimates. Such as:
" The current approximation of the table factors leads to power estimates that on average are 3W off" or "the maximum difference is 11 wthat at speed .." or errors are about 11% at 200-250 watt level."
Now we may differ in opinion on the practical significance of these errors, i don't have a problem with that. But it will make results comparable between our approaches. While opening up the discussion on the implications of these power estimation errors to a much wider audience, in relation to use for training analysis, or fair results in online VR sessions.

cyclingflow, 2020-11-28:
3. You have a changed the implementation of TargetGrade2Resistance(self) compared to antifier and my version. It therefore introduces an additional discretization error, by selecting the correct value only approxximately 50% of the time.
I understand you desire to put in a break statement to cut the loop short, but this is not the way to do it.

4. I'am not sure your statement in the mnual on the nature of these grades is fair to John/antifier. To clarify: these grades are ones i have manually adjusted to satisfy riding some routes in Rouvy. Indeed to allow for the limited resistance range. To be fair, antifiers approach _does_ use the same type of considerations and calculation you do, using an assumed overall eight. A simplified version of the nice gribble formula's you have implemented. It's explicitly in power_curve.py and it's comments.

WouterJD, 2020-11-28:
> I have to admit i'm a bit pissed off right now

Sorry @cyclingflow I do not appreciate your tone of voice.
I have taken all inputs (from you, and all others) very seriously and ask to execute a test with the branch I have published.

Please note that all the work I do is on-top-of the basic purpose: to make my own Fortius work; so please step down a little bit.

My suggestion is to execute a test, showing how this algorithm works in practice.

After an appropriate reaction I will read the other texts.

cyclingflow, 2020-11-28:
**My final comment. Flow's estimates as standard.**

A more regularized approach might be justified by fears of overfitting in the 14 equation approach. 

As @WouterJD has said, people have appreciated the tacx trainers, and although some critique exist on the accuracy of the power readings, they have been widely perceived as useful tools (based upon comments here, multiple reviews.)
So we can consider it a standard to some extent.

Therefore, we way simplify the extraction of formula's for the purpose of FortiusAntifer, by not looking at external bike power reference values, but also looking at the provided power readings of a device like the T1682 or T2202.
I have one, and have some results. Conclusions so far:
1. _neither a two x two factor or a 5 factor approach introducing a quadratic term - for the intercepts in the 14*2 parameterization - does completely cut it._ The approach here  is similar to the approach @WouterJD has explained in the manual for the legacy formula's, with a little improvement by getting all 5 parameters simultaenously, globally optimal (LS framework).
2. Needles to say that the 14*2 has no problem replicating these power readings,  while implicitly providing it's own 'resistance' values, rather than the one provided the unit.
3. My conclusion is, that t_he tacx is doing things slightly differently than FortiusAnt formula's_. And it confirms my suspicion that the _resistance values returned by the unit are optimized for a lightly different objective function,_ such that @WouterJD approach is compromized. Which is a pity. It could have been very simple.

Now, i have come up with a nice mathematical justification for anitfiers approach, but i wont bother. I will cease.

I simply assume our paths will diverge.

WouterJD, 2020-11-28:
> I simply assume our paths will diverge.

Indeed; if there are no test-results the suggested implementation is not proven wrong.
Thanks for your contribution; it certainly helps to understand the magnetic brake.

cyclingflow, 2020-11-28:
Our posts have crossed.
> Sorry @cyclingflow <https://github.com/cyclingflow> I do not appreciate ...

I will apologize. And I will step down.

> I have taken all inputs (from you, and all others) very seriously..

Well i might have misconstrued your intentions after reading the new
additions to the manual.

> Please note that all the work I do is on-top-of ..

Same for me. And I do recognize that you have put in a lot of work in
the project, while i have been (rather busy)
with it for just one week.

I actually assumed we both just wanted to help out a few fellow users.

> if there are no test-results the suggested implementation is not proven wrong.

This really confuses me. I have done a lot of test runs, simply did
not provide you with the log files.
I did not know how to get them at that point.

I honestly believe, i have presented results that attest to the
accuracy of several
approaches, including results on average differences between estimates
and reference powers,
and the estimates of parameters.

You could calculate and compare any approach with these results,
without me ever riding the bike again.

[17 more replies]

closed 2020-12-07 by WouterJD