WouterJD/FortiusANT #144: Calibrate does not work well for Tacx e6be T1942

WouterJD, 2020-11-17:
Hi @huishuis, I created this issue for you

I have a question which might be caused by me misreading the manual, I'm using a e6be T1942 headunit starting Fortiusant with the standard parameters. It loads the firmware and is ready for use. According to the manual it should now be ready for callibration which 'starts when you turn the pedal as if starting to cycle'. The issue is that the callibration won't start. I can cycle and the GUI indicates speed, cadence and power and it works nicely in Zwift. 
The 'callibrate' function seems to become partly active though once i press 'stop'. When i now turn the pedal, the motor will kick in and the wheel rotates at 20 KM/h.   It will however only stop again once i press 'start' and 'stop' again. 

_Originally posted by @huishuis in https://github.com/WouterJD/FortiusANT/issues/14#issuecomment-728304486_

WouterJD, 2020-11-17:
Sounds strange and _should_ work.
Please provide a logfile (-d127) so that I can check the commands sent to your Tacx

switchabl, 2020-11-17:
I still have one of those (but normally use a T1932) and I can confirm this behaviour. Log is attached.
I seem to recall vaguely that this didn't work in TTS either with a T1942 and it used run-down calibration instead? But I may be wrong.
The curious question is why it goes into calibration mode when you press stop.

[FortiusANT.2020-11-17 16-05-24.log](https://github.com/WouterJD/FortiusANT/files/5554673/FortiusANT.2020-11-17.16-05-24.log)

huishuis, 2020-11-17:
Attached my log as well. I can confirm  the behaviour mentioned by @switchabl. It does not go into calibration mode. This is the sequence of events that generated the log:
1. FortiusAnt.exe -g -H0 -A _(so without the auto start parameter)_
2. Press **Locate HW**
3. Tacx Fortius is found 
4. Turn pedal
5. Motor spins
6. Press **Start**
7. Press **Stop**
8. Motor stops spinning
[FortiusANT.2020-11-17 18-36-26.log](https://github.com/WouterJD/FortiusANT/files/5555595/FortiusANT.2020-11-17.18-36-26.log)

On the original Tacx Fortius software (so not TTS, but the older software) the calibration works as expected.

WouterJD, 2020-11-18:
Two improvements made:
- calibration is now supported for 1942 head unit
- calibration should not be started in idle-mode but when after start-button is pressed (error introduced in version 3.0)

Please test version 3.7 and let me know

huishuis, 2020-11-18:
I just tested it and it works flawlessly. Thanks Wouter!

closed 2020-11-19 by WouterJD

Rendier8, 2020-11-19:
How long does it take to calibrate?

WouterJD, 2020-11-20:
@Rendier8; Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
> How long does it take to calibrate?
Some minutes