WouterJD/FortiusANT #147: run off procedure

cyclingflow, 2020-11-20:
Due to all calibration work, i regularly apply the runoff procedure, and encountered some minor issues:

FortiusAnt selects powermode for the runoff. This might be a good idea for the fortius, however, for the 1901 brake I do not see why you would want to make starting the wheel harder, as well as change the resistance constantly during rundown.
The last also implies that runoff values are dependent on calibration factors, which i would like to avoid.

So, i have implemented a change in my fork, such that when the 1901 brake is selected, it switches to grade mode for the runoff.
I am currently experimenting with the lowest resistance level for the runoff, reasoning that we only want to measure tire/wheel resistance and flywheel effects, not brake effects.

cyclingflow, 2020-11-20:
Furthermore 3 **minor** convenience issues:
1. Max Speed I prefer a lower max speed. 40kph requires gearing up to largest blade on my triple (26inch wheel), which i like to avoid. 30 is the original tacx maximum. Currently I use 36.

2 Over the years i have regularly find myself wondering, did I cleanly stop pedaling in time? I propose a change such that
after reaching the max speed, the program signals STOP PEDALLING, and then after a slight further dip say 2kph (40-2=38kph),
the programs starts measuring and tells you so.

3 Minimum speed. I have noticed several times that the behavior of the wheel just before stopping is inconsistent (may be smudge or something), which made be start wondering if stopping measuring slightly above zero speed would be better.

Anyway, I implemented these changes in my fork through an additional command line argument, such that normal program behavior is not changed, but you can specify a different runoff procedure to your liking.
-r(maxSpeed,dipSpeed,minSpeed,targetTime)
the last parameter simply displays _your_ target time needed to get consistent results across sessions.

WouterJD, 2020-11-20:
Good ideas. I suggest to get the powercurve right first and then pick-up all other suggestions to create ani-Magic release

cyclingflow, 2020-11-20:
Working hard on it, requires quite some testing. But close to release.

WouterJD, 2020-11-20:
I see it; I'm studying your code now

switchabl, 2020-12-11:
Found this only now. This is already very close to what we need for the rolling resistance calibration. Basically:
1) Set brake to lowest resistance (not grade or power mode)
2) "Pedal at X km/h"
3) "Stop pedaling!"
4) Wait for speed to drop below Y km/h, then start timer.
5) Wait for speed to drop below Z km/h, then stop timer.
6) Calculate estimated rolling resistance value from time and display that (also maybe display "warning: too high" or "too low" if we can decide on a sensible range)

We will need some more data eventually to decide on a) good values for X, Y, Z b) the formula to use for time -> rolling resistance, but that test protocol will be a lot shorter and simpler than for the power curve.

WouterJD, 2020-12-16:
Implemented in Version 4.1

closed 2020-12-16 by WouterJD