WouterJD/FortiusANT #148: i-Magic Problem

kiwisibk, 2020-11-20:
Hello,

I followed everything from the tutorial, the VR-interface looks installed fine.

[image]

But FortiusANT is still not recognizing my interface/tainer... :(

[image]

What I'm doing wrong??

Many thanks for any help!!

BR
Arno

cyclingflow, 2020-11-20:
Wrong driver type. Your VR driver should be listed under libusb32-win devices. 

You probably will have to uninstall this driver (also select removing driver to be shure). (However, your tacx software will no longer work)

TIP: there are detailed instruction somewhere, i forgot. 
But if you download the libusb drivers and unpack, there will be a tool in the BIN folder called inf-wizard.exe.
Use it to make a custom ".inf" file for your vr-device, that allows you to provide a name ("VR-interface") such that it is listed
as "VR-device" under linusn-win32 in your device manager.

Otherwise, it will be listed as USb stick, leaving you with 3 such devices; confusing.

cyclingflow, 2020-11-20:
And you might want to download my FortiusAnt version from my fork. See issue 143.

kiwisibk, 2020-11-20:
That was exactly the issue! Thank you so much! Works like a charm now! Great 😊👍

WouterJD, 2020-11-20:
@kiwisibk hi; please leave a note how I can improve the instructions in the manual
Thanks!

kiwisibk, 2020-11-20:
Good point! Just add a tutorial how to enable unsigned drivers on Windows
10. And then, the process is following: connect the i-magic interface, go
to device manager, find the i-magic interface (almost on the top
highlighted with an erroneous driver), click on update driver, select
locate on hard drive, and select the folder where you firstly created the
.inf usblib driver you already are describing in your tutorial.

Thats it! 👍

Thanks once again for your outstanding work!! 😊

BR
Arno

P.S. if you want to send me a Word file of your tutorial I can edit it for
you and send it back for approval!

switchabl, 2020-11-23:
@kiwisibk It should not normally be necessary to install unsigned drivers. Did Zadig not work for you?

In general, I think using Zadig (https://zadig.akeo.ie/) should be the recommended way to install libusb-win32 drivers (it is for many other projects that use libusb). It comes with signed drivers, generally just works with a couple of clicks and doesn't require creating custom .inf files either (well, it does create them, but it all happens automatically in the background). It also has the option to replace the Tacx drivers directly.

The only thing to be careful about is selecting the proper driver ("libusb-win32"), because it does other libusb flavours as well.

cyclingflow, 2020-11-23:
Good point. Did not get around to using it before. But does work like a charm.

closed 2020-12-01 by WouterJD