WouterJD/FortiusANT #149: no connection with Tacx

MTBKL, 2020-11-21:
Hi,
i have a question, i like to use my fortius in combination with Zwift but FortiusANT does not recognize the hardware.
I am using an Tacx T1941.50 base unit woth a  t1932 USB connection box. Is this hardware supported?
The PC is a laptop running Windows10 (64bit).

WouterJD, 2020-11-21:
Yes it is see #148 whoch might help

MTBKL, 2020-11-22:
Wouter,

thanks for your quick response, i tested with the usb drivers as written in issue #148 and now FortiusAnt makes a connection with the Tacx :-)
I need to hook up the sensors and then i expect also the speed meters will show values.
Thank you !!

WouterJD, 2020-11-22:
Great and Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

closed 2020-11-22 by WouterJD