WouterJD/FortiusANT #150: Possible to use old Tacx 32-bit head units on 64-bit Windows 10?

BrRoBo, 2020-11-24:
Hi Wouter,

I have a couple of the original, first generation Tacx trainers... the green I-Magic Head unit 0x1902, and the first model of the Fortius blue head unit (which you have dubbed the "0xe6be").  

These trainers are currently connected to an old Windows XP machine that is used to run the old blue Fortius software and TTS1 software along with a long list of RLVs (ride videos) installed.  It also has Fitcentric v2 still working.  Lots of archaic stuff, but all functional to serve up the dinosaur smart trainers.

I would like to use a newer Windows 10 64-bit laptop for running Zwift and FortiusANT on the same, single machine, but not sure if the latter USB part will work in this Windows version.  AFAIK, both of these head units can use only 32-bit USB drivers... correct me if I'm wrong, so I am not sure if this single machine setup can work.    

A typical workaround may be to use the old XP as the 32-bit USB driver handler, but I would like to get away from needing this obsolete machine.  

Thoughts?

WouterJD, 2020-11-24:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
32/64 bit issue does not  affect functioning of USB.

martin-vi, 2020-11-24:
The Tacx software (TTS1) has been compiled for 32bit and therefore no longer works under Windows10. The protocol that the trainers speak is not interested in which CPU architecture is used. FortiusAnt implements the USB control completely by itself and therefore the software can basically run on any architecture where Python is running.
For example, I run the 0x1902 Head Unit on a RasberryPi with FortiusAnt. This is a 32 bit ARM CPU.

BrRoBo, 2020-11-24:
> 
> 
> The Tacx software (TTS1) has been compiled for 32bit and therefore no longer works under Windows10. The protocol that the trainers speak is not interested in which CPU architecture is used. FortiusAnt implements the USB control completely by itself and therefore the software can basically run on any architecture where Python is running.
> For example, I run the 0x1902 Head Unit on a RasberryPi + Linux with FortiusAnt. This is a 32 bit ARM CPU.

Thanks martin.  Yeah, I never expected to run the old Tacx software in Windows 10 or on a 64-bit OS.  My concern was that these older, first generation head units maybe only worked on a 32-bit OS... because Tacx did not publish 64-bit drivers for this units but instead came out with new models.

I'll give this FortiusANT solution a try.

Regards.

WouterJD, 2020-11-24:
> I'll give this FortiusANT solution a try.

You'll be amazed; Zwift, Rouvy, Trainer Road with your old tacx👍

closed 2020-11-25 by WouterJD