WouterJD/FortiusANT #151: Tacx keeps running after calibration and run off

sofiedp, 2020-11-24:
Dag Wouter,

Bedankt voor deze software, maar ik heb nog een probleem waardoor er iets misgaat met de power.
Na de kalibratie en de run off blijven de tacx rollen aangestuurd aan ongeveer 7km/u. 
De weerstand blijft ook op 100W staan. Ik kan wel fietsen met zwift maar dan blijft de weerstand ook hoog. Ik doe waarschijnlijk iets verkeerd. Ben ook niet echt goed met software enzo. Hopelijk kan je mij helpen..
Vriendelijke groeten,
Sofie

WouterJD, 2020-11-25:
Hoi @sofiedp 
Welkom in de FortiusANT community
- - -
Ik ben altijd nieuwsgierig met wie ik communiceer, waar FortiusANT wordt gebruikt en in welke configuratie.
Welke bundle heb je gekocht, en welke rem en head-unit gebruik je?
Vertel meer over jezelf, eventueel onder issue #14.
- - -
Voorgaande helpt ook om antwoord te geven op je vragen.
Heb je de nieuwste versie gedownload? Voorgaand was er een issue met calibratie.
Is het dat niet, doe dan een test met de debugging optie aan (-d127) en stuur de logfile even terug

Groeten Wouter

sofiedp, 2020-11-25:
Dag Wouter,
Ik ben Sofie De Proft uit Dworp in België en heb een Tacx fortius rem T1941
en head unit T1932.
Daarnet heb ik eens de laatste versie gedownload, toen ik dan opstartte
leek het probleem met de kalibratie en run-off opgelost.
Als ik in Zwift reed bleef de power weer op 100Watt staan. Verandert deze
normaal automatisch of moet ik zelf nog iets aanpassen?

Daarna leek het erop dat ik de connectie met de power verloren was en heb
ik de FortiusAnt opnieuw opgestart en had ik het probleem met de kalibratie
en run off weer. Hiervan heb ik een log file in bijlage.

Vriendelijke groeten,
Sofie

WouterJD, 2020-11-26:
Hoi Sofie, de Fortius zou goed moeten werken inclusief de calibratie.

Ik zie geen log in de bijlage; wanneer je antwoordt via mail komt er geen bijlage mee.
Daarvoor moet je de bijlage uploaden in github

sofiedp, 2020-11-28:
Hej Wouter,

Vandaag werkte kalibratie weer goed.
In zwift blijft de weerstand wel 100W.
Voorziet het programma dat de weerstand aangepast wordt volgend zwift?
Mss scheelt er iets met mijn power meter.

Vriendelijke groeten,
Sofie

sofiedp, 2020-11-28:
Dag Wouter,

Ik ken niks van ant+ maar ik heb zwift op mijn gsm. Die kan blijkbaar ant+
signalen lezen zonder extra dongle. Maar mss kan deze geen uitsturen?

Mvg
Sofie

sofiedp, 2020-11-28:
Dag Wouter,
Sorry voor het storen.
Maar het werkt nu :). Ik heb zwift op een andere pc geïnstalleerd.

Vriendelijke groeten,
Sofir

sofiedp, 2020-11-28:
werkt idd goed! Merci!!

closed 2020-11-28 by sofiedp

WouterJD, 2020-11-28:
Tof. Dit is natuurlijk het mooiste dat de problemen zich vanzelf oplossen.
Ik hoor wel goed het verder gaat!