WouterJD/FortiusANT #152: Please implement ZWIFT steering

WouterJD, 2020-11-25:
The Blacktrack steering thing was also included with my Genius. For now that is maybe not so interesting, but once Bluetooth is available, we could emulate an Elite Sterzo Smart to get steering on Zwift (https://github.com/kwakeham/zwift-steerer/).

_Originally posted by @switchabl in https://github.com/WouterJD/FortiusANT/issues/101#issuecomment-733266357_

This might be usable for Fortius VR with steering unit as well.
FortiusANT already knows the steering movement, but I did not know what to do with it :-)

WouterJD, 2020-11-27:
To be honest: perhaps using the Zwift Companion app is the way to go.
- It is supported by Zwift
- Do not build what already exists

switchabl, 2020-11-27:
If you mean the one where it uses the phone sensors on the handlebar: that was only a tech demo on a short mountainbike loop. Apparently it doesn't work very well because on many phones it drifts quite badly and it looks like they have more or less abandoned it. At least for now, steering on normal Zwift "roads" works only with the Sterzo unit (https://www.elite-it.com/en/products/home-trainers/trainers-accessories/sterzo-smart). They have an exclusive deal with Zwift until some time next year I think.

On the down side, the protocol is proprietary and works only on Zwift. It is also more complicated than it should be (probably on purpose). But it has already been reverse-engineered and it would give all those Fortius/Blacktrack steering frames out there some purpose.

switchabl, 2020-11-27:
Just collecting some links here for future reference:
https://github.com/fiveohhh/zwift-steerer
https://github.com/kwakeham/zwift-steerer/
https://hackaday.com/2020/09/11/a-deep-dive-into-the-sterzo-steering-plate/
https://titanlab.co/zwift-steering-sterzo-pwned/
https://www.youtube.com/watch?v=BPVFjz5zD4g

I don't think it makes sense to begin work on this before BLE support is stabilized. It is also somewhat low priority for me. I do have an original Sterzo I can use for debugging/comparing purposes.

WouterJD, 2020-11-27:
If you build it; I'm happy to test and integrate

BrRoBo, 2020-12-14:
In addition to the Fortius I currently have hooked up and using with FortiusAnt, we also have a i-Magic steering frame in storage.  I always found steering while riding was more of a hassle than a benefit because your Avatar would not stick to a route and so your ride results would not match any other historical results.

Were I do think the steering would be nice to use is if Tacx (Phil Stone, etc) would implement a suggestion I made years ago to do 360 videos and allow you to turn your head (pan the view) around while riding.  Apparently Phil tried mocking up a 360 video as a test, but it made him nauseous, like motion sick :/

switchabl, 2020-12-14:
On Zwift it is more "veering" than "steering" (except for the MTB trail where you actually have to steer to stay on track)., which I think is mostly a good thing. So if you don't do anything, you just stick to your line. It takes a bit getting used to, but I think it is quite fun on the more twisting routes to go for the tighter lines. Adds a new dimension to racing too (which you may or may not like). Overall certainly something you can live without, but if we can let everyone with an old steering frame/Blacktrack try it for free, why not?

This needs to wait until we have Bluetooth though, there is no steering profile for ANT.

marcoveeneman, 2020-12-21:
Just want to let you guys know i'm looking into this atm

WouterJD, 2020-12-21:
:-)

WouterJD, 2022-08-16:
@marcoveeneman has created the first version, using NodeJs
@switchabl has added blacktrack
It's now integrated, also in -bb (ble/bless).

Ready for testing

closed 2023-03-07 by WouterJD