WouterJD/FortiusANT #154: Problems with calibraion perhaps related to #144 - Brake 1941, head unit 1942

jujfalus, 2020-11-27:
STDOUT shows 

"Calibration stopped because of unexpected resistance value"

Power values show in gui and in zwift are consistent each other, however powertap measurements are about 50% of the taxc/zwift power values.

I hope to create a spreadsheet of tacx values vs garmin values, however have not collected all the date.

Attached is -d127  output log.

Not sure if this interelates to #144. I pulled the most recent windows exe this morning (3.8).
[FortiusANT.2020-11-27 21-51-02.log](https://github.com/WouterJD/FortiusANT/files/5609692/FortiusANT.2020-11-27.21-51-02.log)
[FortiusANT.2020-11-27 21-51-02.json.txt](https://github.com/WouterJD/FortiusANT/files/5609693/FortiusANT.2020-11-27.21-51-02.json.txt)

jujfalus, 2020-11-27:
btw thanks for your time. This must keep you busy.

jujfalus, 2020-11-27:
Attached is a spreadsheet showing zwift and gramin power. Much of the time zwift (frotiusant) power is 200% of powertap (garmin) power. Although not consistently 200%. I suspect that zwift is in slope mode and 200% is when slope=0. The course was fairly flat.

[FortiusANT-Power-154.xlsx](https://github.com/WouterJD/FortiusANT/files/5609803/FortiusANT-Power-154.xlsx)

jujfalus, 2020-11-27:
Also note that Cadence value was > 150 sometimes as high as 200 for one or two seconds.

I had configured zwift to use the tacx-power, tacx-cad and tacx-controlled.

Thank you for all your investment of time into this product. Not sure if there is some way I can help in debugging. Lots of troublshooting and programming experience. Although python is fairly new to me.

jujfalus, 2020-12-04:
Hi, I now understand the 

"Calibration stopped because of unexpected resistance value"

I was mistakenly pedaling for more than a single stroke, which was messing with the calibration. 

Now that I understand the calibration is working. I see that "Counter" is now only getting to 4 before the resistance values become negative which is < the Counter limit of 10.

After all that I find that even after the calibrate cycle where fortius goes to ~20km and takes Resistance value samples the fortius (redid a couple times first time Resistance value of -140 second time resistance value of -110) estimated power is still about half of the powertap values. I'm hoping to better understand the code so that I can get the powertap values and perhaps either log them to debug log or even present them within the gui. Also wondering about exposing the powerfactor parameter, so that its adjustable via the gui, so that it could be tuned on the fly.

WouterJD, 2020-12-10:
Made some adjustments to documentation
Calibration was improved 

Closed for now

closed 2020-12-10 by WouterJD

jujfalus, 2020-12-10:
I'm hoping to modify the code so that I can log timestamp, tacx power and powertap power all at more or less the same time. I'm hoping that the data will yield some insight into why the power is so far off for me. I have not had my powertap recalibrated, but I think its values are quite reasonable and consistent with my second powertap wheel. Also have you seen some spurious cadence values?

WouterJD, 2020-12-10:
I thought of it: in stead of broadcasting power, pair to a powermeter and resister the received value in the JSON file.
Did not have time to do it and I have no power meter to test.

jujfalus, 2020-12-10:
Hi, I have run zwift using power controlled by fortiusant and power from powertap. Which is fine if I'm just riding around. However in ERG mode this will not work. Maybe, I'm not understanding what you were proposing.

WouterJD, 2020-12-11:
Did not intend to suggest a proposal; the idea was there but I will not build it shortly