WouterJD/FortiusANT #155: Follow up to #111, perhaps document solution

BrRoBo, 2020-11-29:
re: "GetTrainer - Unable to load firmware" on 1942 head unit #111

I seem to have had the exact same issue as "bradyap", but found a quick solution to my case.  Brady's screenshots are identical to what I was seeing and we had the same head unit... T1942.

What I think happens is that we do the lib32 USB driver install and successfully see the Fortius unit listed under libusb-win32 devices along with how-ever-many Ant USB stick 2 are connected.  Then when we run the FortiusAnt application, it seems to be looking for a different device or registration, causes the Fortius to unload from the list of "libusb-win32 devices" and then there is a new "Tacx Fortius" listed under "Other devices".  This unregistration, re-registration cycle is reproducible over and over by disconnecting the Fortius USB, then plug it in to load the lib32 driver and see it properly listed, then use the FortiusAnt "Locate HW" which again will cause the Fortius to get re-registered under "Other devices".

The simple solution at this point is to open the properties for the "Tacx Fortius" listed under "Other devices", then "Update drivers", select "libusb-win32 devices", then "Tacx", then select whatever name you used ("VR-Interface", "Fortius Virtual Trainer", etc).  Bingo, the Fortius head unit is reloaded into the correct list of "libusb-win32 devices".  Issue fixed.  I think these steps should be added to the installation guide section.

WouterJD, 2020-12-10:
added to manual

PeterKassenaar, 2021-01-15:
Hi @BrRoBo,

While installing FortiusAnt, I just ran into this problem ("Unable to load firmware"). The Tacx Trainer in my case was also unregistered in the libusb-win32 devices tree and then re-registered in the Other Devices tree. So I was also stuck in this cycle.

Using your suggestions, I was able to solve it. Thanks! 

However, it looks like you missed one (important) point, after the "Update drivers" step. You'll also have to choose "Let me pick from a list of available drivers on my computer" in the next step, otherwise you'll never see the "libusb-win32"  option.

For clarity I made screenshots of the complete process, hopefully other people that also run into this cycle can benefit.

1.  Tacx Fortius was first correctly installed, but after starting FortiusANt re-registered as Unknown Device.
[image]

2. Rightclick, then "Update Drivers". Select "Let me pick from a list..." in the next screen (this step was omitted in the solution above).

[image]

3. Select category "libusb-win32 devices"

[image]

4. Select Tacx, Tacx Fortius (or whatever you called your trainer) and then Next.

[image]

5. Click "Yes" in the Update Driver Warning dialog box.

[image]

6. After that, the device should be working properly. Click Close.

[image]

7. Tacx is now listed in the correct category. You can start FortiusAnt and should NOT get the "Unable to load firmware" message anymore. However, repeat this process for every USB-port that you connect the Tacx to.

[image]

switchabl, 2021-01-17:
This should not be necessary anymore if you use the all-in-one driver package (https://github.com/switchabl/FortiusANT/releases/download/v4.0/libusb-win32_T19xx.exe, #168).

WouterJD, 2021-02-22:
Agree, and is now documented in manual

closed 2021-02-22 by WouterJD