WouterJD/FortiusANT #156: Run FortiusANT on MacOS

gpurinto, 2020-11-30:
Hello,

I'm new to all of this so please be patient with me. I've got an iMagic with the solid green head unit T1902. I've downloaded the FortiusANT software from GitHub to my MacBook - running Catalina v 10.15.7. I've also downloaded Python 3.9 and used Brew installed libusb - all per the instructions in the wiki. I've also purchased two ANT dongles - although I've not gotten to the point of using them yet.

Here's my trouble. From what I've read the instructions for starting the process on a Mac are as follows:
Install python 3 - **done**
Download the code - **which code?**
Run the code and see where we get - **Which code exactly and how do I run the code on a Mac?**

If I go to my FortiusANT folder and try to run any of the Windows Executable - ExplorANT.exe or FortiusANT.exe they obviously don't work as they are not recognizable to the Mac. Clearly I'm missing a step (again, be patient - I'm new to this stuff). 
Do I have to run a Windows environment on my MacBook to make this work, or is there another way to do this?

I'm hoping to run everything from my MacBook using the two dongle method.

Thank you in advance for any assistance!!

Greg Brown
Monroe, Maine USA

tjerkh, 2020-11-30:
Hello Greg,

You have already downloaded the code if you downloaded the FortiusANT Software from GitHub. You will need to run the code from the command line. You probably also need to install some required python modules/ 

These are the steps I did: 
```
git clone https://github.com/WouterJD/FortiusANT.git
cd FortiusANT-master/pythoncode
pip3 install -r requirements.txt
```

In case pygame fails: `python3 -m pip install -U pygame==2.0.0.dev6 --user`

Run FortiusAnt:
`python3 FortiusAnt.py -g`

You can skip the `git clone https://github.com/WouterJD/FortiusANT.git` since you already downloaded this.

Feel free to ask any more questions if you need to.

regards,
Tjerk

gpurinto, 2020-11-30:
Thanks Tjerk for the reply.

Sorry but much of this is foreign to me. I'm assuming I do this through Terminal on my MacBook?

I attempted the first part you provided - 
git clone https://github.com/WouterJD/FortiusANT.git
cd FortiusANT-master/pythoncode

it did something....

but then I got an error - 
[image]

I then tried the last command you suggested and got this
[image]

Thanks again for your help!

Greg

tjerkh, 2020-11-30:
Hello Greg,

Yes, you should do that in the terminal. I see there is a mistake in the step cd `FortiusANT-master/pythoncode` this should have been `cd FortiusANT/pythoncode`, just make sure you are in the folder pythoncode (in the terminal) before you execute  `pip3 install -r requirements.txt`. It is also best to execute all commands (lines) one by one, this way it's better to see when things fail.

gpurinto, 2020-11-30:
Ah OK. See I'm very new to this...

I ran steps in this order:
cd FortiusANT/pythoncode
then
pip3 install -r requirements.txt
then got this error....
[image]

Thank you for your help...making some progress I think.

tjerkh, 2020-11-30:
Unfortunately I don't know what the problem here is either. Something you could do is try to install the requirements one by one to see which module gives the problem:
```
pip3 install keyboard
pip3 install numpy
pip3 install pyusb
pip3 install pygame
pip3 install pynput
pip3 install wxPython
pip3 install wheel
```

gpurinto, 2020-11-30:
Hi Tjerkh,

Here's what I got - 

pip3 install keyboard - **error**
pip3 install numpy - installed
pip3 install pyusb - installed
pip3 install pygame - installed
pip3 install pynput - **error**
pip3 install wxPython - installed
pip3 install wheel - installed

[image]

[image]

gpurinto, 2020-11-30:
Success??!!

I ran the app and hooked up my iMagic. No cadence yet - no heart rate ( i haven't hooked those up yet) but it found my trainer and it looks like the dongles are working. This is exciting!! Now to get it all put together and try Zwift. Thank you!!!!!
[image]

WouterJD, 2020-12-10:
Checked with documentation what improvements could be made

closed 2020-12-10 by WouterJD

ted-westwood, 2021-01-04:
Hey Guys! I appear to be on the right track with this. I am yet to order the ANT dongles but I am getting this in terminal (GUI obviously loads). Are these lines anything to be concerned about before I order?

2021-01-04 00:07:18.610 Python[53946:5725895] Metal library creation failed: Error Domain=MTLLibraryErrorDomain Code=3 "compiler service could not be reached" UserInfo={NSLocalizedDescription=compiler service could not be reached}

2021-01-04 00:07:19.127 Python[53946:5725908] +[NSXPCSharedListener endpointForReply:withListenerName:]: an error occurred while attempting to obtain endpoint for listener 'ClientCallsAuxiliary': Connection invalid

Cheers,
Ted

ted-westwood, 2021-01-04:
Pressing Locate Hardware in the GUI also yields:

00:13:04,888: GetDongle - Exception: No backend available
00:13:04,889: GetDongle - Exception: No backend available
00:13:04,890: GetDongle - Exception: No backend available
00:13:04,894: GetTrainer - No backend, check libusb: No backend available

WouterJD, 2021-01-04:
Hi @ted-westwood 
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

WouterJD, 2021-02-01:
Does it all work?