WouterJD/FortiusANT #157: Question regarding heart rate support from Tacx head unit to CTPs

BrRoBo, 2020-12-01:
Hi,

I have the FortiusAnt running fine with my Fortius trainer and Zwift, plus 1 Suunto Ant+ from a few years ago and 1 new Anself ANT+ I recently bought via Amazon US.  No issues.  Along with this setup, I pair my Garmin Ant+ HR strap to Zwift to complete the needed devices for full telemetry in that CTP.

My question is why the HR data captured from the Tacx head unit is not rebroadcast as an Ant HR so that Zwift (and other CTP) could simply read that data stream?

Although I have an Ant+ HR strap, I would like to continue using my older straps that worked fine with the Tacx head unit, and not need to wear out my nicer/newer Ant+ straps.

Is it possible or of any interest to implement the HR data transfer from the Tacx head unit and transmit that as a Ant+ HR for pairing in a CTP?

WouterJD, 2020-12-01:
There are two modes:
- Use the Tacx connected strap and then FortiusANT broadcasts a HRM to be used by Zwift (and sends the HR-data in the FE-C interface)
- Use the HRM to display the HR in FortiusANT and connect the HRM to Zwift as well. (-H command line option)

So, yes you can use your old HRM. Do not use the -H command line option.

BrRoBo, 2020-12-03:
I have not tried with the -H option yet.  Simply been running the FortiusAnt.exe without any options.  

My old, non-Ant HR strap does display in the FortiusAnt gui and matches up with the flashing heart on the Tacx head unit.  However, although Zwift shows the Fortius trainer device (connected as a controllable) speed and power components, it does not report finding any HR device.  Tried HR search as well, without success.  

The only good HR so far is by using my Garmin Ant HR strap.

WouterJD, 2020-12-03:
I would expect FortiusANT to broadcast the heartrate and display the following message omn the console:
`FortiusANT broadcasts data as an ANT+ Heart Rate Monitor (HRM), id=57592`

If that's not the case, please send a logfile as created with the -d127 command-line flag.

BrRoBo, 2020-12-03:
Yes, based on the output in the console/command line window I am currently seeing, I thought it was odd to not see the broadcast line you mentioned above..  Currently, the only line mentioning HR is one saying it was ready to receive Ant+ HR.

BrRoBo, 2020-12-03:
Ok, I can confirm using "FortiusANT with -a -g -A" does start up with the HR **broadcast** message and in the gui shows it will use the Tacx HR.  

Without those extra startup options, the terminal window shows only that it will **receive** Ant+ HR, and the gui also shows it waiting for a Ant+ HR.

The FortiusAnt HR preference behavior does not seem to be correct since none of the "-a -g -A" startup options are described as effecting which type of HR will be used.

WouterJD, 2020-12-03:
I have marked this issue for documentation; will check on next round

BrRoBo, 2020-12-03:
So what I am seeing in the debug info for the startup options...

1 - By overriding/providing any startup options but omitting any  "-H" option, this triggers a HR = "None", and expects to use  the built-in Tacx HR and then will **broadcast** as an Ant+ HR signal.

2 - By not overriding/providing any startup options, "-H0" is assumed.  This will start up expecting an Ant+ HR, and then indicates the app will only receive Ant+ data, there is no broadcast of Ant+ HR.

3 - By specifying "-H-1", app will startup with the same "AntHRM - Heartrate expected from ANT+ HRM", but then will not bother to receive that Ant+ HR data.

log file attached with these 3 scenarios.

I suggest indicating in the documentation what is needed (and why) to use the Tacx HR instead of an Ant+ HR.

Thank you!

[FortiusAnt-debug HR options.txt](https://github.com/WouterJD/FortiusANT/files/5638671/FortiusAnt-debug.HR.options.txt)

WouterJD, 2020-12-10:
Manual describes: is now in #153 branch

-H HRM	Pair this Heart Rate Monitor (0: any, -1: none). Tacx HRM is used if not specified.
FortiusAnt -a -g		 will use Tacx heartrate monitor (analog HRM paired to Tacx)
FortiusAnt -a -g -H-1	 no HRM is used at all
FortiusAnt -a -g -H0	 FortiusANT will pair with first HRM found
FortiusAnt -a -g -H1234	 FortiusANT will pair with HRM with ID=1234

closed 2020-12-10 by WouterJD