WouterJD/FortiusANT #158: Fortius T1942 unable to load firmware

emptyc71, 2020-12-02:
Any help appreciated. I have the hex file for 1942 in the directory with the Windows exe file - but  I consistently receive the error "Unable to load Firmware". I am on Win 10 - I do not see a power management option re: wake for USB's, although I did enable USB wake option under power management  in BIOS. 
So how to I get the firmware into my T1942 (technically it is a T1940 but I doubt that matters). Thanks in advance

WouterJD, 2020-12-02:
@emptyc71 Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -
Sorry that I am not able to assist; I do not have this hardware - perhaps trying the python-version helps because then you can experiment, for example by using a longer waittime.

Perhaps other users with the T1942 can help:  @RamonFigueras @JordiCostaLo @bradyap @omedirk @PeterKassenaar @Hasoth @huishuis

Hasoth, 2020-12-02:
@emptyc71 You need to use "libusbK-dev-kit" and modify driver for your T1942 to enable wake-up option. I suggest to export .exe file and do not install from the wizard as it showed success but didn't work for me. When you have .exe file exported make sure to open it with the administrator rights. It worked for me. (please reade PDF carefully as it was in the user guide ;) )

[image]
[image]
[image]

WouterJD, 2020-12-02:
@Hasoth thanks for the support!
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

emptyc71, 2020-12-04:
Thank you but that did not work. Part of the issue is that the link to inf-wizard now has an older version, which is why I wasn't seeing the power option. I downloaded a separate EXE file for V2.0, and made a driver VR-Interface using parameters as above. However, the firmware still will not download. I have the hex file in the same directory as the Windows executable, and have tried the python-directory version as well. Both files have the initiation of disconnecting, then reconnecting the TACX trainer, but after a 5 second pause, I still receive the  error unable to download firmware. 
To be clear: 
1) The directory of my inf-wizard is below - I only was able to find the libusbK-inf-wizard.exe file, by itself, not able to find an entire installation package with this file. However, the installation matched the file above. 
[image]

2) I have run the Windows exectuable file in c:\Github\FortiusAnt\FortiusAnt-master\Startup and the WindowsExecutable directories with the 1942 hex file in those directories as well. 

I am out of ideas. TIme to buy a new trainer?

WouterJD, 2020-12-04:
Always a nice present for Christmas. FortiusANT tries to help to delay that investment.
A T1932 head unit might be enough...

Hasoth, 2020-12-04:
@emptyc71 
I've downloaded libusb from https://sourceforge.net/projects/libusbk/ 
It's not intent to copy this wizard to FortiusAnt folder or to store hex file there... You need to prepare USB driver for the device and then overwrite current one with the new one with changed power managment opion. After that you don't need wizard or the isntallation file for the driver.

emptyc71, 2020-12-05:
Excellent - thank you, thank you , thank you! There were 2 problems - the libusbk worked well, and forgotten to me, an old Fortius driver popped up. I must have installed TTS at some point a decade ago on this laptop. Worked very well last night. Hopefully will continue to do so, as I am familiar with Tacx drivers issues. Thank you so much - saves me a lot of money in the States. I can now add my name to comment # 11 re: where I am from - Asheville, North Carolina, known as "Beer City USA". Equipment is Tacx 1942, Garmin ANT+ x 2, and Bianchi CDI w Mirage 9 speed - old school for sure.

WouterJD, 2020-12-05:
Hurray @emptyc71 and welcome again!
So issue is: no tacx drivers?

Hasoth, 2020-12-05:
@emptyc71 great news! 👍

closed 2020-12-10 by WouterJD