WouterJD/FortiusANT #161: Using FortiusAnt and controlling an FE-C trainer

JibberJim, 2020-12-08:
First up, thanks for this, it's a great starting app for my needs, I've taken the code and have it controlling an FE-C trainer - a tacx neo - essentially working as a proxy for the Zwift control, I've done this because zwift doesn't send WindResistance that is useful to me and I want my neo to work with a lot more inertia than the default (ie a lower frontal area, so the same watts produce a faster speed in the tacx), it also allows me to limit gradients to a lower max gradient whilst still allowing a more normal feel on the lower gradients, as my same lack of inertia means I hate those 10% parts too.

Whilst FortiusAnt was a great starting place for my needs, I've realised it actually might be useful for others and it does make some sense to offer it up, so would people be interested in my tidying this up for more than my needs?

My other thing was wondering about how to offer remote control from a webpage that could be opened on any phone or other device - I was considering using a webpage talking websockets over MQTT to mosquitto, as it's a nice simple workflow that nicely disaccociates the two sides and runs really easy on the raspberry pi I use, but that may not be as simple for windows users, so I wondered if there was another obvious alternative that someone might have?

mattipee, 2020-12-08:
I have a Bushido Smart which is native ANT+ but want to use a single speed bike on it, so have been thinking about this too, an ANT+ proxy with which so simulate virtual gearing.

WouterJD, 2020-12-08:
Welcome to the FortiusANT community
- - -
I'm always curious to know who I communicate with, where FortiusANT is used and what configuration is used.
Please tell me what bundle did you buy, and what brake and what head unit do you use?
I would therefore appreciate that you introduce yourself; perhaps leave a comment under issue #14.
- - -

switchabl, 2020-12-10:
> 
> 
> I have a Bushido Smart which is native ANT+ but want to use a single speed bike on it, so have been thinking about this too, an ANT+ proxy with which to simulate virtual gearing. Originally thought I could do this by doing grade2power calcs in FortiusANT and then controlling trainer in ERG mode with a continuously changing set-point, but there may be a problems with that. I'm very interested to try your FE-C backend to try out some ideas.

@mattipee I'm facing a similar problem with the Genius (non-smart) implementation in #101, which is using something that is conceptually very similar to FE-C (just not compatible). There is no really straight-forward way to implement the FortiusANT virtual shifting.  I would love to hear your ideas.

I have also considered using power mode with grade2power but it may be too slow and I don't think it will accepts negative powers, so I wouldn't be able to drive the motor downhill (not a problem with the Bushido I guess).

My current idea is to modify the usual simulation parameters to achieve the same effect:
- changing the rider+bike mass should take care of rolling resistance, slope and intertia
`effective mass = gear ratio * mass`
If the effective mass ends up outside the supported range, slope and rolling resistance could also be adjusted directly:
- `effective slope = gear ratio * target slope`
- `effective c_rr = gear ratio * c_rr`
Wind resistance is where it gets tricky, because the relationship to speed is quadratic, so changing C_d doesn't work. So the idea is to manipulate the wind speed parameter:
`effective wind speed = wind speed + (gear ratio - 1) * speed`
so the brake uses the correct speed (`wind speed + gear ratio * speed`) for the drag calculation. The problem is that this needs to be updated continuously.

@JibberJim For just the wind resistance, you could probably get away with pairing the trainer directly to Zwift and just sending wind resistance pages occasionally. The advantage would be avoiding the additional latency and complexity of the FortiusANT layer and also avoid interfering with unrelated functionality (things like road feel would just continue to work without passing them through explicitly).

I have also entertained the idea of a modular framework with neatly decoupled USB trainer, ANT and BLE services exposing physical/virtual sensors and trainers through a standard MQTT interface, a controller service routing between them and a WebSocket/HTML or desktop GUI (or both) on top. That would be very easy to hack and to extend. Problem is that has very little to do with the current FortiusANT and realistically you would write that more or less from scratch. That is not something I am really keen on doing.

mattipee, 2020-12-10:
@switchabl the target grade needs to be left alone with virtual shifting, I think. In my head, virtual shifting (take my single speed for example, or Wouter's tendancy to stay in top physical gear all the time) is simply saying "trainer wheel speed can be higher, or lower, than simulated speed", but should present the correct amount of resistance as it would for the simulated speed on the simulated grade with the appropriate wind resistance, etc.

One issue is with inertia... The Fortius has a configurable virtual flywheel and its effects are noticeable. In this case, FortiusANT doesn't do the inertial calculation, the trainer does. I think I need to do something be here to get the setpoint correct.

The other issue I envisage is that in ERG mode, the Bushido wheel speed will be allowed to drift wrt the "geared" virtual speed, but I'm not sure whether the update frequency of the setpoint will mask that or not. I think it basically means that power becomes the only input from trainer to a FortiusANT inertial model on the way back up, to get back to what the new virtual speed is, but I don't know where the trainer's wheel speed fits in here.

Keen to have a play, anyway. I appear serially unable to resolve and articulate all the factors in my head, so do let me know if I'm either not making sense, or am missing something.

mattipee, 2020-12-10:
@oh, I see... You're trying to keep trainer in grade mode, and alter the params to compensate? Tricky. Maybe we should open a new issue...?

switchabl, 2020-12-10:
Well, the formula used internally by the trainer to calculate the required power in simulation mode (according to the FE-C specification) is basically this:
```
power = speed * (mass * grade / 100 * 9.81 m/s^2
+ mass * rolling resistance coefficient * 9.81 m/s^2
+ 0.5 * wind resistance coefficient * (speed + wind speed) * drafting factor)
```
I think virtual shifting just ideally just changing the speed to `gear ratio * speed`. But we can't tell the trainer what speed to use. So the idea is to adjust all the parameters to new "effective parameters" that produce the same effect. Easiest way (probably):
- `effective mass = gear ratio * mass`
- `effective wind speed = wind speed + (gear ratio - 1) * speed`
- `effective wind resistance coefficient = gear ratio * wind resistance coefficient` (that one I had missed earlier, that is why you should always write it down properly)

Maybe we can move this to #101 ? That is where I am working on the Genius currently.

JibberJim, 2020-12-10:
@switchabl yes, you do get away with it just sending the wind-resistance every now and then, on a seperate pair you're right, although I've got the virtual one with FortiusAnt involved and can do grade adjustment too (So I can ditch steep downhills and limit the steepest uphills too while keeping lower grades more "realistic" - ie something more complicated than just Zwift's limit.   I'm not passing through the 252 messages - as I hate the road feel too, the latency really doesn't seem any more noticeable than zwift's latency - I am using my power meter with zwift though and not the FE-C power so haven't yet re-implemented re-broadcasting that may make some difference.

It's working well, I'll get it actually into a fork soon, and will certainly look at the MQTT integration, as you say not directly into FortiusAnt, but there are actually still a lot of useful parts from FortiusAnt to get you there.

JibberJim, 2020-12-13:
So my fork now has working mqtt support (hard coded localhost) takes messages from /neo/wind/resistance and sets the WindResistance parameter in the trainer.  I have an Alexa node_red set up so I can say "alexa set neo to 31" and it changes the wind resistance to 0.31 - giving me lots more resistance, if I then find I'm spinning out, I can swap it to 0.41.

Surprisingly easy to add voice control - adding web page control with mqtt in browser web sockets will also be simple.

mattipee, 2020-12-13:
Hi @JibberJim .

I've fetched your branch and it seems to pair happily with my Bushido Smart. I noticed it wasn't picking up and rebroadcasting speed, cadence or power, but Zwift seemed to be content to control FortiusANT and get power/cadence from Bushido, but after a few hundred meters it kept dropping out. I tried a USB extension cable and also moving Bushido closer to the laptop but it wasn't happy.

I'll try again another day, but the new FE-C class was happy to pair with the Bushido, so I'm exited. Thanks so far!

switchabl, 2020-12-13:
Just a note: FE-C is probably very much a niche feature right now, but if we do get Bluetooth support this would allow FortiusANT to act as an ANT <-> BLE bridge which could be quite interesting as well.

JibberJim, 2020-12-14:
@mattipee I hadn't seen any drop outs before, but I did a run with my garmin recording the speed cadence sensor of the tacx, and that dropped out loads, so it may well be something I'm doing wrong - perhaps I need to acknowledge some of the messages, it looks like ExplorAnt does have some responses, so I probably need to look more into that.

JibberJim, 2020-12-15:
So I've made some more changes, I now have mqtt control of various things (min/max grade so you can decide to cap zwift if you're feeling knackered now and don't want  that 15% bit, without having to stop on zwift to access the menu, the actual target grade if you want to test without any zwift, as well as the wind stuff which was my main reason.)  the MQTT also gives me voice control of the features via alexa.

I still seem to be a bit strange with the pairing, whilst it's obviously paired and sending commands, there is the issue @mattipee mentioned above with dropouts of data being send, and there's no messages coming from the neo that I expect, the ANT FE-C spec makes me think I should have data pages 16 and 25 providing data about the trainer.  Yet all I ever see is page 238 - which I have no idea what it is, except it's not regular enough for any actual data.

Does anyone know anything about pairing with a neo and getting data off, who could point me in the right direction?

mattipee, 2020-12-15:
@JibberJim I'm not familiar enough with ANT+ to know the answer immediately.

As an aside, I discovered page 252 is used to set trainer road feel, and but I don't know what 238 is used for.

I've not been able to find time yet to look into it. Thanks for your efforts, I hope I can join you soon.

closed 2021-11-22 by WouterJD